dotintent/FlutterBleLib #10: package sets application@label value

emdete, 2018-03-15:
i get the error 

```
Attribute application@label value=... from AndroidManifest.xml:17:9-36
        is also present at [com.polidea.rxandroidble:rxandroidble:1.4.3] AndroidManifest.xml:16:18-50 value=(@string/app_name).
```

when compiling when using the package. without everything works fine.

KKonradMC, 2018-03-23:
Thanks for reporting this issue! We’re working on it, library is still in progress.

Regards
Konrad Krakowiak

alishir, 2018-06-12:
I have the same problem too.

pawelByszewski, 2019-11-01:
@emdete @alishir It has been a long time since we stopped maintaining the library but now we come with great news! We are currently working with Frontside on a new BLE Flutter library and BLE simulator. Beta version has been released already and we are going to release 2.0.0 in the first half of November

closed 2019-11-01 by pawelByszewski

Drkstr, 2020-01-10:
Still seeing this issue, can we reopen it, please ?

pawelByszewski, 2020-01-10:
@Drkstr could you provide the logs, please?

Drkstr, 2020-01-11:
Sure.
Here is the error log when I run a flutter project with the flutter_ble_lib package(version 2.0.1) without adding "tools:replace="android:label" to the manifest file. 

/Users/rakshak/personal_projects/ble_test_flutter/android/app/src/debug/AndroidManifest.xml:16:9-41 Error:
	Attribute application@label value=(ble_test_flutter) from AndroidManifest.xml:16:9-41
	is also present at [com.polidea.rxandroidble:rxandroidble:1.7.0] AndroidManifest.xml:16:18-50 value=(@string/app_name).
	Suggestion: add 'tools:replace="android:label"' to <application> element at AndroidManifest.xml:14:5-35:19 to override.

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':app:processDebugManifest'.
> Manifest merger failed : Attribute application@label value=(ble_test_flutter) from AndroidManifest.xml:16:9-41
  	is also present at [com.polidea.rxandroidble:rxandroidble:1.7.0] AndroidManifest.xml:16:18-50 value=(@string/app_name).
  	Suggestion: add 'tools:replace="android:label"' to <application> element at AndroidManifest.xml:14:5-35:19 to override.

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 5s
Finished with error: Gradle task assembleDebug failed with exit code 1

flutter doctor output 
Doctor summary (to see all details, run flutter doctor -v):
[✓] Flutter (Channel stable, v1.12.13+hotfix.5, on Mac OS X 10.14.5 18F132, locale en-AU)

[✓] Android toolchain - develop for Android devices (Android SDK version 29.0.2)
[✓] Xcode - develop for iOS and macOS (Xcode 11.2.1)
[✓] Android Studio (version 3.5)
[!] IntelliJ IDEA Ultimate Edition (version 2018.1.8)
    ✗ Flutter plugin not installed; this adds Flutter specific functionality.
    ✗ Dart plugin not installed; this adds Dart specific functionality.
[!] Connected device
    ! No devices available

! Doctor found issues in 2 categories.

mikolak, 2020-02-18:
@Drkstr should be fixed in 2.2.0

Drkstr, 2020-02-18:
Awesome thanks :D