dotintent/FlutterBleLib #13: Build failures, iOS

jack4git, 2018-03-20:
I am taking a project that was working with the flutter_blue library and attempting to switch it over to flutter_ble_lib but builds are failing.

First failure is on pod install, I get:
`Installing flutter_ble_lib (0.0.1)
[!] Pods written in Swift can only be integrated as frameworks; add `use_frameworks!` to your Podfile or target to opt into using it. The Swift Pods being used are: RxBluetoothKit, RxSwift, and flutter_ble_lib`

I tried adding add `use_frameworks!` to the Podfile but that leads to more problems relating to Swift:

`=== BUILD TARGET RxSwift OF PROJECT Pods WITH CONFIGURATION Debug ===
           
           Check dependencies
           The “Swift Language Version” (SWIFT_VERSION) build setting must be set to a supported value for targets which use Swift. This setting can be set in the build settings editor.
           
           
[   +1 ms] Could not build the precompiled application for the device.
[   +2 ms] Error launching application on JSI iPhone X.
[   +3 ms] "flutter run" took 8,823ms.

#0      throwToolExit (package:flutter_tools/src/base/common.dart:28)
#1      RunCommand.runCommand (package:flutter_tools/src/commands/run.dart:351)
<asynchronous suspension>
#2      FlutterCommand.verifyThenRunCommand (package:flutter_tools/src/runner/flutter_command.dart:290)
<asynchronous suspension>
#3      FlutterCommand.run (package:flutter_tools/src/runner/flutter_command.dart:228)
<asynchronous suspension>
#4      CommandRunner.runCommand (package:args/command_runner.dart:194)
<asynchronous suspension>
#5      FlutterCommandRunner.runCommand (package:flutter_tools/src/runner/flutter_command_runner.dart:286)
<asynchronous suspension>
#6      CommandRunner.run.<anonymous closure> (package:args/command_runner.dart:109)
#7      new Future.sync (dart:async/future.dart:222)
#8      CommandRunner.run (package:args/command_runner.dart:109)
#9      FlutterCommandRunner.run (package:flutter_tools/src/runner/flutter_command_runner.dart:166)
#10     run.<anonymous closure> (package:flutter_tools/runner.dart:90)
<asynchronous suspension>
#11     AppContext._run (package:flutter_tools/src/base/context.dart:76)
<asynchronous suspension>
#12     AppContext.runInZone.<anonymous closure> (package:flutter_tools/src/base/context.dart:66)
#13     _rootRun (dart:async/zone.dart:1126)
#14     _CustomZone.run (dart:async/zone.dart:1023)
#15     runZoned (dart:async/zone.dart:1501)
#16     AppContext.runInZone (package:flutter_tools/src/base/context.dart:65)
#17     run (package:flutter_tools/runner.dart:61)
<asynchronous suspension>
#18     main (package:flutter_tools/executable.dart:48)
<asynchronous suspension>
#19     main (file:///Users/jackivers/Projects/Flutter/flutter/packages/flutter_tools/bin/flutter_tools.dart:16)
#20     _startIsolate.<anonymous closure> (dart:isolate-patch/dart:isolate/isolate_patch.dart:277)
#21     _RawReceivePortImpl._handleMessage (dart:isolate-patch/dart:isolate/isolate_patch.dart:165)`

Flutter Doctor:
`[✓] Flutter (Channel beta, v0.1.5, on Mac OS X 10.13.3 17D102, locale en-US)
    • Flutter version 0.1.5 at /Users/jackivers/Projects/Flutter/flutter
    • Framework revision 3ea4d06340 (4 weeks ago), 2018-02-22 11:12:39 -0800
    • Engine revision ead227f118
    • Dart version 2.0.0-dev.28.0.flutter-0b4f01f759

[✓] Android toolchain - develop for Android devices (Android SDK 27.0.3)
    • Android SDK at /Users/jackivers/Library/Android/sdk
    • Android NDK location not configured (optional; useful for native profiling support)
    • Platform android-27, build-tools 27.0.3
    • Java binary at: /Applications/Android Studio.app/Contents/jre/jdk/Contents/Home/bin/java
    • Java version OpenJDK Runtime Environment (build 1.8.0_152-release-915-b08)

[✓] iOS toolchain - develop for iOS devices (Xcode 9.2)
    • Xcode at /Applications/Xcode.app/Contents/Developer
    • Xcode 9.2, Build version 9C40b
    • ios-deploy 1.9.2
    • CocoaPods version 1.4.0

[✓] Android Studio (version 3.0)
    • Android Studio at /Applications/Android Studio.app/Contents
    • Java version OpenJDK Runtime Environment (build 1.8.0_152-release-915-b08)

[✓] VS Code (version 1.21.1)
    • VS Code at /Applications/Visual Studio Code.app/Contents
    • Dart Code extension version 2.10.0

[✓] Connected devices (1 available)
    • JSI iPhone X • 7abb1638e5f94117408869c9c5236c5acb0d7f0c • ios • iOS 11.2.6`

jack4git, 2018-03-20:
I got a bit further by editing the Swift Language Version setting in targets RxSwift, RxBluetoothKit, and flutter_ble_lib (tried both 4.0 and 3.2) but this led to further problems:
Include of non-modular header inside framework module 'flutter_ble_lib': '/Users/jackivers/.pub-cache/hosted/pub.dartlang.org/flutter_ble_lib-1.0.0/ios/Classes/BleClientManager/BleClientManager.h'`

jack4git, 2018-03-20:
I got past this issue by following this suggestion:
[https://stackoverflow.com/questions/24103169/swift-compiler-error-non-modular-header-inside-framework-module?utm_medium=organic&utm_source=google_rich_qa&utm_campaign=google_rich_qa](url)
But this led to more issues:
`/Users/jackivers/.pub-cache/hosted/pub.dartlang.org/flutter_ble_lib-1.0.0/ios/Classes/BleData/Bledata.pbobjc.m:67:56: Cast of an indirect pointer to an Objective-C pointer to 'void *volatile *' is disallowed with ARC
/Users/jackivers/.pub-cache/hosted/pub.dartlang.org/flutter_ble_lib-1.0.0/ios/Classes/BleData/Bledata.pbobjc.m:68:15: ARC forbids explicit message send of 'release'
/Users/jackivers/.pub-cache/hosted/pub.dartlang.org/flutter_ble_lib-1.0.0/ios/Classes/BleData/Bledata.pbobjc.m:110:56: Cast of an indirect pointer to an Objective-C pointer to 'void *volatile *' is disallowed with ARC
/Users/jackivers/.pub-cache/hosted/pub.dartlang.org/flutter_ble_lib-1.0.0/ios/Classes/BleData/Bledata.pbobjc.m:111:15: ARC forbids explicit message send of 'release'
/Users/jackivers/.pub-cache/hosted/pub.dartlang.org/flutter_ble_lib-1.0.0/ios/Classes/BleData/Bledata.pbobjc.m:143:19: ARC forbids Objective-C objects in struct
/Users/jackivers/.pub-cache/hosted/pub.dartlang.org/flutter_ble_lib-1.0.0/ios/Classes/BleData/Bledata.pbobjc.m:214:13: ARC forbids Objective-C objects in struct
/Users/jackivers/.pub-cache/hosted/pub.dartlang.org/flutter_ble_lib-1.0.0/ios/Classes/BleData/Bledata.pbobjc.m:290:28: ARC forbids Objective-C objects in struct
/Users/jackivers/.pub-cache/hosted/pub.dartlang.org/flutter_ble_lib-1.0.0/ios/Classes/BleData/Bledata.pbobjc.m:369:13: ARC forbids Objective-C objects in struct
/Users/jackivers/.pub-cache/hosted/pub.dartlang.org/flutter_ble_lib-1.0.0/ios/Classes/BleData/Bledata.pbobjc.m:435:19: ARC forbids Objective-C objects in struct
/Users/jackivers/.pub-cache/hosted/pub.dartlang.org/flutter_ble_lib-1.0.0/ios/Classes/BleData/Bledata.pbobjc.m:486:13: ARC forbids Objective-C objects in struct
/Users/jackivers/.pub-cache/hosted/pub.dartlang.org/flutter_ble_lib-1.0.0/ios/Classes/BleData/Bledata.pbobjc.m:563:19: ARC forbids Objective-C objects in struct
/Users/jackivers/.pub-cache/hosted/pub.dartlang.org/flutter_ble_lib-1.0.0/ios/Classes/BleData/Bledata.pbobjc.m:623:13: ARC forbids Objective-C objects in struct
/Users/jackivers/.pub-cache/hosted/pub.dartlang.org/flutter_ble_lib-1.0.0/ios/Classes/BleData/Bledata.pbobjc.m:775:13: ARC forbids Objective-C objects in struct`

jack4git, 2018-03-20:
Any thoughts? This seems a long way from building cleanly ...

jack4git, 2018-03-20:
Note also, I get the same build failure on the example app:
`/Users/jackivers/Projects/Flutter/FlutterBleLib/ios/Classes/BleData/Bledata.pbobjc.m:67:56: Cast of an indirect pointer to an Objective-C pointer to 'void *volatile *' is disallowed with ARC
/Users/jackivers/Projects/Flutter/FlutterBleLib/ios/Classes/BleData/Bledata.pbobjc.m:68:15: ARC forbids explicit message send of 'release'
/Users/jackivers/Projects/Flutter/FlutterBleLib/ios/Classes/BleData/Bledata.pbobjc.m:110:56: Cast of an indirect pointer to an Objective-C pointer to 'void *volatile *' is disallowed with ARC
/Users/jackivers/Projects/Flutter/FlutterBleLib/ios/Classes/BleData/Bledata.pbobjc.m:111:15: ARC forbids explicit message send of 'release'
/Users/jackivers/Projects/Flutter/FlutterBleLib/ios/Classes/BleData/Bledata.pbobjc.m:143:19: ARC forbids Objective-C objects in struct
/Users/jackivers/Projects/Flutter/FlutterBleLib/ios/Classes/BleData/Bledata.pbobjc.m:214:13: ARC forbids Objective-C objects in struct
/Users/jackivers/Projects/Flutter/FlutterBleLib/ios/Classes/BleData/Bledata.pbobjc.m:290:28: ARC forbids Objective-C objects in struct
/Users/jackivers/Projects/Flutter/FlutterBleLib/ios/Classes/BleData/Bledata.pbobjc.m:369:13: ARC forbids Objective-C objects in struct
/Users/jackivers/Projects/Flutter/FlutterBleLib/ios/Classes/BleData/Bledata.pbobjc.m:435:19: ARC forbids Objective-C objects in struct
/Users/jackivers/Projects/Flutter/FlutterBleLib/ios/Classes/BleData/Bledata.pbobjc.m:486:13: ARC forbids Objective-C objects in struct
/Users/jackivers/Projects/Flutter/FlutterBleLib/ios/Classes/BleData/Bledata.pbobjc.m:563:19: ARC forbids Objective-C objects in struct
/Users/jackivers/Projects/Flutter/FlutterBleLib/ios/Classes/BleData/Bledata.pbobjc.m:623:13: ARC forbids Objective-C objects in struct
/Users/jackivers/Projects/Flutter/FlutterBleLib/ios/Classes/BleData/Bledata.pbobjc.m:775:13: ARC forbids Objective-C objects in struct
ARC Casting Rules Group
/Users/jackivers/Projects/Flutter/FlutterBleLib/ios/Classes/BleData/Bledata.pbobjc.m:67:48: Implicit conversion of Objective-C pointer type 'GPBEnumDescriptor *' to C pointer type 'void *' requires a bridged cast
/Users/jackivers/Projects/Flutter/FlutterBleLib/ios/Classes/BleData/Bledata.pbobjc.m:67:48: Use __bridge to convert directly (no change in ownership)
/Users/jackivers/Projects/Flutter/FlutterBleLib/ios/Classes/BleData/Bledata.pbobjc.m:67:48: Use CFBridgingRetain call to make an ARC object available as a +1 'void *'
/Users/jackivers/Projects/Flutter/FlutterBleLib/ios/Classes/BleData/Bledata.pbobjc.m:110:48: Implicit conversion of Objective-C pointer type 'GPBEnumDescriptor *' to C pointer type 'void *' requires a bridged cast
/Users/jackivers/Projects/Flutter/FlutterBleLib/ios/Classes/BleData/Bledata.pbobjc.m:110:48: Use __bridge to convert directly (no change in ownership)
/Users/jackivers/Projects/Flutter/FlutterBleLib/ios/Classes/BleData/Bledata.pbobjc.m:110:48: Use CFBridgingRetain call to make an ARC object available as a +1 'void *'
Semantic Issue Group
/Users/jackivers/Projects/Flutter/FlutterBleLib/ios/Classes/BleData/Bledata.pbobjc.m:68:15: 'release' is unavailable: not available in automatic reference counting mode
/Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Developer/SDKs/iPhoneOS11.2.sdk/usr/include/objc/NSObject.h:37:1: 'release' has been explicitly marked unavailable here
/Users/jackivers/Projects/Flutter/FlutterBleLib/ios/Classes/BleData/Bledata.pbobjc.m:111:15: 'release' is unavailable: not available in automatic reference counting mode`

jack4git, 2018-03-20:
Tracked down the ARC issues, saw a comment here that suggested turning off ARC:
[https://github.com/google/protobuf/issues/2127](url)
Did that (for target FlutterBleLib) and now it is building.

Can some or all of these required fixes be integrated into the package?

jack4git, 2018-03-20:
While the example builds, it doesn't run correctly. Gets the following rendering error:
`Xcode build done
══╡ EXCEPTION CAUGHT BY RENDERING LIBRARY ╞═════════════════════════════════════════════════════════
The following message was thrown during layout:
A RenderFlex overflowed by 2.0 pixels on the right.
The overflowing RenderFlex has an orientation of Axis.horizontal.
The edge of the RenderFlex that is overflowing has been marked in the rendering with a yellow and
black striped pattern. This is usually caused by the contents being too big for the RenderFlex.
Consider applying a flex factor (e.g. using an Expanded widget) to force the children of the
RenderFlex to fit within the available space instead of being sized to their natural size.
This is considered an error condition because it indicates that there is content that cannot be
seen. If the content is legitimately bigger than the available space, consider clipping it with a
ClipRect widget before putting it in the flex, or using a scrollable container rather than a Flex,
like a ListView.
`
Crashes when I hit the search button.

ryanhanks, 2018-03-26:
I experienced this also, but never got passed the point where disabling ARC gets it to build. @jack4git how did you disable ARC? Where did you disable it?

jack4git, 2018-03-26:
In Xcode, you need to select Pods and then the FlutterBleLib project, and then under build settings, search for "reference" and set Objective-C Automatic Reference Counting to No, see screenshot.

[image]

ryanhanks, 2018-03-27:
Thanks @jack4git, I'll try this out.

ryanhanks, 2018-03-27:
Thanks @jack4git, the above set of steps, all of which proved to be required, enabled me to successfully build the example app.

Upon doing so, I also experienced the crash. The example app is currently broken and crashes when you hit the search button.

ryanhanks, 2018-03-27:
@jack4git, it seems like your question above still needs to be answered.

Can the changes required to fix this be implemented by this library? Are changes outside of this library required for this to be fully fixed?

It seems like we have a number of things incompatibly configured here. It would be nice to hear from a knowledgeable contributor on this. @KKonradMC, is this issue on your radar at all? Any ideas on a fix?

FCMHUB, 2018-04-13:
Same here, excellent package, but only works on Android. Any news on this?

paweljaneczek, 2018-05-24:
Hey, could you try to run it from `feature/ios_fixes` branch? 
I've made there fixes to ARC problem. Also updated readme and added steps that needs to be made in order to make library work on iOS.

pawelByszewski, 2019-11-01:
@jack4git @ryanhanks @FCMHUB It has been a long time since we stopped maintaining the library but now we come with great news! We are currently working with Frontside on a new BLE Flutter library and BLE simulator. Beta version has been released already and we are going to release 2.0.0 in the first half of November

closed 2019-11-01 by pawelByszewski