dotintent/FlutterBleLib #15: iOS Crash on instance.startDeviceScan

jack4git, 2018-03-22:
I am getting an immediate crash as soon as `FlutterBleLib.instance.startDeviceScan` is called. Crash log attached. Same general symptoms as #13 -- immediate crash in FlutterBleLib code.
[Runner  3-22-18, 2-20 PM.crash.zip](https://github.com/Polidea/FlutterBleLib/files/1838955/Runner.3-22-18.2-20.PM.crash.zip)

`[✓] Flutter (Channel beta, v0.1.5, on Mac OS X 10.13.3 17D102, locale en-US)
    • Flutter version 0.1.5 at /Users/jackivers/Projects/Flutter/flutter
    • Framework revision 3ea4d06340 (4 weeks ago), 2018-02-22 11:12:39 -0800
    • Engine revision ead227f118
    • Dart version 2.0.0-dev.28.0.flutter-0b4f01f759

[✓] Android toolchain - develop for Android devices (Android SDK 27.0.3)
    • Android SDK at /Users/jackivers/Library/Android/sdk
    • Android NDK location not configured (optional; useful for native profiling support)
    • Platform android-27, build-tools 27.0.3
    • Java binary at: /Applications/Android Studio.app/Contents/jre/jdk/Contents/Home/bin/java
    • Java version OpenJDK Runtime Environment (build 1.8.0_152-release-915-b08)

[✓] iOS toolchain - develop for iOS devices (Xcode 9.2)
    • Xcode at /Applications/Xcode.app/Contents/Developer
    • Xcode 9.2, Build version 9C40b
    • ios-deploy 1.9.2
    • CocoaPods version 1.4.0

[✓] Android Studio (version 3.0)
    • Android Studio at /Applications/Android Studio.app/Contents
    • Java version OpenJDK Runtime Environment (build 1.8.0_152-release-915-b08)

[✓] VS Code (version 1.21.1)
    • VS Code at /Applications/Visual Studio Code.app/Contents
    • Dart Code extension version 2.10.0

[✓] Connected devices (1 available)
    • JSI iPhone X • 7abb1638e5f94117408869c9c5236c5acb0d7f0c • ios • iOS 11.2.6

• No issues found!
`

jack4git, 2018-03-22:
Hey Polidea, just pinging on the several issues posted on FlutterBleLib. Things are unstable / unusable for me, possibly something on my side but I would appreciate feedback / confirmation that you've testing on the current flutter beta channel. Neither my code nor the Polidea example code are stable for me.

dariuszseweryn, 2018-03-22:
Hey @jack4git 
Thanks for the reports! They are noticed
I am not directly involved with this lib but I will nag the right people
Best Regards

jack4git, 2018-03-22:
Thanks @dariuszseweryn !

pawelByszewski, 2019-11-01:
It has been a long time since we stopped maintaining the library but now we come with great news! We are currently working with Frontside on a new BLE Flutter library and BLE simulator. Beta version has been released already and we are going to release 2.0.0 in the first half of November

closed 2019-11-01 by pawelByszewski