dotintent/FlutterBleLib #16: Always getting isConnected = false.

PabloPL, 2018-03-24:
I'm using this library to connect to some arduino board with BLE support (https://www.dfrobot.com/wiki/index.php/Bluno_SKU:DFR0267).
After calling FlutterBleLib.instance.connectToDevice, i'm getting BleDevice object
which has isConnected always set to false.
The same is for FlutterBleLib.instance.onDeviceConnectionChanged().

I'm using this library by calling methods in following order
```
FlutterBleLib.instance.createClient
FlutterBleLib.instance.onDeviceConnectionChanged()
FlutterBleLib.instance.connectToDevice
```
And later 
```
FlutterBleLib.instance.discoverAllServicesAndCharacteristicsForDevice
FlutterBleLib.instance.writeCharacteristicForDevice
```
But this parts is working fine.

KKonradMC, 2018-03-27:
Can you paste full part of your code?
 The method `FlutterBleLib.instance.createClient(null)` returns Future object it is something like Promise in ReactNative. You can call next step int then block as is shown in example.
```
    FlutterBleLib.instance.createClient(null).then(
            (data) =>
            // Do something with data.
    );
```
https://github.com/Polidea/FlutterBleLib/blob/e24b1e175d5002aec981d1ccd68d51c97331801d/example/lib/ui/screen_on_boarding.dart

PabloPL, 2018-03-27:
At first in constructor of MyApp widget (which is building MaterialApp). i'm calling
```
FlutterBleLib.instance.createClient(null);
```

Later i start listening to device connection state change using following code
```
FlutterBleLib.instance
        .onDeviceConnectionChanged()
        .listen((device) => updateUi(device))
```
After connection i'm getting here (in updateUi) device object with correct device name but isConnected = false

Next in my app, after button click i'm connecting to device (which is working fine)
```
FlutterBleLib.instance.connectToDevice(_deviceMac, isAutoConnect: false)
        .then((dev) {
      FlutterBleLib.instance
          .discoverAllServicesAndCharacteristicsForDevice(_deviceMac)
          .then((device) {
             // here i update ui (enable buttons which i use to move robot)
             // the same issue here - after connection i'm getting device object but isConnected = false
```

Later, if  user press button (for example left/right/up/down arrow), i'm using following code to send command to device over ble (this is also working fine)
```
FlutterBleLib.instance
        .writeCharacteristicForDevice(
        _deviceMac, _serviceUUID, _characteristicUUID, data, false,
        new Uuid().v1());
```
In android logcat i can see that i'm correctly connecting to device.

PabloPL, 2018-05-18:
I got the same behaviour when using application from example directory. After clicking connect it's connecting to device but showing  "is connected: false".
By quick look at source code, answer is clear - isConnected field of BleDevice is never set.
Also BleDevice contains _id, name, mtu, rssi, isConnected_ fields but BleDeviceMessage contains only id, name rssi, mtu fields, so it's also never passed from android/ios framework.

I've locally implemented passing this value (and now it's working fine), but can't to this for iOS. Will be this accepted as PR?

PabloPL, 2018-05-23:
Fixed (for Android only) in https://github.com/Polidea/FlutterBleLib/pull/21. Closing.

closed 2018-05-23 by PabloPL