dotintent/FlutterBleLib #19: FlutterBleLib.instance.createClient method failed

inv-jithin, 2018-05-09:
**Failed to create FlutterBleLib instance using createClient(..) method.**. 
**_OS: Mac
Platform: Android
Device: Nexus 5_** 
 
**Exception stack trace**.

###############type 'Future<dynamic>' is not a subtype of type 'Future<Null>' where
  Future is from dart:async
  Future is from dart:async
  Null is from dart:core
EXCEPTION: _TypeError
type 'Future<dynamic>' is not a subtype of type 'Future<Null>' where
  Future is from dart:async
  Future is from dart:async
  Null is from dart:core
STACK TRACE:
#0      FlutterBleLib.createClient (file:///Users/user1/FlutterDevelopment/flutter/.pub-cache/hosted/pub.dartlang.org/flutter_ble_lib-1.0.0/lib/source/lib_core.dart:38:26)
#1      _MyHomePageState._incrementCounter (file:///Users/user1/FlutterDevelopment/FlutterProjects/flutter_ble/lib/main.dart:59:30)
#2      _InkResponseState._handleTap (package:flutter/src/material/ink_well.dart:494:14)
#3      _InkResponseState.build.<anonymous closure> (package:flutter/src/material/ink_well.dart:549:30)
#4      GestureRecognizer.invokeCallback (package:flutter/src/gestures/recognizer.dart:102:24)
#5      TapGestureRecognizer._checkUp (package:flutter/src/gestures/tap.dart:161:9)
#6      TapGestureRecognizer.handlePrimaryPointer (package:flutter/src/gestures/tap.dart:94:7)
#7      PrimaryPointerGestureRecognizer.handleEvent (package:flutter/src/gestures/recognizer.dart:315:9)
#8      PointerRouter._dispatch (package:flutter/src/gestures/pointer_router.dart:73:12)
#9      PointerRouter.route (package:flutter/

PabloPL, 2018-05-18:
Isn't it the same issue as https://github.com/Polidea/FlutterBleLib/issues/18 ?
If yes, just disable Dart 2 mode (in AndroidStudio go to Settings -> Languages & Framework -> Flutter -> Run applications in Dart 2.0 mode: -> Disable Dart 2.

KKonradMC, 2018-05-24:
Fixed in newest version

closed 2018-05-24 by KKonradMC