dotintent/FlutterBleLib #24: Is it possible to receive BLE service notification while activity running in background?

alishir, 2018-06-13:
Is it possible to receive BLE service notification in the activity that is running in background? I want to continuously monitor hear rate measurement while the user switching between different applications.

kitolog, 2018-12-19:
Same here
I need to read characteristics updates running in background, iOS

mikolak, 2019-11-13:
@alishir 
You may need to use foreground service to keep your application alive and avoid background work limit.

@kitolog 
[This link should help you](https://developer.apple.com/library/archive/documentation/NetworkingInternetWeb/Conceptual/CoreBluetooth_concepts/CoreBluetoothBackgroundProcessingForIOSApps/PerformingTasksWhileYourAppIsInTheBackground.html)

closed 2019-11-13 by mikolak

Lootwig, 2021-01-22:
For anyone arriving through googling, https://github.com/Polidea/FlutterBleLib/issues/530#issuecomment-700412023 worked for me!