dotintent/FlutterBleLib #25: example app crashes on disconnect (ios only)

marcmo, 2018-06-19:
steps to reproduce:
build included example, run on ios, connect -> disconnect

seems you get your types wrong when calling `convertToBleDeviceMessage`. this is the function:
```
+ (BleDataServiceMessage* _Nonnull) convertBleDataServiceMessage: (id _Nonnull) value {
    BleDataServiceMessage* bleDataServiceMessage = [[BleDataServiceMessage alloc] init];
    bleDataServiceMessage.id_p = ((NSNumber *) value[@"id"]).doubleValue;
    bleDataServiceMessage.device =  [[BleDataBleDeviceMessage alloc] init];
    bleDataServiceMessage.device.id_p = value[@"deviceID"];
    bleDataServiceMessage.uuid = value[@"uuid"];
    bleDataServiceMessage.isPrimary = value[@"isPrimary"];
    
    return bleDataServiceMessage;
}
```
connecting you handle it correclty:
[image]
but not so much when disconnecting:
[image]

I saw you have started some major refactorings for ios on the `refactoring_ios` branch...are you still thinking about continuing your efforts? seems this has gone stale

NOA-Charles, 2018-07-06:
Is this problem solved? I also encountered the same problem.

NOA-Charles, 2018-07-06:
Dude, I saw your update in pull requests.

NOA-Charles, 2018-07-06:
This solves the problem, thank you

pawelByszewski, 2019-11-01:
@marcmo It has been a long time since we stopped maintaining the library but now we come with great news! We are currently working with Frontside on a new BLE Flutter library and BLE simulator. Beta version has been released already and we are going to release 2.0.0 in the first half of November

closed 2019-11-01 by pawelByszewski