dotintent/FlutterBleLib #28: Running example with "flutter run" fails on both Android and iOS.

ssarangi, 2018-06-25:
On the surface, it feels like a flutter issue but it is not. Flutter run fails to install on both iOS (running 11.4) and Android device. Running the example project directly with xcode succeeds but unfortunately I can't debug any issues with it.

Launches on simulator as well. Also, all the other flutter apps I have run fine on the device so I am guessing something with the project config is off.

duncanfinney, 2018-08-03:
I had to change the bundle identifier and the provisioning profile in the project and then it worked for me.

pawelByszewski, 2019-11-01:
@ssarangi It has been a long time since we stopped maintaining the library but now we come with great news! We are currently working with Frontside on a new BLE Flutter library and BLE simulator. Beta version has been released already and we are going to release 2.0.0 in the first half of November

closed 2019-11-01 by pawelByszewski