dotintent/FlutterBleLib #29: Writing to a BT Characteristic fails encoding

ssarangi, 2018-06-25:
While running with xcode on device, I discover my BT device, connect to it and find the service I want to. On trying to write to the "characteristic" a hex value of 0x289 / 0x0289 / 0x2, all of them fail with the following error. 

Invalid argument (Not a byte value at index 0: 0x289): Instance(length:1) of '_GrowableList'
#0      _Base64Encoder.encodeChunk (dart:convert/base64.dart:392:5)
#1      _Base64Encoder.encode (dart:convert/base64.dart:346:9)
#2      Base64Encoder.convert (dart:convert/base64.dart:253:32)
#3      Codec.encode (dart:convert/codec.dart:25:32)
#4      FlutterBleLib.writeCharacteristic (file:///Users/satyajits/dev/flutter/FlutterBleLib/lib/source/lib_core.dart:268:30)

This however works with the ios bluetooth app, LightBlue.

pawelByszewski, 2019-11-01:
It has been a long time since we stopped maintaining the library but now we come with great news! We are currently working with Frontside on a new BLE Flutter library and BLE simulator. Beta version has been released already and we are going to release 2.0.0 in the first half of November

closed 2019-11-01 by pawelByszewski

gersonfa, 2020-03-10:
Sorry, but why this is closed? How it can be fixed?

mikolak, 2020-03-10:
Does it happen? Do you have logs? 

Whole library has been rewritten from scratch between August and October, that's why this is closed.