dotintent/FlutterBleLib #34: Android Crash on connect

crosas-krasamo, 2018-07-13:
When trying to connect, sometimes, I'm getting the attached crash, I haven't identified any steps but it's quite constant, I can see the same crash when I try to cancel the connection (disconnect) after requesting a connection.

`D/BluetoothGatt( 1277): onClientConnectionState() - status=133 clientIf=9 device=20:91:48:CA:DF:XX
D/BluetoothManager( 1277): getConnectionState()
D/BluetoothManager( 1277): getConnectedDevices
W/BluetoothGatt( 1277): Unhandled exception in callback
W/BluetoothGatt( 1277): rx.exceptions.OnErrorFailedException: Error occurred when trying to propagate error to Observer.onError
W/BluetoothGatt( 1277): 	at rx.observers.SafeSubscriber._onError(SafeSubscriber.java:187)
W/BluetoothGatt( 1277): 	at rx.observers.SafeSubscriber.onError(SafeSubscriber.java:115)
W/BluetoothGatt( 1277): 	at rx.observers.Subscribers$5.onError(Subscribers.java:230)
W/BluetoothGatt( 1277): 	at rx.observers.Subscribers$5.onError(Subscribers.java:230)
W/BluetoothGatt( 1277): 	at rx.observers.Subscribers$5.onError(Subscribers.java:230)
W/BluetoothGatt( 1277): 	at rx.observers.Subscribers$5.onError(Subscribers.java:230)
W/BluetoothGatt( 1277): 	at rx.internal.operators.OperatorUnsubscribeOn$1.onError(OperatorUnsubscribeOn.java:47)
W/BluetoothGatt( 1277): 	at rx.internal.operators.OperatorSubscribeOn$SubscribeOnSubscriber.onError(OperatorSubscribeOn.java:80)
W/BluetoothGatt( 1277): 	at rx.observers.Subscribers$5.onError(Subscribers.java:230)
W/BluetoothGatt( 1277): 	at rx.observers.Subscribers$5.onError(Subscribers.java:230)
W/BluetoothGatt( 1277): 	at rx.internal.operators.OperatorMerge$MergeSubscriber.reportError(OperatorMerge.java:266)
W/BluetoothGatt( 1277): 	at rx.internal.operators.OperatorMerge$MergeSubscriber.checkTerminate(OperatorMerge.java:818)
W/BluetoothGatt( 1277): 	at rx.internal.operators.OperatorMerge$MergeSubscriber.emitLoop(OperatorMerge.java:579)
W/BluetoothGatt( 1277): 	at rx.internal.operators.OperatorMerge$MergeSubscriber.emit(OperatorMerge.java:568)
W/BluetoothGatt( 1277): 	at rx.internal.operators.OperatorMerge$InnerSubscriber.onError(OperatorMerge.java:852)
W/BluetoothGatt( 1277): 	at rx.internal.operators.OperatorMerge$MergeSubscriber.reportError(OperatorMerge.java:266)
W/BluetoothGatt( 1277): 	at rx.internal.operators.OperatorMerge$MergeSubscriber.checkTerminate(OperatorMerge.java:818)
W/BluetoothGatt( 1277): 	at rx.internal.operators.OperatorMerge$MergeSubscriber.emitLoop(OperatorMerge.java:579)
W/BluetoothGatt( 1277): 	at rx.internal.operators.OperatorMerge$MergeSubscriber.emit(OperatorMerge.java:568)
W/BluetoothGatt( 1277): 	at rx.internal.operators.OperatorMerge$InnerSubscriber.onError(OperatorMerge.java:852)
W/BluetoothGatt( 1277): 	at rx.internal.operators.OnSubscribeThrow.call(OnSubscribeThrow.java:44)
W/BluetoothGatt( 1277): 	at rx.internal.operators.OnSubscribeThrow.call(OnSubscribeThrow.java:28)
W/BluetoothGatt( 1277): 	at rx.Observable.unsafeSubscribe(Observable.java:10256)
W/BluetoothGatt( 1277): 	at rx.internal.operators.OperatorMerge$MergeSubscriber.onNext(OperatorMerge.java:248)
W/BluetoothGatt( 1277): 	at rx.internal.operators.OperatorMerge$MergeSubscriber.onNext(OperatorMerge.java:148)
W/BluetoothGatt( 1277): 	at rx.internal.operators.OnSubscribeMap$MapSubscriber.onNext(OnSubscribeMap.java:77)
W/BluetoothGatt( 1277): 	at rx.internal.operators.NotificationLite.accept(NotificationLite.java:135)
W/BluetoothGatt( 1277): 	at rx.internal.operators.OperatorReplay$UnboundedReplayBuffer.replay(OperatorReplay.java:903)
W/BluetoothGatt( 1277): 	at rx.internal.operators.OperatorReplay$ReplaySubscriber.replay(OperatorReplay.java:636)
W/BluetoothGatt( 1277): 	at rx.internal.operators.OperatorReplay$ReplaySubscriber.onNext(OperatorReplay.java:452)
W/BluetoothGatt( 1277): 	at rx.internal.operators.OperatorMerge$MergeSubscriber.emitScalar(OperatorMerge.java:395)
W/BluetoothGatt( 1277): 	at rx.internal.operators.OperatorMerge$MergeSubscriber.tryEmit(OperatorMerge.java:355)
W/BluetoothGatt( 1277): 	at rx.internal.operators.OperatorMerge$InnerSubscriber.onNext(OperatorMerge.java:846)
W/BluetoothGatt( 1277): 	at com.jakewharton.rxrelay.RelaySubscriptionManager$RelayObserver.onNext(RelaySubscriptionManager.java:205)
W/BluetoothGatt( 1277): 	at com.jakewharton.rxrelay.PublishRelay.call(PublishRelay.java:47)
W/BluetoothGatt( 1277): 	at com.polidea.rxandroidble.internal.connection.DisconnectionRouter.onDisconnectedException(DisconnectionRouter.java:65)
W/BluetoothGatt( 1277): 	at com.polidea.rxandroidble.internal.connection.RxBleGattCallback$2.onConnectionStateChange(RxBleGattCallback.java:76)
W/BluetoothGatt( 1277): 	at android.bluetooth.BluetoothGatt$1.onClientConnectionState(BluetoothGatt.java:228)
W/BluetoothGatt( 1277): 	at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:70)
W/BluetoothGatt( 1277): 	at android.os.Binder.execTransact(Binder.java:573)
W/BluetoothGatt( 1277): Caused by: rx.exceptions.CompositeException: 2 exceptions occurred. 
W/BluetoothGatt( 1277): 	... 40 more
W/BluetoothGatt( 1277): Caused by: rx.exceptions.CompositeException$CompositeExceptionCausalChain: Chain of Causes for CompositeException In Order Received =>
W/BluetoothGatt( 1277): 	at android.util.Log.printlns(Log.java:584)
W/BluetoothGatt( 1277): 	at android.util.Log.w(Log.java:276)
W/BluetoothGatt( 1277): 	at android.bluetooth.BluetoothGatt$1.onClientConnectionState(BluetoothGatt.java:230)
W/BluetoothGatt( 1277): 	... 2 more
W/BluetoothGatt( 1277): Caused by: com.polidea.rxandroidble.exceptions.BleDisconnectedException: Disconnected from 20:91:48:CA:DF:6F
W/BluetoothGatt( 1277): 	at com.polidea.rxandroidble.internal.connection.RxBleGattCallback$2.onConnectionStateChange(RxBleGattCallback.java:76)
W/BluetoothGatt( 1277): 	at android.bluetooth.BluetoothGatt$1.onClientConnectionState(BluetoothGatt.java:228)
W/BluetoothGatt( 1277): 	... 2 more
W/BluetoothGatt( 1277): Caused by: java.lang.IllegalArgumentException: Unsupported value: com.polidea.rxandroidble.exceptions.BleDisconnectedException: Disconnected from 20:91:48:CA:DF:6F
W/BluetoothGatt( 1277): 	at io.flutter.plugin.common.StandardMessageCodec.writeValue(StandardMessageCodec.java:293)
W/BluetoothGatt( 1277): 	at io.flutter.plugin.common.StandardMethodCodec.encodeErrorEnvelope(StandardMethodCodec.java:70)
W/BluetoothGatt( 1277): 	at io.flutter.plugin.common.MethodChannel$IncomingMethodCallHandler$1.error(MethodChannel.java:199)
W/BluetoothGatt( 1277): 	at com.polidea.flutterblelib.FlutterBleLibPlugin$DefaultOnErrorAction.onError(FlutterBleLibPlugin.java:449)
W/BluetoothGatt( 1277): 	at com.polidea.flutterblelib.utils.SafeAction.onError(SafeAction.java:21)
W/BluetoothGatt( 1277): 	at com.polidea.flutterblelib.BleHelper$11.onError(BleHelper.java:401)
W/BluetoothGatt( 1277): 	at rx.internal.util.ObserverSubscriber.onError(ObserverSubscriber.java:39)
W/BluetoothGatt( 1277): 	at rx.observers.SafeSubscriber._onError(SafeSubscriber.java:153)
W/BluetoothGatt( 1277): 	at rx.observers.SafeSubscriber.onError(SafeSubscriber.java:115)
W/BluetoothGatt( 1277): 	at rx.observers.Subscribers$5.onError(Subscribers.java:230)
W/BluetoothGatt( 1277): 	at rx.observers.Subscribers$5.onError(Subscribers.java:230)
[... 10 lines cut ...]
W/BluetoothGatt( 1277): 	at rx.internal.operators.OperatorMerge$InnerSubscriber.onError(OperatorMerge.java:852)
W/BluetoothGatt( 1277): 	at rx.internal.operators.OperatorMerge$MergeSubscriber.reportError(OperatorMerge.java:266)
W/BluetoothGatt( 1277): 	at rx.internal.operators.OperatorMerge$MergeSubscriber.checkTerminate(OperatorMerge.java:818)
W/BluetoothGatt( 1277): 	at rx.internal.operators.OperatorMerge$MergeSubscriber.emitLoop(OperatorMerge.java:579)
W/BluetoothGatt( 1277): 	at rx.internal.operators.OperatorMerge$MergeSubscriber.emit(OperatorMerge.java:568)
W/BluetoothGatt( 1277): 	at rx.internal.operators.OperatorMerge$InnerSubscriber.onError(OperatorMerge.java:852)
W/BluetoothGatt( 1277): 	at rx.internal.operators.OnSubscribeThrow.call(OnSubscribeThrow.java:44)
W/BluetoothGatt( 1277): 	at rx.internal.operators.OnSubscribeThrow.call(OnSubscribeThrow.java:28)
W/BluetoothGatt( 1277): 	at rx.Observable.unsafeSubscribe(Observable.java:10256)
W/BluetoothGatt( 1277): 	at rx.internal.operators.OperatorMerge$MergeSubscriber.onNext(OperatorMerge.java:248)
W/BluetoothGatt( 1277): 	at rx.internal.operators.OperatorMerge$MergeSubscriber.onNext(OperatorMerge.java:148)
[... cut ...]
W/BluetoothGatt( 1277): 	at rx.internal.operators.OperatorMerge$InnerSubscriber.onNext(OperatorMerge.java:846)
W/BluetoothGatt( 1277): 	at com.jakewharton.rxrelay.RelaySubscriptionManager$RelayObserver.onNext(RelaySubscriptionManager.java:205)
W/BluetoothGatt( 1277): 	at com.jakewharton.rxrelay.PublishRelay.call(PublishRelay.java:47)
W/BluetoothGatt( 1277): 	at com.polidea.rxandroidble.internal.connection.DisconnectionRouter.onDisconnectedException(DisconnectionRouter.java:65)
W/BluetoothGatt( 1277): 	... 4 more
D/BluetoothGatt( 1277): close()
D/BluetoothGatt( 1277): unregisterApp() - mClientIf=9`

JamesMcIntosh, 2018-09-26:
This is the part of the stack trace which outlines the cause of the problem:
```
     Caused by: java.lang.IllegalArgumentException: Unsupported value: com.polidea.rxandroidble.exceptions.BleDisconnectedException: Disconnected from 88:6B:0F:00:A7:DD with status -1 (UNKNOWN)
        at io.flutter.plugin.common.StandardMessageCodec.writeValue(StandardMessageCodec.java:293)
        at io.flutter.plugin.common.StandardMethodCodec.encodeErrorEnvelope(StandardMethodCodec.java:70)
        at io.flutter.plugin.common.MethodChannel$IncomingMethodCallHandler$1.error(MethodChannel.java:199)
        at com.polidea.flutterblelib.FlutterBleLibPlugin$DefaultOnErrorAction.onError(FlutterBleLibPlugin.java:449)
```

In FlutterBleLibPlugin.java, line number 449 should not be passing in the exception as the third parameter because the StandardMessageCodec doesn't know how to parse it.
```
public void onError(Throwable t) {
  this.result.error("Error occurred", t.getMessage(), t);
}
```

The interface definition is a bit vague, see:
https://github.com/flutter/engine/blob/b1d3c98fd0ccecb4730de65d268fe87ddf8eea63/shell/platform/android/io/flutter/plugin/common/MethodChannel.java#L147
```
/**
 * Handles an error result.
 *
 * @param errorCode An error code String.
 * @param errorMessage A human-readable error message String, possibly null.
 * @param errorDetails Error details, possibly null
 */
void error(String errorCode, String errorMessage, Object errorDetails);
```
My recommended solution is to just pass null in as the third parameter.

JamesMcIntosh, 2018-09-26:
Here is the fix in my fork of the repository
https://github.com/JamesMcIntosh/FlutterBleLib/commit/3a61b2006968656ebccb54c5ff46a0b3d93456d4

pawelByszewski, 2019-11-01:
@crosas-dev @JamesMcIntosh It has been a long time since we stopped maintaining the library but now we come with great news! We are currently working with Frontside on a new BLE Flutter library and BLE simulator. Beta version has been released already and we are going to release 2.0.0 in the first half of November

closed 2019-11-01 by pawelByszewski