dotintent/FlutterBleLib #36: FlutterBleLib.instance.createClient(null) crashes on Android

schdck, 2018-07-24:
This is very similar to #19, if not equal.

Don't know if I'm doing something wrong (likely) or the issue is actually back.

The steps to reproduce are:
 1. Create a new Flutter App on Android Studio
 2. Change the Android `minSdkVersion` to 18 (build.gradle)
 3. Add `tools:replace="android:label"` to the application tag (in AndroidManifest.xml)
 4. Replace your `main.dart` with [this](https://gist.github.com/schdck/7f16a5dcc07db472223fb8870c4ecb11) one

I don't know if Items 2 and 3 are necessary, but I wasn't able to debug without them.

When I click the button that calls `_createClient` (and `FlutterBleLib.instance.createClient(null)`) I get the following:

    I/flutter (12924): ══╡ EXCEPTION CAUGHT BY GESTURE ╞═══════════════════════════════════════════════════════════════════
    I/flutter (12924): The following assertion was thrown while handling a gesture:
    I/flutter (12924): type 'Future<dynamic>' is not a subtype of type 'Future<Null>'
    I/flutter (12924): 
    I/flutter (12924): Either the assertion indicates an error in the framework itself, or we should provide substantially
    I/flutter (12924): more information in this error message to help you determine and fix the underlying cause.
    I/flutter (12924): In either case, please report this assertion by filing a bug on GitHub:
    I/flutter (12924):   https://github.com/flutter/flutter/issues/new
    I/flutter (12924): 
    I/flutter (12924): When the exception was thrown, this was the stack:
    I/flutter (12924): #0      FlutterBleLib.createClient (file:///C:/flutter/.pub-cache/hosted/pub.dartlang.org/flutter_ble_lib-1.0.0/lib/source/lib_core.dart:38:26)
    I/flutter (12924): #1      _MyHomePageState._createClient (file:///C:/Users/mauri/AndroidStudioProjects/irrigacao_ble/lib/main.dart:31:28)
    I/flutter (12924): #2      _InkResponseState._handleTap (package:flutter/src/material/ink_well.dart:494:14)
    I/flutter (12924): #3      _InkResponseState.build.<anonymous closure> (package:flutter/src/material/ink_well.dart:549:30)
    I/flutter (12924): #4      GestureRecognizer.invokeCallback (package:flutter/src/gestures/recognizer.dart:102:24)
    I/flutter (12924): #5      TapGestureRecognizer._checkUp (package:flutter/src/gestures/tap.dart:161:9)
    I/flutter (12924): #6      TapGestureRecognizer.handlePrimaryPointer (package:flutter/src/gestures/tap.dart:94:7)
    I/flutter (12924): #7      PrimaryPointerGestureRecognizer.handleEvent (package:flutter/src/gestures/recognizer.dart:315:9)
    I/flutter (12924): #8      PointerRouter._dispatch (package:flutter/src/gestures/pointer_router.dart:73:12)
    I/flutter (12924): #9      PointerRouter.route (package:flutter/src/gestures/pointer_router.dart:101:11)
    I/flutter (12924): #10     _WidgetsFlutterBinding&BindingBase&GestureBinding.handleEvent (package:flutter/src/gestures/binding.dart:143:19)
    I/flutter (12924): #11     _WidgetsFlutterBinding&BindingBase&GestureBinding.dispatchEvent (package:flutter/src/gestures/binding.dart:121:22)
    I/flutter (12924): #12     _WidgetsFlutterBinding&BindingBase&GestureBinding._handlePointerEvent (package:flutter/src/gestures/binding.dart:101:7)
    I/flutter (12924): #13     _WidgetsFlutterBinding&BindingBase&GestureBinding._flushPointerEventQueue (package:flutter/src/gestures/binding.dart:64:7)
    I/flutter (12924): #14     _WidgetsFlutterBinding&BindingBase&GestureBinding._handlePointerDataPacket (package:flutter/src/gestures/binding.dart:48:7)
    I/flutter (12924): #15     _invoke1 (dart:ui/hooks.dart:134:13)
    I/flutter (12924): #16     _dispatchPointerDataPacket (dart:ui/hooks.dart:91:5)
    I/flutter (12924): 
    I/flutter (12924): Handler: onTap
    I/flutter (12924): Recognizer:
    I/flutter (12924):   TapGestureRecognizer#f8bd3(debugOwner: GestureDetector, state: possible, won arena, finalPosition:
    I/flutter (12924):   Offset(332.0, 597.3), sent tap down)
    I/flutter (12924): ════════════════════════════════════════════════════════════════════════════════════════════════════

npanagop, 2018-08-09:
I was having the same issue. Pull request https://github.com/Polidea/FlutterBleLib/pull/38 fixed it for me.

pawelByszewski, 2019-11-01:
@schdck @npanagop It has been a long time since we stopped maintaining the library but now we come with great news! We are currently working with Frontside on a new BLE Flutter library and BLE simulator. Beta version has been released already and we are going to release 2.0.0 in the first half of November

closed 2019-11-01 by pawelByszewski