dotintent/FlutterBleLib #42: Compatability Issue

roshan-rajendran, 2018-08-29:
The current Dart SDK version is 2.1.0-dev.0.0.flutter-be6309690f.

Because flutter_ble_lib depends on protobuf >=0.2.0 <=0.9.0 which requires SDK version >=0.7.5 <2.0.0-∞, version solving failed.
pub upgrade failed (1)

roshan-rajendran, 2018-08-29:
Temporarily solved it like this xD 

dependencies:
  # ...
  xml: any 
  intl_translation: any

closed 2018-08-29 by roshan-rajendran

npanagop, 2018-09-06:
Hi @roshan-rajendran ,
I'm having the same issue, can you explain your solution a bit better?

roshan-rajendran, 2018-09-06:
Hi @npanagop 
The problem is the version of dart which is creating an issue . 
So  change  the following and follow the rest of the procedures like get dependencies and build the app . You shouldn't have any issues :). 

1. In the pubspec.yaml of the site directory change the dart version to which ever you are using like so . 

**Before**
```
environment:
  sdk: ">=1.8.0 <2.0.0"
```
**After**
```
environment:
  sdk: ">=1.8.0 <2.1.0"
```
2. Go to the pubspec.yaml file in the test directory and change the following 
**Before**
```
dependencies:
  flutter:
    sdk: flutter
  cupertino_icons: ^0.1.0
  uuid: 0.5.3
```
**After**
```
dependencies:
  flutter:
    sdk: flutter
  cupertino_icons: ^0.1.0
  uuid: any
```