dotintent/FlutterBleLib #50: Flutter support

votruk, 2019-06-28:
Any updates of when you are planning to continue to support this library?

Btw I'm a huge fan of your android lib and it's very frustrating to see that you cannot use your library on flutter.

passos, 2019-07-12:
looking for flutter update +1

pawelByszewski, 2019-11-01:
@votruk @passos It has been a long time since we stopped maintaining the library but now we come with great news! We are currently working with Frontside on a new BLE Flutter library and BLE simulator. Beta version has been released already and we are going to release 2.0.0 in the first half of November

closed 2019-11-01 by pawelByszewski

votruk, 2019-11-01:
Such great news! I was already looking after your progress and documentation for some time. You do a great job, thank you!

pawelByszewski, 2019-11-01:
Thank you @votruk, if you are going to use the lib then please take a look at BLE Simulator (https://github.com/Polidea/blemulator_flutter)

passos, 2019-11-03:
cool~ that's really a great news. Thanks