dotintent/FlutterBleLib #295: bleManager.connectToPeripheral(String peripheralId)

mikolak, 2019-10-22:


wl386123298, 2020-07-10:
the method is not exist in BleManager

mikolak, 2020-07-10:
No, it does not. This issue is to create it.

wl386123298, 2020-07-10:
i have a idea :
filter scan -- > connect --> done

mikolak, 2020-07-10:
The idea behind this method is to avoid scanning and try to connect to a peripheral known from previous sessions.

wl386123298, 2020-07-10:
yes, this method is developping ?

mikolak, 2020-07-10:
It will be created, yes. I'm unable to give you any timeline, we're stretched a little thin.

wl386123298, 2020-07-10:
greate

cameronkloot, 2020-07-23:
This is a feature I needed for our apps. Creating a peripheral from saved identifier and name works if you have access to the InternalBleManager, so I added a property to BleManager to access the internal manager. If anyone needs it: 

In `ble_manager.dart`
```
InternalBleManager get internalManager;
```

In `internal_ble_manager.dart`
```
@override
  InternalBleManager get internalManager {
    return _bleLib._manager;
  }
```

To use: 
```
var json = {'id': 'saved identifier', 'name': 'saved name'};
Peripheral peripheral = Peripheral.fromJson(json, _manager.internalManager);
```

This is sort of contrary to how the library is set up, with the peripheral being an intermediary between the internal manager and the api, but it was quick and easy.

mikolak, 2020-07-24:
I'll try to get it working properly next week. :) It might require some work to make it compatible with BLEmulator as well.

mikolak, 2020-08-20:
Hi!
We're stretched thin and a bit overwhelmed with other work right now and need a little help.
Could one of you (@cameronkloot @cbreezier @tomstokes @raycardillo @jtrovato @slole @jpuerto @TheLastGimbus) help us and test the PR #502 ?

You'd have to switch dependency in your app to the git branch and see if you can connect to the peripheral using `BleManager.createUnsafePeripheral(id).connect()`.
```
dependencies:
  flutter_ble_lib:
    git:
      url: git://github.com/Polidea/FlutterBleLib.git
      ref: feature/#295-connect-without-scan
```

VinciShark, 2020-08-20:
I test this function with:

`
var peripheral = bleManager.createUnsafePeripheral({MY_ID_HERE});
await peripheral.connect();
bool connected = await peripheral.isConnected();
print("!!!!" + connected.toString());
`

Works fine!

@mikolak

mikolak, 2020-08-20:
@VinciShark great, thanks for help!

PR #502 (2020-08-20): Add unsafe way of creating peripherals
Resolves #295
closed 2020-08-20 by mikolak