dotintent/FlutterBleLib #337: BLE Client: not able to write data into characteristics

gurgenDP, 2019-12-04:
BLE Client is not able to write data into characteristics after receiving first ad-hoc notification. The scenario is the following: 
1. Mobile app(hereafter Client) connects to the BLE device.
1. The client enables notifications on the device (plugin enables notifications automatically).
1. Client receives ad-hoc notification
1. Client writes data to characteristics
1. Device reads the request data
1. Device writes the response data to characteristics

Scenario breaks when the Client tries to write the data to characteristics. The device doesn't receive any data from the Client. 

The problem occurs with a real BLE device and Mobile app running Android OS. Scenario above works with BLE emulator + Android, BLE emulator + iOS, Real Device + iOS.

Also some info about device: 
* Uses the Generic Attribute Profile (GATT).
* Based on one Service including only one characteristic.
* Characteristic supports the following operations
  * Read
  * Write
  * Notification

pawelByszewski, 2019-12-04:
@gurgenDP thanks for the issue!

mikolak, 2019-12-05:
Hi, @gurgenDP!

Could you set log level to verbose, retry your scenario and paste the logs?

gurgenDP, 2019-12-06:
Thanks for follow up @mikolak. I will be able to provide logs early next week.

gurgenDP, 2019-12-10:
Hello @mikolak . Providing you the logs.

I've went over the logs and separated the part that may be most useful.
```
                        [30;102m I [0m set log level to verbose
[34mlib.FlutterBleLibPlugin[0m [30;104m D [0m on native side observed method: setLogLevel
[35mlegate.LogLevelDelegate[0m [30;104m D [0m set log level to: verbose
[34mlib.FlutterBleLibPlugin[0m [30;104m D [0m on native side observed method: stopDeviceScan
                        [30;104m D [0m on native side observed method: startDeviceScan
[36mle#ClientOperationQueue[0m [30;104m D [0m QUEUED   ScanOperationApi21(167638979)
                        [30;104m D [0m STARTED  ScanOperationApi21(167638979)
[31m   RxBle#QueueOperation[0m [30;102m I [0m Scan operation is requested to start.
[33m       BluetoothAdapter[0m [30;104m D [0m STATE_ON
[32m     BluetoothLeScanner[0m [30;104m D [0m onClientRegistered() - status=0 clientIf=5
[36mle#ClientOperationQueue[0m [30;104m D [0m FINISHED ScanOperationApi21(167638979) in 63 ms
[32m                flutter[0m [30;102m I [0m ========tf37004c001251373239333039
                        [30;102m I [0m null
                        [30;102m I [0m device ===============
                        [30;102m I [0m [Peripheral{
                        [30;102m I [0m     name: tf37004c001251373239333039,
                        [30;102m I [0m     identifier: F8:F0:05:D3:AA:8D
                        [30;102m I [0m }]
[34mlib.FlutterBleLibPlugin[0m [30;104m D [0m on native side observed method: stopDeviceScan
[32m                flutter[0m [30;102m I [0m ========No Name
                        [30;102m I [0m null
[35mCancellableSubscription[0m [30;102m I [0m Scan operation is requested to stop.
[33m       BluetoothAdapter[0m [30;104m D [0m STATE_ON
[32m                flutter[0m [30;102m I [0m Set actibe device =============tf37004c001251373239333039
[34mlib.FlutterBleLibPlugin[0m [30;104m D [0m on native side observed method: isDeviceConnected
                        [30;104m D [0m on native side observed method: connectToDevice
[32m                flutter[0m [30;102m I [0m =============== PeripheralConnectionState.connecting
[36mle#ClientOperationQueue[0m [30;104m D [0m QUEUED   ConnectOperation(96922694)
                        [30;104m D [0m STARTED  ConnectOperation(96922694)
[31mBle#BleConnectionCompat[0m [37;100m V [0m Connecting without reflection
[35m          BluetoothGatt[0m [30;104m D [0m connect() - device: F8:F0:05:D3:AA:8D, auto: false
                        [30;104m D [0m registerApp()
                        [30;104m D [0m registerApp() - UUID=89a4d508-7e69-4b08-9706-a6657b16d6fe
                        [30;104m D [0m onClientRegistered() - status=0 clientIf=5
                        [30;104m D [0m onClientConnectionState() - status=0 clientIf=5 device=F8:F0:05:D3:AA:8D
[33m    RxBle#BluetoothGatt[0m [30;104m D [0m onConnectionStateChange newState=2 status=0
[36mle#ClientOperationQueue[0m [30;104m D [0m FINISHED ConnectOperation(96922694) in 333 ms
[34monnectionOperationQueue[0m [30;104m D [0m QUEUED   MtuRequestOperation(69802773)
                        [30;104m D [0m STARTED  MtuRequestOperation(69802773)
[35m          BluetoothGatt[0m [30;104m D [0m configureMTU() - device: F8:F0:05:D3:AA:8D mtu: 23
                        [30;104m D [0m onConfigureMTU() - Device=F8:F0:05:D3:AA:8D mtu=23 status=0
[33m    RxBle#BluetoothGatt[0m [30;104m D [0m onMtuChanged mtu=23 status=0
[32m                flutter[0m [30;102m I [0m =============== PeripheralConnectionState.connected
[34monnectionOperationQueue[0m [30;104m D [0m FINISHED MtuRequestOperation(69802773) in 654 ms
[34mlib.FlutterBleLibPlugin[0m [30;104m D [0m on native side observed method: discoverAllServicesAndCharacteristics
[34monnectionOperationQueue[0m [30;104m D [0m QUEUED   ServiceDiscoveryOperation(75983566)
                        [30;104m D [0m STARTED  ServiceDiscoveryOperation(75983566)
[35m          BluetoothGatt[0m [30;104m D [0m discoverServices() - device: F8:F0:05:D3:AA:8D
                        [30;104m D [0m onSearchComplete() = Device=F8:F0:05:D3:AA:8D Status=0
[33m    RxBle#BluetoothGatt[0m [30;104m D [0m onServicesDiscovered status=0
[31mrviceDiscoveryOperation[0m [37;100m V [0m Preparing services description
                        [37;100m V [0m --------------- ====== Printing peripheral content ====== ---------------
                        [37;100m V [0m PERIPHERAL ADDRESS: F8:F0:05:D3:AA:8D
                        [37;100m V [0m PERIPHERAL NAME: tf37004c001251373239333039
                        [37;100m V [0m -------------------------------------------------------------------------
                        [37;100m V [0m Primary Service - Generic Access (00001800-0000-1000-8000-00805f9b34fb)
                        [37;100m V [0m Instance ID: 0
                        [37;100m V [0m -> Characteristics:
                        [37;100m V [0m     * Device Name (00002a00-0000-1000-8000-00805f9b34fb)
                        [37;100m V [0m       Properties: [ READ ]
                        [37;100m V [0m     * Appearance (00002a01-0000-1000-8000-00805f9b34fb)
                        [37;100m V [0m       Properties: [ READ ]
[... cut ...]
[36mle#ClientOperationQueue[0m [30;104m D [0m FINISHED ConnectOperation(265361643) in 4 ms
[31mecutors$RunnableAdapter[0m [30;104m D [0m Terminated.
[36mle#ClientOperationQueue[0m [30;104m D [0m STARTED  DisconnectOperation(237394119)
[32m       BluetoothManager[0m [30;104m D [0m getConnectionState()
                        [30;104m D [0m getConnectedDevices
[34mlib.FlutterBleLibPlugin[0m [30;104m D [0m on native side observed method: connectToDevice                            
```

gurgenDP, 2019-12-10:
The complete log file is 4 mb, but the scenario and lines are similar. What I can see from logs is:
* The plugin first enables the notifications and disables them later.
* Sometimes it's throwing an error when trying to connect, event when right before connection I'm checking the device state to be connected.

gurgenDP, 2019-12-10:
[teforia-app-20191208-1.log](https://github.com/Polidea/FlutterBleLib/files/3944173/teforia-app-20191208-1.log)

mikolak, 2019-12-10:
```
[30;104m D [0m on native side observed method: monitorCharacteristicForIdentifier
[35m          BluetoothGatt[0m [30;104m D [0m setCharacteristicNotification() - uuid: 8ba16c63-1810-c6d2-1a3d-09eb265d0001
                             enable: true
[34monnectionOperationQueue[0m [30;104m D [0m QUEUED   DescriptorWriteOperation(1909846)
                        [30;104m D [0m STARTED  DescriptorWriteOperation(1909846)
[33m    RxBle#BluetoothGatt[0m [30;104m D [0m onDescriptorWrite descriptor=00002902-0000-1000-8000-00805f9b34fb status=0
                        [30;104m D [0m onCharacteristicChanged characteristic=8ba16c63-1810-c6d2-1a3d-09eb265d0001
[34monnectionOperationQueue[0m [30;104m D [0m FINISHED DescriptorWriteOperation(1909846) in 70 ms
[36m        RxBle#BleModule[0m [37;100m V [0m Notification from Characteristic(uuid: 8ba16c63-1810-c6d2-1a3d-09eb265d0001,
                             id: 9, value: 44110B060100C0A8446D00000000800000000000)
[32m                flutter[0m [30;102m I [0m ===============log=========== From.CentralNetwork Status2019-12-08 17:46:15.
                            919687
[34mlib.FlutterBleLibPlugin[0m [30;104m D [0m on native side observed method: writeCharacteristicForIdentifier
[34monnectionOperationQueue[0m [30;104m D [0m QUEUED   CharacteristicWriteOperation(47207905)
                        [30;104m D [0m STARTED  CharacteristicWriteOperation(47207905)
[33m    RxBle#BluetoothGatt[0m [30;104m D [0m onCharacteristicWrite characteristic=8ba16c63-1810-c6d2-1a3d-09eb265d0001 st
                            atus=0
[36m        RxBle#BleModule[0m [37;100m V [0m Write to Characteristic(uuid: 8ba16c63-1810-c6d2-1a3d-09eb265d0001, id: 9, v
                            alue: 281100C7)
[34monnectionOperationQueue[0m [30;104m D [0m FINISHED CharacteristicWriteOperation(47207905) in 18 ms
```

This snippet shows setting up notifications on a characteristic, receiving one and successfully writing to the same characteristic a new value.

I have a few follow-up observations after looking through logs:
* I think there might be two parallel reconnection monitors running and they get in the way of each other
* Error code: 2 means the operation has been cancelled. If you provide the same transactionId to two different operations (eg. monitoring and writing to characteristic), the latter will cancel the former. By default transactionId is null and it is a valid value, so two consequent operations without provided transactionId will cause the first one to be cancelled (which will only discard the result, btw). This behaviour should perhaps be changed, we'll discuss it internally.

Could you send us a sample reproducing the issue or some of you BLE-related code?

gurgenDP, 2019-12-10:
Thanks for the reply @mikolak. 
The points you mentioned can be helpful. I will try to give unique IDs to operations and see how it behaves. 

On iOS device and BLE emulator I had an issue when after the first write operation the characteristics data stream was not updated anymore. As a workaround, I cancel the transactions each time the write operation's Future gets completed. That solved the issue but may cause another problem of canceling all pending operations(including monitoring) because all have the same ID.

The code snippet for writing into characteristics
```dart
  /// Writes the value of an active characteristic.
  Future<void> characteristicWrite(List<int> value,
      {String transactionId}) async {
    await activeCharacteristic.write(Uint8List.fromList(value), false,
        transactionId: 'characteristicWrite');

    await _bleManager.cancelTransaction('characteristicWrite');

    logDataFromClient(Uint8List.fromList(value));
  }
```
Snippet for monitoring the characteristics value. Single Broadcast Stream accross the app.
```dart
  /// Listen to changes of this stream to be notified when
  ///  the active characteristic's value is changed.
  Stream<Uint8List> messageListener() {
    // Enable change notifications
    if (characteristicsData == null &&
        status == PeripheralConnectionState.connected) {
      characteristicsData = activeCharacteristic.monitor().asBroadcastStream();
    }

    // Return the stream from active device with active characteristic
    return characteristicsData;
  }
```

Just wondering why it's works on iOS and doesn't on Android. 
Thanks again, will try the recommendations above and keep updated here.

gurgenDP, 2019-12-18:
@mikolak . I'm still having the issue with writing into Characteristics. 

According to the logs, the write operation is finished, but the device doesn't get any data from the app. Exact same code works on iOS.

mikolak, 2019-12-18:
Hi, @gurgenDP! We haven't looked into details on your issue yet, sorry. It would help us tremendously if you could provide us with a minimalistic app that reproduces the issue to speed things up.

gurgenDP, 2019-12-18:
Thanks @mikolak . I've created a minimal app with the functions I use. 
To reproduce the issue:
- Connect to the device via `Connect` button.
- Listen for messages
- Write data

https://github.com/gurgenDP/ble_test

bartoszwilk, 2019-12-21:
Hi @gurgenDP. Thanks for reporting this issue.
I built your project, ran the app and was able to write and receive notifications on the same characteristic seamlessly.
Here are the logs from my test.
```ruby
2019-12-21 17:21:19.537 8790-8790/com.example.ble_test D/com.polidea.flutter_ble_lib.FlutterBleLibPlugin: on native side observed method: createClient
2019-12-21 17:21:19.597 8790-8790/com.example.ble_test D/com.polidea.flutter_ble_lib.FlutterBleLibPlugin: on native side observed method: getState
2019-12-21 17:21:19.625 8790-8790/com.example.ble_test D/com.polidea.flutter_ble_lib.FlutterBleLibPlugin: on native side observed method: getState
2019-12-21 17:21:34.045 8790-8861/com.example.ble_test I/flutter: set log level to verbose
2019-12-21 17:21:34.046 8790-8790/com.example.ble_test D/com.polidea.flutter_ble_lib.FlutterBleLibPlugin: on native side observed method: setLogLevel
2019-12-21 17:21:34.046 8790-8790/com.example.ble_test D/com.polidea.flutter_ble_lib.delegate.LogLevelDelegate: set log level to: verbose
2019-12-21 17:21:34.048 8790-8790/com.example.ble_test D/com.polidea.flutter_ble_lib.FlutterBleLibPlugin: on native side observed method: stopDeviceScan
2019-12-21 17:21:34.054 8790-8790/com.example.ble_test D/com.polidea.flutter_ble_lib.FlutterBleLibPlugin: on native side observed method: startDeviceScan
2019-12-21 17:21:34.071 8790-8790/com.example.ble_test D/RxBle#ClientOperationQueue: QUEUED   ScanOperationApi21(218887017)
2019-12-21 17:21:34.072 8790-8879/com.example.ble_test D/RxBle#ClientOperationQueue: STARTED  ScanOperationApi21(218887017)
2019-12-21 17:21:34.077 8790-8885/com.example.ble_test I/RxBle#QueueOperation: Scan operation is requested to start.
2019-12-21 17:21:34.079 8790-8885/com.example.ble_test D/BluetoothAdapter: isLeEnabled(): ON
2019-12-21 17:21:34.083 8790-8835/com.example.ble_test D/BluetoothLeScanner: onScannerRegistered() - status=0 scannerId=5 mScannerId=0
2019-12-21 17:21:34.089 8790-8879/com.example.ble_test D/RxBle#ClientOperationQueue: FINISHED ScanOperationApi21(218887017) in 17 ms
2019-12-21 17:21:35.096 8790-8861/com.example.ble_test I/flutter: ========CC2650 SensorTag
2019-12-21 17:21:35.101 8790-8861/com.example.ble_test I/flutter: [0000aa80-0000-1000-8000-00805f9b34fb]
2019-12-21 17:21:35.111 8790-8790/com.example.ble_test D/com.polidea.flutter_ble_lib.FlutterBleLibPlugin: on native side observed method: stopDeviceScan
2019-12-21 17:21:35.112 8790-8861/com.example.ble_test I/flutter: ========CC2650 SensorTag
2019-12-21 17:21:35.112 8790-8861/com.example.ble_test I/flutter: [0000aa80-0000-1000-8000-00805f9b34fb]
2019-12-21 17:21:35.114 8790-8885/com.example.ble_test I/RxBle#CancellableSubscription: Scan operation is requested to stop.
2019-12-21 17:21:35.117 8790-8885/com.example.ble_test D/BluetoothAdapter: isLeEnabled(): ON
2019-12-21 17:21:35.126 8790-8861/com.example.ble_test I/flutter: Set active device =============CC2650 SensorTag
2019-12-21 17:21:35.136 8790-8790/com.example.ble_test D/com.polidea.flutter_ble_lib.FlutterBleLibPlugin: on native side observed method: isDeviceConnected
2019-12-21 17:21:35.141 8790-8790/com.example.ble_test D/com.polidea.flutter_ble_lib.FlutterBleLibPlugin: on native side observed method: connectToDevice
2019-12-21 17:21:35.142 8790-8861/com.example.ble_test I/flutter: PeripheralConnectionState.disconnected
2019-12-21 17:21:35.161 8790-8790/com.example.ble_test W/xample.ble_tes: Accessing hidden field Lsun/misc/Unsafe;->theUnsafe:Lsun/misc/Unsafe; (light greylist, reflection)
2019-12-21 17:21:35.178 8790-8892/com.example.ble_test D/RxBle#ClientOperationQueue: QUEUED   ConnectOperation(107248145)
2019-12-21 17:21:35.180 8790-8879/com.example.ble_test D/RxBle#ClientOperationQueue: STARTED  ConnectOperation(107248145)
2019-12-21 17:21:35.184 8790-8885/com.example.ble_test V/RxBle#BleConnectionCompat: Connecting without reflection
2019-12-21 17:21:35.192 8790-8885/com.example.ble_test D/BluetoothGatt: connect() - device: 54:6C:0E:53:41:49, auto: false
2019-12-21 17:21:35.192 8790-8885/com.example.ble_test D/BluetoothGatt: registerApp()
2019-12-21 17:21:35.192 8790-8885/com.example.ble_test D/BluetoothGatt: registerApp() - UUID=f292ece2-56a6-4bc9-847f-121a63895a32
2019-12-21 17:21:35.198 8790-8835/com.example.ble_test D/BluetoothGatt: onClientRegistered() - status=0 clientIf=5
2019-12-21 17:21:35.693 8790-8835/com.example.ble_test D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=5 device=54:6C:0E:53:41:49
2019-12-21 17:21:35.696 8790-8835/com.example.ble_test D/RxBle#BluetoothGatt$1: onConnectionStateChange newState=2 status=0
2019-12-21 17:21:35.704 8790-8879/com.example.ble_test D/RxBle#ClientOperationQueue: FINISHED ConnectOperation(107248145) in 523 ms
2019-12-21 17:21:35.716 8790-8892/com.example.ble_test D/RxBle#ConnectionOperationQueue: QUEUED   MtuRequestOperation(245733756)
2019-12-21 17:21:35.719 8790-8891/com.example.ble_test D/RxBle#ConnectionOperationQueue: STARTED  MtuRequestOperation(245733756)
2019-12-21 17:21:35.728 8790-8885/com.example.ble_test D/BluetoothGatt: configureMTU() - device: 54:6C:0E:53:41:49 mtu: 23
2019-12-21 17:21:36.134 8790-8835/com.example.ble_test D/BluetoothGatt: onConnectionUpdated() - Device=54:6C:0E:53:41:49 interval=6 latency=0 timeout=500 status=0
2019-12-21 17:21:37.376 8790-8835/com.example.ble_test D/BluetoothGatt: onConfigureMTU() - Device=54:6C:0E:53:41:49 mtu=23 status=0
[... cut ...] h it.
Logcat will provide us more information about what is actually happening on your phone during the test. If you will be using Android Studio or VSCode plugin for that, it is important to disable all logcat filters (especially the one responsible for printing only logs for application that is currently running) and set logcat log level to verbose.
4. (if answer on 1st question is no) Could you try this test scenario on another android device (with different OS version for instance)?

Thanks!

gurgenDP, 2019-12-21:
Thanks for testing and reply @bartoszwilk . 
1. The scenario above was tested on 3 different android devices with 3 units of device.
2. I will check the versions of 2 other devices, but the one I've tested on was Android 9.
3. I can run the app and send you logs from logcat tomorrow or on Monday.
4. I should be able to test on Android 5, 8, 9, 10 I think. Will send you the logs from different versions.

Thanks again, will update here once I get chance to test and the logs.

gurgenDP, 2019-12-22:
@bartoszwilk Please see attached the logs from Android 9 device. 
[logs.txt](https://github.com/Polidea/FlutterBleLib/files/3992331/logs.txt)

bartoszwilk, 2020-01-09:
Hi @gurgenDP,

I've looked into your logs and obviously you are able to write (or at least init writing) a value to your characteristic.

I've noticed this line in your logs.
```ruby
12-22 13:44:56.205 21532 21558 D RxBle#BluetoothGatt$1: onCharacteristicWrite characteristic=8ba16c63-1810-c6d2-1a3d-09eb265d0001 status=0
```

_BluetoothGattCallback#onCharacteristicWrite_ method is called each time when peripheral, which we are connected to, sends us a confirmation that writing operation is finished (with some status).
In our case we write to characteristic in "no response" mode, but status=0 says that this operation was at least invoked at system level and no error occurred during this operation 
(we are quite sure about that because underlying RxAndroidBle library checks the result of _BluetoothGatt#writeCharacteristic_ and throws BleGattCannotStartException when its false, so wasn't initiated successfully).

I was wondering if your problem could be related to the pairing/bonding process that was running during the test. It seems that your peripheral automatically (or maybe you handle it by your own?) triggers this process after establishing a connection with your phone. 

Note that you set the subscription for characteristic changes before the pairing/bonding process ends and you try to write to that characteristic after this process ends.

Could you please make another test where you:
1. Connect to your peripheral
2. Wait for pairing/bonding process to be finished
3. Wait couple of seconds
4. Discover all services
5. Subscribe for characteristic changes
6. Write to characteristic

I will also ask you to paste logs from this test too, if possible.

If it won't work I will have to ask you for sending us HCI logs so we could look into them and verify what is happening in the radio layer between the phone and the peripheral during writing operation.

Thanks! :)

mikolak, 2020-02-10:
Hi, @gurgenDP! 

Does the issue still occur for you?

gurgenDP, 2020-02-10:
Hi @mikolak @bartoszstelmaszuk . 

Unfortunately, I had to switch the project back to flutter_blue plugin because of the following issue. 

You can close the issue until someone reproduces it or faces with similar on. 

Thanks and keep going on!

mikolak, 2020-02-11:
Sorry to hear that! I take it the issue does not occur on that plugin?

gurgenDP, 2020-02-11:
It has it's own problems, but the issue with writing data is not exist.

mikolak, 2020-02-12:
Closing for now, since we couldn't reproduce it.

closed 2020-02-12 by mikolak

gurgenDP, 2020-04-18:
I've decided to have a try one more time and find out the issue. 

```
await activeCharacteristic.write(Uint8List.fromList(value), false,
        transactionId: 'characteristicWrite');
```

`withResponse ` parameter was set to false, and it seems like the device is not supporting write without response. Setting withResponse to true, fixed the issue.

mikolak, 2020-04-21:
Welcome back! Thanks for the update! :)

yash10019coder, 2024-05-23:
> @mikolak . I'm still having the issue with writing into Characteristics.
> 
> According to the logs, the write operation is finished, but the device doesn't get any data from the app. Exact same code works on iOS.

this fixed for me now actually you also need to send the response to every request that request is received on the server or client. 

As you do this it'll work smoothly