dotintent/FlutterBleLib #347: how to get the device notify value

spd-heshuip, 2019-12-11:
I connected to the device ,and  i need to get the message from the device which it notify ,
what should i do?

mikolak, 2019-12-11:
Hi, @spd-heshuip!

You want to get a notification from a characteristic, correct? 

After connecting to peripheral you can use `peripheral.monitorCharacteristic()` or filter `peripheral.characteristics()` result to get the characteristic you're interested in and call `characteristic.monitor()`.

Take a look at the example. https://github.com/Polidea/FlutterBleLib/blob/master/example/lib/test_scenarios/peripheral_test_operations.dart#L207

spd-heshuip, 2019-12-12:
> Hi, @spd-heshuip!
>
> You want to get a notification from a characteristic, correct?
>
> After connecting to peripheral you can use `peripheral.monitorCharacteristic()` or filter [...]

Hi  @mikolak ,
there is my code 

[image]

when i  connected to the devices,i use `peripheral.monitorCharacteristic()`,but it failed;
this is the error message:
`D/com.polidea.flutter_ble_lib.FlutterBleLibPlugin(27613): on native side observed method: createClient
D/com.polidea.flutter_ble_lib.FlutterBleLibPlugin(27613): on native side observed method: startDeviceScan
D/BluetoothAdapter(27613): STATE_ON
D/BluetoothLeScanner(27613): onClientRegistered() - status=0 clientIf=5
I/flutter (27613): Scanned Peripheral Zephyr, RSSI -61
I/flutter (27613): find device
D/com.polidea.flutter_ble_lib.FlutterBleLibPlugin(27613): on native side observed method: stopDeviceScan
D/BluetoothAdapter(27613): STATE_ON
I/flutter (27613): onDone
I/flutter (27613): start connect....
D/com.polidea.flutter_ble_lib.FlutterBleLibPlugin(27613): on native side observed method: connectToDevice
D/BluetoothGatt(27613): connect() - device: 54:59:EA:CC:50:7F, auto: false
D/BluetoothGatt(27613): registerApp()
D/BluetoothGatt(27613): registerApp() - UUID=3143e654-db03-4efc-b501-559cab8586e2
D/BluetoothGatt(27613): onClientRegistered() - status=0 clientIf=5
D/BluetoothGatt(27613): onClientConnectionState() - status=0 clientIf=5 device=54:59:EA:CC:50:7F
D/BluetoothGatt(27613): configureMTU() - device: 54:59:EA:CC:50:7F mtu: 23
D/BluetoothGatt(27613): onConfigureMTU() - Device=54:59:EA:CC:50:7F mtu=23 status=0
I/flutter (27613): Peripheral 54:59:EA:CC:50:7F connection state is PeripheralConnectionState.connected
D/com.polidea.flutter_ble_lib.FlutterBleLibPlugin(27613): on native side observed method: isDeviceConnected
I/flutter (27613): connect device successs
D/com.polidea.flutter_ble_lib.FlutterBleLibPlugin(27613): on native side observed method: discoverAllServicesAndCharacteristics
D/BluetoothGatt(27613): discoverServices() - device: 54:59:EA:CC:50:7F
D/BluetoothGatt(27613): onSearchComplete() = Device=54:59:EA:CC:50:7F Status=0
D/com.polidea.flutter_ble_lib.FlutterBleLibPlugin(27613): on native side observed method: monitorCharacteristicForDevice
D/BluetoothGatt(27613): setCharacteristicNotification() - uuid: 12345678-1234-5678-1234-56789abcdef1 enable: true
D/BluetoothGatt(27613): setCharacteristicNotification() - uuid: 12345678-1234-5678-1234-56789abcdef1 enable: false
I/flutter (27613): monitor value:Error while monitoring characteristic 
I/flutter (27613): BleError (Error code: 403, ATT error code: null, iOS error code: null, Android error code: null, reason: Cannot write client characteristic config descriptor (code 3) with characteristic UUID 12345678-1234-5678-1234-56789abcdef1, internal message: null, device ID: null, service UUID: 12345678-1234-5678-1234-56789abcdef0, characteristic UUID: 12345678-1234-5678-1234-56789abcdef1, descriptor UUID: null)`

if i use `peripheral.monitorCharacteristic()`,it always report the error ,and then i can not  write the devices, this is the error message:
`FlutterBleLibPlugin(27613): on native side observed method: writeCharacteristicForDevice
I/flutter (27613): error:
I/flutter (27613): BleError (Error code: 4, ATT error code: null, iOS error code: null, Android error code: null, reason: GATT exception from MAC address 54:59:EA:CC:50:7F, with type BleGattOperation{description='CHARACTERISTIC_WRITE'}, internal message: null, device ID: null, service UUID: null, characteristic UUID: null, descriptor UUID: null)
`
please help me ,thank you very much!!

mikolak, 2019-12-12:
```
D/BluetoothGatt(27613): setCharacteristicNotification() - uuid: 12345678-1234-5678-1234-56789abcdef1 enable: false I/flutter (27613): monitor value:Error while monitoring characteristic I/flutter (27613): **BleError (Error code: 403**, ATT error code: null, iOS error code: null, Android error code: null, reason: **Cannot write client characteristic config descriptor (code 3) with characteristic UUID 12345678-1234-5678-1234-56789abcdef1**, internal message: null, device ID: null, service UUID: 12345678-1234-5678-1234-56789abcdef0, characteristic UUID: 12345678-1234-5678-1234-56789abcdef1, descriptor UUID: null)
```
403 is `CharacteristicNotifyChangeFailed`.

It seems the characteristic you're attempting to monitor does not support it. Could you try the following?
```
  List<Characteristic> characteristics =
      await peripheral.characteristics("12345678-1234-5678-1234-56789abcdef0");
  Characteristic characteristic = characteristics.firstWhere(
      (e) => e.uuid == "12345678-1234-5678-1234-56789abcdef1",
      orElse: () => null);

  if (characteristic == null) {
    print("Characteristic not found");
  } else {
    print("Characteristic ${characteristic.uuid} can be monitored: "
        "${characteristic.isNotifiable || characteristic.isIndicatable}");
  }
```

How do `initiObtaining(Peripheral)` and `monitorCharacteristicForPeripheral(Peripheral)` look?

spd-heshuip, 2019-12-12:
initiObtaining(Peripheral) and monitorCharacteristicForPeripheral(Peripheral):
[image]

and i try your code:
[image]

spd-heshuip, 2019-12-12:
initiObtaining(Peripheral) and monitorCharacteristicForPeripheral(Peripheral):

`
  initObtaining(Peripheral peripheral)async{

    await peripheral.discoverAllServicesAndCharacteristics();
  }

  StreamSubscription monitoringStreamSubscription;
  Future<void> monitorCharacteristicForPeripheral(Peripheral peripheral) async {
    await _runWithErrorHandling(() async {
      _startMonitoringTemperature(
          peripheral
              .monitorCharacteristic(
              serviceUUID,
              charUUID,
              transactionId: "monitor")
              .map((characteristic) => characteristic.value),
          (value){
            print("monitor value:" + value);
          });
    });
  }`

and i try your code,the log is:
`D/com.polidea.flutter_ble_lib.FlutterBleLibPlugin( 9187): on native side observed method: characteristics
I/flutter ( 9187): Characteristic 12345678-1234-5678-1234-56789abcdef1 can be monitored: true`

mikolak, 2019-12-12:
1) Can you try running:
```
List<Characteristic> characteristics =
      await peripheral.characteristics("12345678-1234-5678-1234-56789abcdef0");
  Characteristic characteristic = characteristics.firstWhere(
      (e) => e.uuid == "12345678-1234-5678-1234-56789abcdef1",
      orElse: () => null);

  if (characteristic == null) {
    print("Characteristic not found");
  } else {
    print(
      "Characteristic:\n"
      "uuid: ${characteristic.uuid}\n"
      "isReadable: ${characteristic.isReadable}\n"
      "isWritableWithoutResponse: ${characteristic.isWritableWithoutResponse}\n"
      "isWritableWithResponse: ${characteristic.isWritableWithResponse}\n"
      "isIndicatable: ${characteristic.isIndicatable}\n"
      "isNotifiable: ${characteristic.isNotifiable}\n",
    );
```
and attaching the print?

2) Can you set log level to LogLevel.verbose, run your scenario and attach the logs?

BTW for long snippets use triple back ticks, instead of single:
>```

spd-heshuip, 2019-12-13:
it is work for now! the reason is that device notification data is too long,now we change the data's length,it work fine;

mikolak, 2019-12-13:
Cool, happy to hear that. Hope you have a good time with the library!

closed 2019-12-13 by mikolak