dotintent/FlutterBleLib #356: support indicate

spd-heshuip, 2019-12-17:
how  to receive the device indicate value?

pawelByszewski, 2019-12-17:
You can check if  what mechanism is supported bu a characteristic directly on an instance 
 of `Characteristic` https://github.com/Polidea/FlutterBleLib/blob/589ed5770d9796e390c73945d3b520d12176ef16/lib/characteristic.dart#L22

If you want to observe a value then just call `monitor` method in the same way as if the characteristic would be notifiable. 
On vanilla Android BLE API it is possible to choose whether notification or indication is setup if a characteristic supports both but the iOS does not allow to choose—that is why it is not possible to do so in this library.

spd-heshuip, 2019-12-17:
OK，I got it ，thank you very much！

closed 2019-12-17 by pawelByszewski