dotintent/FlutterBleLib #358: about indicate

spd-heshuip, 2019-12-20:
i use this library to receive the packet by other bluetooth device indicate ,but sometimes it will lost some packet;

mikolak, 2019-12-20:
Hi!

Sorry, but you'll need to be more clear. Is it a random indication that gets lost or right at the start of subscription? How do you know that it gets lost, are you sniffing communication? Does it happen on both platforms?

spd-heshuip, 2019-12-21:
it is get lost right at the start of subscription.i did not  sniffing communication.and i know that it gets lost because i had check the packet which  send by the bluetooth devices; the bluetooth devices show has indicate success on the log,  but the phone lost one of the packet; for the moment  i  just  test it on the android ;

mikolak, 2020-01-24:
Version 2.1.0 with fix has just been released.

PR #378 (2020-01-17): Delay start of characteristic monitoring until sink is created
Fixes #358
closed 2020-01-24 by mikolak