dotintent/FlutterBleLib #361: Can't get the mtu value from the connection.

NOA-Charles, 2019-12-23:
I can't find in the source code to get the mtu value from the connection.
The requestMtu() will get the invalid PDU error.
Any idea?

@mikolak

_Originally posted by @NOA-Charles in https://github.com/Polidea/FlutterBleLib/issues/315#issuecomment-567773838_

NOA-Charles, 2019-12-23:
###### Cause:
It is found through the firmware's log that when the APP connects, the APP requests mtu.
##### Isuee:
The Android APP has successfully obtained the MTU value when it was successfully connected.I don't know where to get it.

pawelByszewski, 2019-12-23:
Could you set log level to `VERBOSE` and provide logs? The valid MTU value range is from 23 to 512, could you try to request some value from that range instead of `517`?

NOA-Charles, 2019-12-23:
> Could you set log level to `VERBOSE` and provide logs? The valid MTU value range is from 23 to 512, could you try to request some value from that range instead of `517`?
Thank you for your reply.
No matter what value of mtu is used on Android, the following error will be reported.
##### log:
```
[ERROR:flutter/lib/ui/ui_dart_state.cc(157)] Unhandled Exception: BleError (Error code: 206, ATT error code: 4, iOS error code: null, Android error code: null, reason: GATT exception from MAC address EC:F7:19:2C:71:DA, status 4 (GATT_INVALID_PDU), type BleGattOperation{description='ON_MTU_CHANGED'}. (Look up status 0x04 here https://android.googlesource.com/platform/external/bluetooth/bluedroid/+/android-5.1.0_r1/stack/include/gatt_api.h), internal message: null, device ID: EC:F7:19:2C:71:DA, service UUID: null, characteristic UUID: null, descriptor UUID: null)
```

mikolak, 2020-01-07:
Hi, @NOA-Charles!

Could you paste the fragment responsible for connecting?

NOA-Charles, 2020-01-08:
Happy New Year! @mikolak 

This is the code for my connection part:
```
      await peripheral.connect(requestMtu: 517).catchError((_, __) {
        print("connect err: $_");
      });
      bool connected = await peripheral.isConnected();
      if (!connected) {
        cb(false);
        return;
      }
      await peripheral.discoverAllServicesAndCharacteristics();
      //TODO: Wait get real mtu value
//      BLEEntity.mtu = await peripheral.requestMtu(512);
//      print("MTU: ${BLEEntity.mtu}");
      cb(true);
```

mikolak, 2020-01-08:
This is undocumented behaviour, which I'm sorry for. I'll add information about this to readme. 

MTU can only be requested once, so if you request it in `peripheral.connect(requestMtu: int)`, the call to `periphera.requestMtu()` will fail.

Additionally, there's a bug in the library with MTU set to a default value, while it should be omitted.

I'll fix this in the library, but in the meantime you can check this workaround:
```
await peripheral.connect(requestMtu: null);
int negoiatedMtu = await peripheral.requestMtu(512);
```

mikolak, 2020-01-08:
I'm sorry, but the afore mentioned workaround won't work after all. Working on a fix now.

NOA-Charles, 2020-01-09:
Very grateful for your help.

In version 2.0.1, we need to use parameters in the connection function:```requestMtu: 0```

##### The following code is available
```
await peripheral.connect(requestMtu: 0);
await peripheral.discoverAllServicesAndCharacteristics();
int mtu = await peripheral.requestMtu(512);
```

closed 2020-01-09 by NOA-Charles

mikolak, 2020-01-24:
Version 2.1.0 with fix has just been released.

PR #371 (2020-01-10): Remove setting MTU to 23 in connect by default
Fixes #361