dotintent/FlutterBleLib #362: Questions about notifications

huaxiao47, 2019-12-23:
When there is a large amount of data notification, the data will be duplicated and overwritten. I wonder if the platform thread and ui thread are thread safe? There is no such problem when using the rxble library natively on Android.

huaxiao47, 2019-12-23:
I tested the rx_ble flutter library and it was fine.

mikolak, 2019-12-23:
Hi!

Could you paste the logs? The issue occurs only on Android?

huaxiao47, 2019-12-23:
[image]
this is correct

[image]
The error looks like this

mikolak, 2020-01-09:
Sorry for the long silence. 

Is it possible for you to check if the issue occurs on iOS as well? 

Is there only one active subscription monitoring any characteristic, ie. do you monitor only one characteristic with only one subscription?

mikolak, 2020-01-29:
Hi @huaxiao47 
Does the problem still occur on new version? Could you set log level to verbose and paste the logs?

posipunk, 2020-02-27:
I'm having same/similar issue on 2.2.1, only tested on Android. I have only 1 monitored characteristic and 1 subscription. 
I'm writing to a characteristic and expect 3 writes back to another characteristic (UART service). 
I observe 3 writes to the characteristic in sequence (lets call them A, B, C) but the data my app gets is a jumble of these and randomly returns duplicates [(B,B,B), (B,C,C), (A,B,C), (A,C,C), (C,C,C), etc.]. They seem to at least always be in order (i.e. A is always before B and C)

mikolak, 2020-02-27:
Hi!
@posipunk could you put
`await Future.delayed(Duration(millis: 500))` between the writes and let me know if it changed anything?

posipunk, 2020-02-28:
> Hi!
> @posipunk could you put
> `await Future.delayed(Duration(millis: 500))` between the writes and let me know if it changed anything?

@mikolak  I'm only doing a single write from the Phone side.
I'm using the Nordic UART service and everything works great on native Android with Nordic's libraries.

Sequence:
1. Phone writes "REQ sample" 
2. Device writes "ACK" (A) in response
3. Device takes sample (~50-100ms)
4. Device writes 2 responses (B, C). 

My thinking is that by the time the Flutter level gets the first characteristic write the second has already been received / queued up and perhaps they're pointing to the same object and the data gets overwritten with this new one.

jornhong, 2020-03-03:
`D/com.polidea.flutter_ble_lib.FlutterBleLibPlugin(10025): on native side observed method: monitorCharacteristicForIdentifier
D/BluetoothGatt(10025): setCharacteristicNotification() - uuid: 0000fff1-0000-1000-8000-00805f9b34fb enable: true
D/BluetoothGatt(10025): onConnectionUpdated() - Device=30:45:11:47:E4:81 interval=36 latency=0 timeout=500 status=0
I/flutter (10025): frame: 0xc0 0x25 0x2 0xa0 0x20 0x30 0xd3 0xc0 
D/com.polidea.flutter_ble_lib.FlutterBleLibPlugin(10025): on native side observed method: writeCharacteristicForIdentifier
I/flutter (10025): in home bluetooth connected
I/flutter (10025): begin dump##############################################
I/flutter (10025): recv data:0xc0 0x25 0x24 0x0 0x0 0x18 0x50 0x4 0x0 0x4c 0x4d 0x54 0x41 0x30 0x30 0x30 0x30 0x20 0x4c 
I/flutter (10025): end dump##############################################
I/flutter (10025): begin dump##############################################
I/flutter (10025): recv data:0x55 0x8e 0xcc 0xc0 
I/flutter (10025): end dump##############################################
I/flutter (10025): begin dump##############################################
I/flutter (10025): recv data:0x55 0x8e 0xcc 0xc0 
I/flutter (10025): end dump##############################################
D/BluetoothGatt(10025): onConnectionUpdated() - Device=30:45:11:47:E4:81 interval=9 latency=0 timeout=300 status=0
D/com.polidea.flutter_ble_lib.FlutterBleLibPlugin(10025): on native side observed method: cancelConnection
D/BluetoothManager(10025): getConnectionState()
D/BluetoothManager(10025): getConnectedDevices
D/BluetoothGatt(10025): cancelOpen() - device: 30:45:11:47:E4:81
D/BluetoothGatt(10025): onClientConnectionState() - status=0 clientIf=7 device=30:45:11:47:E4:81
D/BluetoothGatt(10025): setCharacteristicNotification() - uuid: 0000fff1-0000-1000-8000-00805f9b34fb enable: false`

same problem here. I just write a single command to characteristic, my device should return 42 bytes which begin with 0xc0 and end with 0xc0. but as you see, it just notify 19 bytes and followed with  the last 4 bytes twice.

mikolak, 2020-03-04:
Hi, @jornhong! How often do the notifications occur on your device? Is the length of the notification always trimmed this way? Does it occur on any other characteristic?

jornhong, 2020-03-04:
thanks for your reply.
yes, notification always trimmed. 
I put my code into the example of the FlutterBleLib, it work in order, so i don't know what's the problem.

[image]

jornhong, 2020-03-04:
i modify  monitorCharacteristic() function in the example for test. this problem still occur occasionally.
[image]

mikolak, 2020-03-05:
Hey, @jornhong @posipunk @huaxiao47, did you guys set MTU? 

IIRC Android by default does not negotiate and gets 23 bytes. That would be the 19+4 bytes @posipunk was getting out of 42 intended.

The auto test in the example wasn't doing such a thing, since our peripheral was sending only 4 bytes of data.

Try adding
```dart
int MTU = await peripheral.requestMtu(512)
```
Before operations on characteristics and see what MTU you negotiate.

jornhong, 2020-03-05:
I had try this way before,but change nothing.
I also try the flutter_blue  library, it occur same problem.
flutter_wechat_ble have same problem with the first few commands, and the rest are always in order.
my origin app is programmed by QT/QML, it working fine.

mikolak, 2020-03-06:
@jornhong @posipunk Could you guys set log level to verbose, run your scenarios printing data you receive to logs and paste them here?

posipunk, 2020-03-06:
@mikolak Setting Flutter to verbose didn't add much.
I did enable the Bluetooth HCI logs to get the raw data though it is encrypted you can tell the relative byte lengths of the 3 notifications and see that the library isn't returning the correct data

Relevant flutter verbose log
[   +2 ms] I/flutter ( 2663): 2020-03-06T11:14:36.043483	D	new DeviceDetailsBloc.<ac>:	 Written "[2, 12, 11, 176, 56, 160, 221, 8, 43, 18, 2, 8, 4, 0]" to uart tx config
[        ] D/com.polidea.flutter_ble_lib.FlutterBleLibPlugin( 2663): on native side observed method: writeCharacteristicForDevice
[ +108 ms] I/flutter ( 2663): 2020-03-06T11:14:36.154324	D	new DeviceDetailsBloc.<ac>:	 RX: [2, 12, 10, 59, 14, 221, 180, 8, 1, 16, 43, 26, 1, 0]
[  +48 ms] I/flutter ( 2663): 2020-03-06T11:14:36.202680	D	new DeviceDetailsBloc.<ac>:	 RX: [2, 17, 16, 180, 208, 106, 202, 8, 3, 42, 7, 24, 4, 40, 60, 56, 152, 30, 0]
[   +5 ms] I/flutter ( 2663): 2020-03-06T11:14:36.207290	D	new DeviceDetailsBloc.<ac>:	 RX: [2, 17, 16, 180, 208, 106, 202, 8, 3, 42, 7, 24, 4, 40, 60, 56, 152, 30, 0]

Relevant lines from the Bluetooth HCI log (note - encrypted)
"No.","Time","Source","Destination","Protocol","Length","Info"
"1676","1293.930951","SamsungE_d3:5d:XX (Galaxy Note8)","ea:94:0e:e0:ce:88 (DEVICE)","ATT","26","Sent Write Request, Handle: 0x0018 (Unknown)"
024000150011000400121800020c0bb038a0dd082b1202080400

"1677","1294.029780","ea:94:0e:e0:ce:XX (DEVICE)","SamsungE_d3:5d:XX (Galaxy Note8)","ATT","10","Rcvd Write Response, Handle: 0x0018 (Unknown)"
02402005000100040013

"1678","1294.030796","ea:94:0e:e0:ce:XX (DEVICE)","SamsungE_d3:5d:XX (Galaxy Note8)","ATT","26","Rcvd Handle Value Notification, Handle: 0x001a (Unknown)"
0240201500110004001b1a00020c0a3b0eddb40801102b1a0100

"1679","1294.079221","ea:94:0e:e0:ce:XX (DEVICE)","SamsungE_d3:5d:XX (Galaxy Note8)","ATT","54","Rcvd Handle Value Notification, Handle: 0x001a (Unknown)"
02402031002d0004001b1a0002282776772ad20803221e0d43787e481518e430421d68f4313925dc7479482dc978304235d319353900

"1680","1294.080227","ea:94:0e:e0:ce:XX (DEVICE)","SamsungE_d3:5d:XX (Galaxy Note8)","ATT","31","Rcvd Handle Value Notification, Handle: 0x001a (Unknown)"
0240201a00160004001b1a00021110b4d06aca08032a071804283c38981e00

jornhong, 2020-03-06:
finally I had fix this problem by modify the com.polidea.multiplatformbleadapter package.
in BleModule.java's safeMonitorCharacteristicForDevice method, the onNex(byte[] bytes) method override characteristic's value before flutter listener to read the value:

`     @Override
                    public void onNext(byte[] bytes) {
                        characteristic.logValue("Notification from", bytes);
                        characteristic.setValue(bytes);  
                        onEventCallback.onEvent(characteristic);
              }`

i had changed like this:
  
                        Characteristic newChracteristic = characteristic.clone();
                        newChracteristic.setValue(bytes);
                        newChracteristic.logValue("Notification from", bytes);
                        //characteristic.setValue(bytes);
                        onEventCallback.onEvent(newChracteristic);

I alsa add a clone method to the Class Characteristic and modify the Characteristic's constructor to save the service:

        final private Service service;

    public Characteristic clone(){
        return new Characteristic(this.id,this.service,this.gattCharacteristic);
    }

I'm not a Java programmer, I'm not sure this is the best way.

 Thanks for your work.
Best regards.

mikolak, 2020-03-07:
@posipunk that should have added native logs, which would tell whether issue was in native part or the bridge between Dart and native.

@jornhong great job, thank you very much! I'll release a fix Tuesday tops!

mikolak, 2020-03-10:
2.2.3 with fix published. 

@posipunk @jornhong let me know (and reopen) if it hasn't fixed your issues, please.

closed 2020-03-10 by mikolak

posipunk, 2020-03-10:
Can confirm release 2.2.3 fixed all of the issues I was experiencing. Thank you for your excellent support!