dotintent/FlutterBleLib #363: When scanning for devices on Android, no devices are found

earyzhe, 2019-12-27:
When trying to find peripherals on Android, no scan results are found.

Version   flutter_ble_lib: 2.0.1

Everything is working fine with iOS.

The code I use to scan is:

```dart
if (!_isScanning) {
        print("Starting to scan for peripherals");
        _isScanning = true;
        _scanSubscription =
            _bleManager.startPeripheralScan().listen((ScanResult scanResult) {
          //Scan one peripheral and stop scanning
          print(
              "Scanned Peripheral ${scanResult.peripheral.name}, RSSI ${scanResult.rssi}");
          if (scanResult != null) {
            filterDevices(scanResult);
          }
        });
      }
```

**Output from flutter doctor -v**

[✓] Flutter (Channel dev, v1.13.5, on Mac OS X 10.15.2 19C57, locale de-GB)
    • Flutter version 1.13.5 at /Users/earyzhe/SDKs/flutter
    • Framework revision 41a911099b (8 days ago), 2019-12-19 13:48:02 -0800
    • Engine revision 0f90e6546b
    • Dart version 2.8.0 (build 2.8.0-dev.0.0 aa6709974d)

[✓] Android toolchain - develop for Android devices (Android SDK version 28.0.3)
    • Android SDK at /Users/earyzhe/Library/Android/sdk
    • Android NDK location not configured (optional; useful for native profiling support)
    • Platform android-28, build-tools 28.0.3
    • Java binary at: /Applications/Android Studio.app/Contents/jre/jdk/Contents/Home/bin/java
    • Java version OpenJDK Runtime Environment (build 1.8.0_202-release-1483-b49-5587405)
    • All Android licenses accepted.

[✓] Xcode - develop for iOS and macOS (Xcode 11.3)
    • Xcode at /Applications/Xcode.app/Contents/Developer
    • Xcode 11.3, Build version 11C29
    • CocoaPods version 1.8.3

[✓] Android Studio (version 3.5)
    • Android Studio at /Applications/Android Studio.app/Contents
    • Flutter plugin version 40.1.2
    • Dart plugin version 191.8423
    • Java version OpenJDK Runtime Environment (build 1.8.0_202-release-1483-b49-5587405)

[✓] IntelliJ IDEA Community Edition (version 2018.3.4)
    • IntelliJ at /Applications/IntelliJ IDEA CE.app
    • Flutter plugin version 33.2.2
    • Dart plugin version 183.5901

[✓] VS Code (version 1.41.1)
    • VS Code at /Applications/Visual Studio Code.app/Contents
    • Flutter extension version 3.7.1

[✓] Connected device (2 available)
    • Andrew's iPhone • 6aa0c9719a85c174869a09656237b42f23701a91 • ios        • iOS 13.1.3
    • macOS           • macOS

pawelByszewski, 2019-12-30:
Hey @earyzhe, at first please ensure that your app on android has granted access to Bluetooth and Location services (and both of them are turned on). Then could you set the log level to verbose and then provide logs?

earyzhe, 2019-12-30:
Hey @pawelByszewski . 

I turned on the location services and that fixed it. 

Thanks for that. I have been stuck with it for ages.

Maybe worth a mention in the documentation.

pawelByszewski, 2019-12-31:
Good point @earyzhe.

closed 2019-12-31 by pawelByszewski