dotintent/FlutterBleLib #364: Writing is not permitted

spd-heshuip, 2020-01-02:
i write the message to the Characteristic on the IOS phone ,but it is fail ;( on the android phone ,it is success),here is the log:
[image]

mikolak, 2020-01-07:
Hi, @spd-heshuip!

Have you passed `false` to `withResponse` in write function call?

spd-heshuip, 2020-01-08:
Hi @mikolak , i always passed false to withResponse in write function call. And i use the lightBlue to test ,it can write success ,but when i use this library  it fail ;

mikolak, 2020-01-08:
Is the peripheral bonded or paired? Could you make a minimal example, set the log level in the library to verbose and paste logs?

spd-heshuip, 2020-02-21:
no bonded or paired ,i use RxBluetoothKit to test without response,it work, but use the flutterBleLib's example it fail;
[image]

JesusAlexlara, 2020-02-21:
i have de same problem i can't write messages in Characteristic with IOS, did you find any solution? 

`[VERBOSE-2:ui_dart_state.cc(157)] Unhandled Exception: BleError (Error code: 401, ATT error code: 3, iOS error code: null, Android error code: null, reason: Writing is not permitted., internal message: null, device ID: D96CF31D-A797-4C1E-8DB2-C5C64D928339, service UUID: 68a8f0fe-3686-485c-b149-41666e35d80d, characteristic UUID: dac83ec3-ad95-41a2-b901-ba30135774d1, descriptor UUID: null)`

spd-heshuip, 2020-02-22:
@JesusAlexlara Hi,I did not find the solution,i think it is library's bug;

mikolak, 2020-02-22:
Hi! Would you mind telling me what kind of peripheral are you using or even a specific model? I could try getting it and testing on my end. I don't have much time at the moment, but I'll try to look into it.

spd-heshuip, 2020-02-23:
Hi，i use the this project's flutter example to test write without response,j just change the  service uuid and Characteristic uuid,here is the code:
[image]

mikolak, 2020-02-24:
Is it possible for you to check this on Android, @JesusAlexlara @spd-heshuip ?

spd-heshuip, 2020-02-24:
@mikolak  I had check this on Android ,it is work;  
 on IOS ,it is not;

JesusAlexlara, 2020-02-25:
I am using an esp32 with custom characteristics, I was doing several tests and I think I found the error.

When the characteristics has the property of "write without response" and set flag false in write, produces the error

- Property write without response.
- characteristic.write(Uint8List.fromList([0]), false); 
this produces the error on iOS, on android it works.

but if I change the property only to write and set the flag to true it works.

- Property write
- characteristic.write(Uint8List.fromList([0]), true); 
this works on ios and android.

mikolak, 2020-02-27:
Thanks for info! I'll try to look into it.

spd-heshuip, 2020-03-03:
@mikolak  Hi,  is there any update for this bug?

mikolak, 2020-03-04:
We're a little low on time for now, but we're looking into it.

E-Sh4rk, 2020-03-23:
Hi @mikolak ,
I am also facing this issue and I made the same observations as JesusAlexlara.
Write without response fails on iOS. It works if I write using flutter_blue.

Device: https://www.thecubicle.com/products/giiker-super-cube-i3s
Service: 0000aaaa-0000-1000-8000-00805f9b34fb
Characteristic: 0000aaac-0000-1000-8000-00805f9b34fb

matt39, 2020-03-28:
Same issue here.

TomoLV, 2020-03-31:
Thank you for your patience. We've identified the issue, the fix is on its way.

PR #451 (2020-03-31): [iOS] Fixed casting of Bool arguments received from dart
Fixes #364 
In iOS native part of the library, bool arguments received from Dart part were cast in an inappropriate way.

As a result, `withResponse` flag of write characteristic operation was always received with `true` value on iOS native part, regardless of what was requested by the Dart part, which in turn resulted in an error.
closed 2020-03-31 by mikolak