dotintent/FlutterBleLib #365: Unable to start service because of FlutterBleLib

lettierimarco, 2020-01-02:
You used       
 ```
final FlutterBleLibPlugin plugin = new FlutterBleLibPlugin(registrar.activity().getApplicationContext());
 ```
so when i start service from a broadcast (in my case geolocation via geofencing) i do:
 ```
sBackgroundFlutterView = FlutterNativeView(context, true)
val registry = sBackgroundFlutterView!!.pluginRegistry
sPluginRegistrantCallback.registerWith(registry)
```
but this cause a NullPointerException because activity is null

there 's a bettery way to create the plugin? or a better way to load plugins from service?

mikolak, 2020-01-08:
Hi @lettierimarco !

I've changed the way we're obtaining context. I don't have a ready-to-use example for your use case, so could you test it for me?

Here's how to add a dependency to a git branch
https://flutter.dev/docs/development/packages-and-plugins/using-packages#dependencies-on-unpublished-packages
https://dart.dev/tools/pub/dependencies#git-packages

And this is the branch: `fix/365-unable-to-start-from-service`

lettierimarco, 2020-01-08:
still null
`     Caused by: java.lang.NullPointerException: Attempt to invoke virtual method 'android.content.Context android.app.Activity.getApplicationContext()' on a null object reference
`

lettierimarco, 2020-01-08:
Thanks, it works, why don't do a release?

closed 2020-01-08 by lettierimarco

mikolak, 2020-01-08:
Wanted to make sure it works first. ;) We should release new version this or next week.

mikolak, 2020-01-24:
Version 2.1.0 with fix has just been released.

PR #370 (2020-01-09): Drop unneeded get of activity
Fixes #365