dotintent/FlutterBleLib #366: Manufacturer Data is not correct

raycardillo, 2020-01-03:
I'm debugging with BlueSee and cross referencing when I port the implementation to a Flutter app using this library. I need a way to identify devices from certain manufacturers (not the specific device but a signature of the type of device so I know which model it is, etc). So I was going to use manufacturer data, but it's not working quite right.

- First, it's an `Int8List` and I would have expected a `Uint8List` so we can get the raw bytes and parse them.
- More concerning is that I just tested with 4 different devices, and for each, I'm getting the same Manufacturer Data (see below) across 4 difference devices when using this library but BlueSee is showing me very different data... so I think there is a bug here.

What I'm getting on all of them is this (even though BlueSee shows different info):
`[-103, -87, -18, 125, -89, 45, -70, -73, -85, 13, -85, 90]`

I'm testing on a physical device (`iPhone 11 Pro Max` running `iOS 13.3`).

mikolak, 2020-01-08:
Hi @raycardillo! Thanks for the issue!

Could you see if the same error occurs on Android? Could you paste the code from obtaining Manufacturer Data to the end of parsing it?

mikolak, 2020-01-09:
I've identified the issue, fix will be out shortly.

raycardillo, 2020-01-10:
Sorry for the delayed response but I was busy with work. Glad to hear that you figured out the root cause. I'll be eager to test the fix when it's available (as soon as I get spare cycles).

mikolak, 2020-01-24:
Version 2.1.0 with fix has just been released.

PR #375 (2020-01-15): Fix incorrect Manufacturer's Data
Fixes #366
closed 2020-01-24 by mikolak