dotintent/FlutterBleLib #367: [iOS] Crash in observeBluetoothState method

rmorbach, 2020-01-07:
Using the following code will result in a crash on iOS as soon as the Bluetooth Adapter status changes:

```dart
BleManager bleManager =  BleManager();
await bleManager.createClient();
bleManager.observeBluetoothState().listen((newState) {
    //...
});
```

This will throw the following exception in Xcode: 
```ObjC
-[__NSCFString objectAtIndexedSubscript:]: unrecognized selector sent to instance
```
Using debug breakpoints in Xcode, it seems the error is thrown in the line 25 of the the method: `- (void)onNewAdapterState:(NSArray *)bluetoothAdapterState` of `AdapterStateStreamHandler.m` file.

[image]

It looks like you are passing a `NSString` instance while the method expects a `NSArray`. This method is called in the `FlutterBleLib.m` file.

[image]

The specific method is:
[image]

mikolak, 2020-01-08:
Hi @rmorbach !

Thanks for the ticket and the investigation! I'll try to fix it and release new patch version this week.

mikolak, 2020-01-24:
Version 2.1.0 with fix has just been released.

PR #373 (2020-01-09): Fix crash in observing Bluetooth adapter state on iOS
Fixes #367
closed 2020-01-24 by mikolak