dotintent/FlutterBleLib #380: scanResult.isConnectable allways returns null

EmilHW, 2020-01-23:
I just started using FlutterBleLib and ran into the problem that any instance of ScanResult allways returns null when i call isConnectable on the object. This is how i do the scanning and test the value:

`
final Stream<ScanResult> stream = bleManager.startPeripheralScan();
    stream.listen(
      (scanResult) {
        final DeviceInfo deviceInfo = DeviceInfo.fromScanResult(scanResult);
        print("${scanResult.peripheral.name} : ${scanResult.isConnectable}");
        updateMap(deviceInfo);
      },
    );
`

Am i doing something wrong here, or am i misunderstanding the use of ScanResult?

mikolak, 2020-01-23:
Hi!

Thanks for the info, I'll look into it tomorrow. On what platform does the problem occur?

EmilHW, 2020-01-24:
Android, and i am running it on my OnePlus 6.

mikolak, 2020-01-24:
As it turns out, the `isConnectable` flag, as well as `overflowServiceUuids`, is iOS only. This information should have been included in README, sorry for the omission.

Android supports `isConnectable()` since API 26, but the dependency we're using does not and because of that it'll remain unsupported for the time being.

PR #381 (2020-01-27): Clarify that two of ScanResult fields are iOS specific
Resolves #380
closed 2020-01-27 by mikolak