dotintent/FlutterBleLib #396: Support for AndroidX.

mikolak, 2020-02-07:


KevinRohn, 2020-02-11:
Is AndroidX already supported?

mikolak, 2020-02-13:
@KevinRohn Jetifier should handle it for now. We haven't yet swapped Support for AndroidX in the library.

KevinRohn, 2020-02-13:
@mikolak What does this exactly mean? I've built the example project and run it on my Huawei Mate 20 Pro with AndroidX - There is the same issue like on all other current Flutter BLE packages, that the scan result always seems to be empty. For iOS and < AndroidX it works fine

mikolak, 2020-02-14:
I'm not sure if your issue connected to AndroidX, especially since we only use Not Null annotations in the library code IIRC. Are you granting location permission to the example? Can you set log level to verbose and check the logs after scanning for peripherals?

PR #411 (2020-02-14): Migrate from Support Library to AndroidX
Resolves #396
closed 2020-02-14 by mikolak