dotintent/FlutterBleLib #412: pod install error when use in flutter module

YoshiApp, 2020-02-14:
I try to build this lib in a flutter module in iOS and got the following error

```
Running pod install...
CocoaPods' output:
↳
      Preparing

    Analyzing dependencies

    Inspecting targets to integrate
      Using `ARCHS` setting to build architectures of target `Pods-Runner`: (``)

    Fetching external sources
    -> Fetching podspec for `Flutter` from `Flutter/engine`
    -> Fetching podspec for `FlutterPluginRegistrant` from `Flutter/FlutterPluginRegistrant`
    -> Fetching podspec for `flutter_ble_lib` from `Flutter/.symlinks/flutter_ble_lib/ios`

    Resolving dependencies of `Podfile`
      CDN: trunk Relative path: CocoaPods-version.yml exists! Returning local because checking is only perfomed in repo update
      CDN: trunk Relative path: all_pods_versions_2_a_e.txt exists! Returning local because checking is only perfomed in repo update
      CDN: trunk Relative path: Specs/2/a/e/MultiplatformBleAdapter/0.1.3/MultiplatformBleAdapter.podspec.json exists! Returning local because checking is only perfomed in repo update
      CDN: trunk Relative path: Specs/2/a/e/MultiplatformBleAdapter/0.0.4/MultiplatformBleAdapter.podspec.json exists! Returning local because checking is only perfomed in repo update

    Comparing resolved specification to the sandbox manifest
      A Flutter
      A FlutterPluginRegistrant
      A MultiplatformBleAdapter
      A flutter_ble_lib

    Downloading dependencies

    -> Installing Flutter (1.0.0)

    -> Installing FlutterPluginRegistrant (0.0.1)

    -> Installing MultiplatformBleAdapter (0.0.4)
      > Copying MultiplatformBleAdapter from `/Users/yoshi/Library/Caches/CocoaPods/Pods/Release/MultiplatformBleAdapter/0.0.4-1beec` to `Pods/MultiplatformBleAdapter`

    -> Installing flutter_ble_lib (0.0.1)
      - Running pre install hooks
    [!] Unable to determine Swift version for the following pods:

    - `flutter_ble_lib` does not specify a Swift version and none of the targets (`Runner`) integrating it have the `SWIFT_VERSION` attribute set. Please contact the author or set the `SWIFT_VERSION` attribute in at least one of the targets that integrate this pod.

    /usr/local/Cellar/cocoapods/1.8.3/libexec/gems/cocoapods-1.8.3/lib/cocoapods/installer/xcode/target_validator.rb:125:in `verify_swift_pods_swift_version'
    /usr/local/Cellar/cocoapods/1.8.3/libexec/gems/cocoapods-1.8.3/lib/cocoapods/installer/xcode/target_validator.rb:39:in `validate!'
    /usr/local/Cellar/cocoapods/1.8.3/libexec/gems/cocoapods-1.8.3/lib/cocoapods/installer.rb:590:in `validate_targets'
    /usr/local/Cellar/cocoapods/1.8.3/libexec/gems/cocoapods-1.8.3/lib/cocoapods/installer.rb:158:in `install!'
    /usr/local/Cellar/cocoapods/1.8.3/libexec/gems/cocoapods-1.8.3/lib/cocoapods/command/install.rb:52:in `run'
    /usr/local/Cellar/cocoapods/1.8.3/libexec/gems/claide-1.0.3/lib/claide/command.rb:334:in `run'
    /usr/local/Cellar/cocoapods/1.8.3/libexec/gems/cocoapods-1.8.3/lib/cocoapods/command.rb:52:in `run'
    /usr/local/Cellar/cocoapods/1.8.3/libexec/gems/cocoapods-1.8.3/bin/pod:55:in `<top (required)>'
    /usr/local/Cellar/cocoapods/1.8.3/libexec/bin/pod:23:in `load'
    /usr/local/Cellar/cocoapods/1.8.3/libexec/bin/pod:23:in `<main>'

Error output from CocoaPods:
↳
    /System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/ruby/2.6.0/universal-darwin19/rbconfig.rb:229: warning: Insecure world writable dir /Volumes/External/AndroidSDK in PATH, mode 040777

Finished with error: Error running pod install
```

mikolak, 2020-02-14:
What Swift version are you using in your application?

YoshiApp, 2020-02-15:
Swift 5
I solved this by adding
s.swift_versions = ['4.0', '4.2', '5.0']
to flutter_ble_lib.podspec

mikolak, 2020-02-15:
Cool, thanks for info! I'll patch it on my end.

PR #413 (2020-02-17): Enable Swift 5
Resolve #412 

According to linked issue, this is sufficient to make the library work with Swift 5.
closed 2020-02-17 by mikolak