dotintent/FlutterBleLib #424: Don't work in Android 4.4.4

edsonboldrini, 2020-02-20:
The lib scan for nearby peripherals and never finds any. I've already tested with two different devices with this system version and the result was the same. Please, help me!

edsonboldrini, 2020-02-20:
Here is the debug console:
[image]
Apparently, the device finds a peripheral, but throw a exception right after.

mikolak, 2020-02-24:
Do you have `ACCESS_FINE_LOCATION` permission and location services on?

edsonboldrini, 2020-02-24:
Unfortunately yes :(

edsonboldrini, 2020-02-25:
I build for release and then execute the example project using a device with android 4.4.4 and nothing happens.

[image]

edsonboldrini, 2020-03-01:
Please, don't forget me!!

mikolak, 2020-03-01:
Don't worry, I'll investigate this. Got a device set aside on my desk, but I didn't have time last week and I probably won't this week as well.

edsonboldrini, 2020-03-03:
Here is another log using the example project from inside the lib to improve your investigation:

✓ Built build/app/outputs/apk/debug/app-debug.apk.
I/flutter ( 5947): 2020-03-03T09:19:02.768341	D	DeviceListScreenState.didChangeDependencies:	 DeviceListScreenState didChangeDependencies
I/flutter ( 5947): 2020-03-03T09:19:02.826080	D	DeviceListScreenState._onResume:	 onResume
I/flutter ( 5947): 2020-03-03T09:19:02.864868	D	DevicesBloc.init:	 Init devices bloc
D/com.polidea.flutter_ble_lib.FlutterBleLibPlugin( 5947): on native side observed method: createClient
I/flutter ( 5947): 2020-03-03T09:19:02.978027	D	DevicesBloc.init:	  listen to _devicePickerController.stream
W/dalvikvm( 5947): VFY: unable to find class referenced in signature (Landroid/bluetooth/le/ScanSettings;)
W/dalvikvm( 5947): VFY: unable to find class referenced in signature (Landroid/bluetooth/le/ScanCallback;)
I/dalvikvm( 5947): Could not find method android.bluetooth.BluetoothAdapter.getBluetoothLeScanner, referenced from method com.polidea.rxandroidble.internal.util.RxBleAdapterWrapper.startLeScan
W/dalvikvm( 5947): VFY: unable to resolve virtual method 334: Landroid/bluetooth/BluetoothAdapter;.getBluetoothLeScanner ()Landroid/bluetooth/le/BluetoothLeScanner;
D/dalvikvm( 5947): VFY: replacing opcode 0x6e at 0x0002
W/dalvikvm( 5947): VFY: unable to find class referenced in signature (Landroid/bluetooth/le/ScanCallback;)
I/dalvikvm( 5947): Could not find method android.bluetooth.BluetoothAdapter.getBluetoothLeScanner, referenced from method com.polidea.rxandroidble.internal.util.RxBleAdapterWrapper.stopLeScan
W/dalvikvm( 5947): VFY: unable to resolve virtual method 334: Landroid/bluetooth/BluetoothAdapter;.getBluetoothLeScanner ()Landroid/bluetooth/le/BluetoothLeScanner;
D/dalvikvm( 5947): VFY: replacing opcode 0x6e at 0x0002
W/dalvikvm( 5947): VFY: unable to find class referenced in signature (Landroid/bluetooth/le/ScanResult;)
I/dalvikvm( 5947): Could not find method android.bluetooth.le.ScanResult.getScanRecord, referenced from method com.polidea.rxandroidble.internal.scan.InternalScanResultCreator.create
W/dalvikvm( 5947): VFY: unable to resolve virtual method 414: Landroid/bluetooth/le/ScanResult;.getScanRecord ()Landroid/bluetooth/le/ScanRecord;
D/dalvikvm( 5947): VFY: replacing opcode 0x6e at 0x0002
W/dalvikvm( 5947): VFY: unable to find class referenced in signature (Landroid/bluetooth/le/ScanResult;)
I/dalvikvm( 5947): Could not find method android.bluetooth.le.ScanResult.getScanRecord, referenced from method com.polidea.rxandroidble.internal.scan.InternalScanResultCreator.create
W/dalvikvm( 5947): VFY: unable to resolve virtual method 414: Landroid/bluetooth/le/ScanResult;.getScanRecord ()Landroid/bluetooth/le/ScanRecord;
D/dalvikvm( 5947): VFY: replacing opcode 0x6e at 0x0002
I/flutter ( 5947): 2020-03-03T09:19:03.086608	D	DeviceListScreenState.build:	 build DeviceListScreenState
I/dalvikvm( 5947): Could not find method android.content.Context.createDeviceProtectedStorageContext, referenced from method androidx.core.content.ContextCompat.createDeviceProtectedStorageContext
W/dalvikvm( 5947): VFY: unable to resolve virtual method 455: Landroid/content/Context;.createDeviceProtectedStorageContext ()Landroid/content/Context;
D/dalvikvm( 5947): VFY: replacing opcode 0x6e at 0x0006
I/dalvikvm( 5947): Could not find method android.content.Context.getCodeCacheDir, referenced from method androidx.core.content.ContextCompat.getCodeCacheDir
W/dalvikvm( 5947): VFY: unable to resolve virtual method 462: Landroid/content/Context;.getCodeCacheDir ()Ljava/io/File;
D/dalvikvm( 5947): VFY: replacing opcode 0x6e at 0x0006
I/dalvikvm( 5947): Could not find method android.content.Context.getColor, referenced from method androidx.core.content.ContextCompat.getColor
W/dalvikvm( 5947): VFY: unable to resolve virtual method 463: Landroid/content/Context;.getColor (I)I
D/dalvikvm( 5947): VFY: replacing opcode 0x6e at 0x0006
I/dalvikvm( 5947): Could not find method android.content.Context.getColorStateList, referenced from method androidx.core.content.ContextCompat.getColorStateList
W/dalvikvm( 5947): VFY: unable to resolve virtual method 464: Landroid/content/Context;.getColorStateList (I)Landroid/content/res/ColorStateList;
D/dalvikvm( 5947): VFY: replacing opcode 0x6e at 0x0006
I/dalvikvm( 5947): Could not find method android.content.Context.getDataDir, referenced from method androidx.core.content.ContextCompat.getDataDir
W/dalvikvm( 5947): VFY: unable to resolve virtual method 466: Landroid/content/Context;.getDataDir ()Ljava/io/File;
D/dalvikvm( 5947): VFY: replacing opcode 0x6e at 0x0006
I/dalvikvm( 5947): Could not find method android.content.Context.getDrawable, referenced from method androidx.core.content.ContextCompat.getDrawable
W/dalvikvm( 5947): VFY: unable to resolve virtual method 468: Landroid/content/Context;.getDrawable (I)Landroid/graphics/drawable/Drawable;
D/dalvikvm( 5947): VFY: replacing opcode 0x6e at 0x0006
I/dalvikvm( 5947): Could not find method android.content.Context.getMainExecutor, referenced from method androidx.core.content.ContextCompat.getMainExecutor
W/dalvikvm( 5947): VFY: unable to resolve virtual method 475: Landroid/content/Context;.getMainExecutor ()Ljava/util/concurrent/Executor;
D/dalvikvm( 5947): VFY: replacing opcode 0x6e at 0x0006
I/dalvikvm( 5947): Could not find method android.content.Context.getNoBackupFilesDir, referenced from method androidx.core.content.ContextCompat.getNoBackupFilesDir
[... cut ...]
    <uses-permission android:name="android.permission.INTERNET" />
    
    
    <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
    <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
    <uses-permission android:name="android.permission.ACCESS_BACKGROUND_LOCATION" />

    
    <application
        android:name="io.flutter.app.FlutterApplication"
        android:label="flutter_ble_lib_example"
        android:icon="@mipmap/ic_launcher">
        <activity

mikolak, 2020-03-04:
Ah, it seems you're missing two permissions in the manifest:
```
<uses-permission android:name="android.permission.BLUETOOTH" />
<uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
```

edsonboldrini, 2020-03-04:
Still the same error:
```
D/permissions_handler(12029): No permissions found in manifest for: 2
I/flutter (12029): Location permission status: PermissionStatus.unknown
I/flutter (12029): Location service status: ServiceStatus.notApplicable
```
Manifest:
```
<uses-permission android:name="android.permission.INTERNET" />

<uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
<uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
<uses-permission android:name="android.permission.ACCESS_BACKGROUND_LOCATION" />
<uses-permission android:name="android.permission.ACCESS_LOCATION_EXTRA_COMMANDS" />

<uses-permission android:name="android.permission.BLUETOOTH" />
<uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
<uses-permission android:name="android.permission.BLUETOOTH_PRIVILEGED" />
```

mikolak, 2020-03-04:
Android 4.4 doesn't support runtime permissions, which was not taken into consideration when making the example.

Can you try commenting out [line 61 in example/lib/devices_list/devices_bloc.dart](https://github.com/Polidea/FlutterBleLib/blob/develop/example/lib/devices_list/devices_bloc.dart#L61)?

edsonboldrini, 2020-03-04:
> Android 4.4 doesn't support runtime permissions, which was not taken into consideration when making the example.
> 
> Can you try commenting out [line 61 in example/lib/devices_list/devices_bloc.dart](https://github.com/Polidea/FlutterBleLib/blob/develop/example/lib/devices_list/devices_bloc.dart#L61)?

Already done, same error.

```
I/flutter (12487): 2020-03-04T08:28:11.928247	D	DeviceListScreenState.build:	 build DeviceListScreenState
D/com.polidea.flutter_ble_lib.FlutterBleLibPlugin(12487): on native side observed method: getState
I/flutter (12487): 2020-03-04T08:28:14.318689	D	DevicesBloc._startScan:	 Ble client created
D/com.polidea.flutter_ble_lib.FlutterBleLibPlugin(12487): on native side observed method: startDeviceScan
D/BluetoothAdapter(12487): startLeScan(): null
D/BluetoothAdapter(12487): onClientRegistered() - status=0 clientIf=5
D/BluetoothAdapter(12487): onScanResult() - Device=C5:24:35:92:22:AD RSSI=-81
D/BluetoothAdapter(12487): stopLeScan()
W/BluetoothAdapter(12487): Unhandled exception: rx.exceptions.OnErrorFailedException: Error occurred when trying to propagate error to Observer.onError
```

mikolak, 2020-03-19:
Hi! I hadn't had the time yet, sorry. I forgot to ask: what's the device you're testing on?

edsonboldrini, 2020-03-23:
A Samsung grand duos prime and a Samsung tablet (I don't know its model exactly, maybe an old galaxy tab).

mikolak, 2020-03-26:
Hi!

Good news: finally found time to confirm it.

Bad news: the issue seems to lie in the native dependency RxAndroidBLE. I haven't yet pinpointed the issue there, since it seems like it should work out of the box, and yet it doesn't. I'll try to find the exact issue and get it patched for us, but I'm unable to give you any timeline.

edsonboldrini, 2020-03-30:
Ok!
Thanks for the help!