dotintent/FlutterBleLib #428: [Question] get Paired Devices

BrunoEleodoro, 2020-02-21:
Hello you'all, firstly let me say that you guys did an awesome job creating this library, the example folder is helpful as well.

But I'm not sure where I can find how to get the Bonded devices (Paired ones), and how to connect to this devices?

Thanks a lot.

dannyalbuquerque, 2020-03-04:
This library uses RxAndroidBle and RxBluetoothKit. This librarires offers this feature but unfortunately I did find it in FlutterBleLib... Don't sure why.

For RxBluetoothKit: https://polidea.github.io/RxBluetoothKit/Classes/CentralManager.html#/s:14RxBluetoothKit14CentralManagerC28retrieveConnectedPeripherals12withServicesSayAA10PeripheralCGSaySo6CBUUIDCG_tF

For RxAndroidBle:
https://github.com/Polidea/RxAndroidBle/pull/46

For native iOS: https://developer.apple.com/documentation/corebluetooth/cbcentralmanager/1518924-retrieveconnectedperipherals

For native Android: https://developer.android.com/reference/android/bluetooth/BluetoothAdapter#getBondedDevices()

mikolak, 2020-03-04:
The iOS method is exposed as `BleManager().connectedPeripherals()`.
I think it may not be possible to get them on Android at the moment, but it looks doable. I'll give you an update once I get to it.

BrunoEleodoro, 2020-03-04:
Thank you.

@mikolak @dannyalbuquerque

Peetz0r, 2020-03-16:
Ah, this explains why `BleManager().connectedPeripherals()` was giving me empty lists when I know it shouldn't on Android.

flutter_blue does it like this: https://github.com/pauldemarco/flutter_blue/blob/2ccdc68d474c158ae7cfbcae993e430a578173ec/android/src/main/java/com/pauldemarco/flutter_blue/FlutterBluePlugin.java#L173

Ahmadre, 2020-07-12:
What about this:

https://gist.github.com/joinAero/64e68c9d63987284f7f0#file-bluetoothprofileproxy-java-L234

Ahmadre, 2020-07-12:
I wrote a method in Java to handle fetching boundedDevices:

```java
    @RequiresPermission(Manifest.permission.BLUETOOTH)
    private void getPairedDevices(@NonNull final MethodChannel.Result result) {
        Log.d(TAG, "Get paired devices");

        bleAdapter.getKnownDevices(deviceIdentifiers.toArray(new String[deviceIdentifiers.size()]) ,new OnSuccessCallback<Device[]>() {	        final SafeMainThreadResolver resolver = new SafeMainThreadResolver<>(new OnSuccessCallback<Device[]>() {
            @Override	            @Override
            public void onSuccess(Device[] devices) {	            public void onSuccess(Device[] devices) {
                Log.d(TAG, "Found known devices" + devices.length);	                try {
                resolver.onSuccess(devices);	                    result.success(devicesResultJsonConverter.toJson(devices));
                } catch (JSONException e) {
                    e.printStackTrace();
                    result.error(null, e.getMessage(), null);
                }
            }	            }
        }, new OnErrorCallback() {	        }, new OnErrorCallback() {
            @Override	            @Override
            public void onError(BleError error) {	            public void onError(BleError error) {
                resolver.onError(error);	                Log.e(TAG, "Get paired devices error " + error.reason + "  " + error.internalMessage);
                result.error(String.valueOf(error.errorCode.code), error.reason, bleErrorJsonConverter.toJson(error));
            }	            }
        });	        });

        Set<BluetoothDevice> pairedDevices = mBluetoothAdapter.getBondedDevices();
        List<Device> devices = new ArrayList<>();

        for (int i = 0; i < pairedDevices.size(); i++) {
            BluetoothDevice blDevice =  pairedDevices.iterator().next();
            Device device = new Device(blDevice.getAddress(), blDevice.getName());
            devices.add(device);
        }

        Log.d(TAG, "Found paired devices" + devices.size());
        resolver.onSuccess(devices);
    }	    
}
```

But I am getting an error from my forked Branch:

```bash
MissingPluginException(Noimplementation found for method getPairedDevices on channel flutter_ble_lib)
```

mikolak, 2020-07-12:
@Ahmadre 
You need to handle the method call in the plugin class: https://github.com/Polidea/FlutterBleLib/blob/develop/android/src/main/java/com/polidea/flutter_ble_lib/FlutterBleLibPlugin.java#L120 
(same class name for iOS)

This issue has to be fixed in MultiPlatformBleAdapter, but we didn't have the time to take care of it yet.

ChrisMarxDev, 2020-12-02:
Are there any updates on this? I'm currently in the situation that I need to get a device that could potentially already be paired with my android phone.

mikolak, 2020-12-11:
Sorry, no updates yet. I finally have some time for the libraries, so work on FlutterBleLib and BLEmulator has started, but I'm unable to give you any timeline for this issue right now.

ousanjp, 2021-01-18:
Same problem, hope update to can getting Paired Devices, thanks.

nchsoft, 2021-02-05:
Also need this functionality

tohami, 2023-04-09:
any updated regarding this