dotintent/FlutterBleLib #429: Is there a way to avoid the location permission check before starting a scan?

espositofulvio, 2020-02-23:
I know FlutterBlue is based on RxAndroidBLE, I was wondering if in any way this option is suppted?
I've read from this pull request https://github.com/Polidea/RxAndroidBle/pull/533 that is a supported use case for RxAndroidBLE, but does it bubble up to FlutterBleLib as well?
I'mm writing an app where I could do without that check and my phone has no problem with other apps when location services are disabled.

mikolak, 2020-02-24:
I'm not clear on what is it that you'd want.
If I understand correctly, you'd like some kind of error telling you that though you started a scan, you won't get any results since you don't have a permission?

espositofulvio, 2020-02-24:
Hi @mikolak.
Nope, it's not that. I know Android requires ACCESS_FINE_LOCATION from 10 onwards and ACCESS_COARSE_LOCATION before that to perform a bluetooth scan and also requires location services to be enabled. There are quite a lot of phones that allow a bluetooth scan with location services switched off if the permission has been granted. RxAndroidBle defaults to check for permission and *also* for location services to be on, but an option has been added so that the second check be avoided (see the pull request I linked in my previous message). At the moment I'm getting an error when starting a scan with location services disabled but everything works if they're on. I couldn't find an option in FlutterBleLib to enable the other behaviour and was asking if I'm just missing it or it's not exposed.
I hope my explanation is a bit clearer now.

mikolak, 2020-02-24:
Ah, thanks for clarifying! I'll see what I can do.

VinciShark, 2020-08-20:
using this api would be able to find and connect to devices without needing location permission:
[companion-device-pairing document](https://developer.android.com/guide/topics/connectivity/companion-device-pairing)
[more detial in Youtube](https://www.youtube.com/watch?v=F1mvm_bfNgU)

@espositofulvio @mikolak