dotintent/FlutterBleLib #430: MissingPluginException

slole, 2020-02-23:
I am getting "MissingPluginException (MissingPluginException(No implementation found for method observeConnectionState on channel flutter_ble_lib))"
on line: 
```
connectedDevice.observeConnectionState(emitCurrentValue: true, completeOnDisconnect: true)
    .listen((connectionState) {
      print("Peripheral ${connectedDevice.identifier} connection state is $connectionState");
      stateCallback(connectionState == PeripheralConnectionState.connected);
    });
```

after calling connectedDevice.disconnectOrCancelConnection();

mikolak, 2020-02-24:
Could you paste logs set to verbose?

slole, 2020-02-24:
This?:
```
MethodChannel.invokeMethod
(c:\Users\Jan\development\flutter\packages\flutter\lib\src\services\platform_channel.dart:319)
<asynchronous gap> (Unknown Source:0)
DeviceConnectionMixin.observePeripheralConnectionState (c:\Users\Jan\development\flutter\.pub-cache\hosted\pub.dartlang.org\flutter_ble_lib-2.2.1\lib\internal\bridge\device_connection_mixin.dart:46)
<asynchronous gap> (Unknown Source:0)
InternalBleManager.observePeripheralConnectionState (c:\Users\Jan\development\flutter\.pub-cache\hosted\pub.dartlang.org\flutter_ble_lib-2.2.1\lib\internal\internal_ble_manager.dart:94)
Peripheral.observeConnectionState (c:\Users\Jan\development\flutter\.pub-cache\hosted\pub.dartlang.org\flutter_ble_lib-2.2.1\lib\peripheral.dart:33)
Bluetooth.connect (c:\Users\Jan\development\bokashi_organko\lib\bluetooth.dart:33)
_HomePageState._deviceFound.<anonymous closure> (c:\Users\Jan\development\bokashi_organko\lib\pages\home.page.dart:82)
_Closure.call (dart:core-patch/function.dart:0)
InkWell.onTap (c:\Users\Jan\development\flutter\packages\flutter\lib\src\material\ink_well.dart:0)
_InkResponseState._handleTap (c:\Users\Jan\development\flutter\packages\flutter\lib\src\material\ink_well.dart:706)
_InkResponseState.build.<anonymous closure> (c:\Users\Jan\development\flutter\packages\flutter\lib\src\material\ink_well.dart:789)
GestureRecognizer.invokeCallback (c:\Users\Jan\development\flutter\packages\flutter\lib\src\gestures\recognizer.dart:182)
TapGestureRecognizer.handleTapUp (c:\Users\Jan\development\flutter\packages\flutter\lib\src\gestures\tap.dart:486)
BaseTapGestureRecognizer._checkUp (c:\Users\Jan\development\flutter\packages\flutter\lib\src\gestures\tap.dart:264)
BaseTapGestureRecognizer.handlePrimaryPointer (c:\Users\Jan\development\flutter\packages\flutter\lib\src\gestures\tap.dart:199)
PrimaryPointerGestureRecognizer.handleEvent (c:\Users\Jan\development\flutter\packages\flutter\lib\src\gestures\recognizer.dart:467)
PrimaryPointerGestureRecognizer.handleEvent (c:\Users\Jan\development\flutter\packages\flutter\lib\src\gestures\recognizer.dart:0)
PointerRouter._dispatch (c:\Users\Jan\development\flutter\packages\flutter\lib\src\gestures\pointer_router.dart:76)
PointerRouter._dispatchEventToRoutes.<anonymous closure> (c:\Users\Jan\development\flutter\packages\flutter\lib\src\gestures\pointer_router.dart:117)
_LinkedHashMapMixin.forEach (dart:collection-patch/compact_hash.dart:379)
PointerRouter._dispatchEventToRoutes (c:\Users\Jan\development\flutter\packages\flutter\lib\src\gestures\pointer_router.dart:115)
PointerRouter.route (c:\Users\Jan\development\flutter\packages\flutter\lib\src\gestures\pointer_router.dart:101)
GestureBinding.handleEvent (c:\Users\Jan\development\flutter\packages\flutter\lib\src\gestures\binding.dart:218)
GestureBinding.dispatchEvent (c:\Users\Jan\development\flutter\packages\flutter\lib\src\gestures\binding.dart:198)
GestureBinding._handlePointerEvent (c:\Users\Jan\development\flutter\packages\flutter\lib\src\gestures\binding.dart:156)
GestureBinding._flushPointerEventQueue (c:\Users\Jan\development\flutter\packages\flutter\lib\src\gestures\binding.dart:102)
GestureBinding._handlePointerDataPacket (c:\Users\Jan\development\flutter\packages\flutter\lib\src\gestures\binding.dart:86)
GestureBinding._handlePointerDataPacket (c:\Users\Jan\development\flutter\packages\flutter\lib\src\gestures\binding.dart:0)
_rootRunUnary (dart:async/zone.dart:1138)
_rootRunUnary (dart:async/zone.dart:0)
_CustomZone.runUnary (dart:async/zone.dart:1031)
_CustomZone.runUnaryGuarded (dart:async/zone.dart:933)
_invoke1 (dart:ui/hooks.dart:273)
_dispatchPointerDataPacket (dart:ui/hooks.dart:182)
```

mikolak, 2020-02-24:
It seems `destroyClient()` has been called while there still were running operations and it should be called only after all operations are finished. Be sure to await for the returned futures. 

Could you take a look at this? https://github.com/flutter/flutter/issues/10912

slole, 2020-02-24:
Yes, I'm looking into if cancelling the subscription on the .listen of the device state will help.

mikolak, 2020-02-25:
Can you try running the example from the library? The auto test will fail a lot since it has hardcoded UUIDs of services and characteristic of a Sensor Tag CC2541, but you could use it to check if the exception still occurs.

slole, 2020-02-25:
I partially fixed the error by not calling .destroyClient();, but I'm still getting:

```
D/BluetoothGatt(21092): close()
D/BluetoothGatt(21092): unregisterApp() - mClientIf=8
E/flutter (21092): [ERROR:flutter/lib/ui/ui_dart_state.cc(157)] Unhandled Exception: BleError (Error code: 201, ATT error code: 0, iOS error code: null, Android error code: null, reason: Disconnected from C1:9E:79:91:D0:7E with status 0 (GATT_SUCCESS), internal message: null, device ID: C1:9E:79:91:D0:7E, service UUID: null, characteristic UUID: null, descriptor UUID: null)
[38;5;248mE/flutter (21092): #0      CharacteristicsMixin._throwErrorIfMatchesWithTransactionId[39;49m
[38;5;248mE/flutter (21092): #1      CharacteristicsMixin._createMonitoringStream.<anonymous closure>[39;49m
[38;5;244mE/flutter (21092): #2      _invokeErrorHandler  (dart:async/async_error.dart:19:29)[39;49m
[38;5;244mE/flutter (21092): #3      _HandleErrorStream._handleError  (dart:async/stream_pipe.dart:288:9)[39;49m
[38;5;244mE/flutter (21092): #4      _ForwardingStreamSubscription._handleError  (dart:async/stream_pipe.dart:170:13)[39;49m
[38;5;244mE/flutter (21092): #5      _rootRunBinary  (dart:async/zone.dart:1146:38)[39;49m
[38;5;244mE/flutter (21092): #6      _CustomZone.runBinary  (dart:async/zone.dart:1039:19)[39;49m
[38;5;244mE/flutter (21092): #7      _CustomZone.runBinaryGuarded  (dart:async/zone.dart:941:7)[39;49m
[38;5;244mE/flutter (21092): #8      _BufferingStreamSubscription._sendError.sendError  (dart:async/stream_impl.dart:357:15)[39;49m
[38;5;244mE/flutter (21092): #9      _BufferingStreamSubscription._sendError  (dart:async/stream_impl.dart:375:16)[39;49m
[38;5;244mE/flutter (21092): #10     _BufferingStreamSubscription._addError  (dart:async/stream_impl.dart:274:7)[39;49m
[38;5;244mE/flutter (21092): #11     _ForwardingStreamSubscription._addError  (dart:async/stream_pipe.dart:139:11)[39;49m
[38;5;244mE/flutter (21092): #12     _ForwardingStream._handleError  (dart:async/stream_pipe.dart:104:10)[39;49m
[38;5;244mE/flutter (21092): #13     _ForwardingStreamSubscription._handleError  (dart:async/stream_pipe.dart:170:13)[39;49m
[38;5;244mE/flutter (21092): #14     _rootRunBinary  (dart:async/zone.dart:1146:38)[39;49m
[38;5;244mE/flutter (21092): #15     _CustomZone.runBinary  (dart:async/zone.dart:1039:19)[39;49m
[38;5;244mE/flutter (21092): #16     _CustomZone.runBinaryGuarded  (dart:async/zone.dart:941:7)[39;49m
[38;5;244mE/flutter (21092): #17     _BufferingStreamSubscription._sendError.sendError  (dart:async/stream_impl.dart:357:15)[39;49m
[38;5;244mE/flutter (21092): #18     _BufferingStreamSubscription._sendError  (dart:async/stream_impl.dart:375:16)[39;49m
[38;5;244mE/flutter (21092): #19     _BufferingStreamSubscription._addError  (dart:async/stream_impl.dart:274:7)[39;49m
[38;5;244mE/flutter (21092): #20     _ForwardingStreamSubscription._addError  (dart:async/stream_pipe.dart:139:11)[39;49m
[38;5;244mE/flutter (21092): #21     _ForwardingStream._handleError  (dart:async/stream_pipe.dart:104:10)[39;49m
[38;5;244mE/flutter (21092): #22     _ForwardingStreamSubscription._handleError  (dart:async/stream_pipe.dart:170:13)[39;49m
[38;5;244mE/flutter (21092): #23     _rootRunBinary  (dart:async/zone.dart:1146:38)[39;49m
[38;5;244mE/flutter (21092): #24     _CustomZone.runBinary  (dart:async/zone.dart:1039:19)[39;49m
[38;5;244mE/flutter (21092): #25     _CustomZone.runBinaryGuarded  (dart:async/zone.dart:941:7)[39;49m
[38;5;244mE/flutter (21092): #26     _BufferingStreamSubscription._sendError.sendError  (dart:async/stream_impl.dart:357:15)[39;49m
[38;5;244mE/flutter (21092): #27     _BufferingStreamSubscription._sendError  (dart:async/stream_impl.dart:375:16)[39;49m
[38;5;244mE/flutter (21092): #28     _DelayedError.perform  (dart:async/stream_impl.dart:604:14)[39;49m
[38;5;244mE/flutter (21092): #29     _StreamImplEvents.handleNext  (dart:async/stream_impl.dart:709:11)[39;49m
[38;5;244mE/flutter (21092): #30     _PendingEvents.schedule.<anonymous closure>  (dart:async/stream_impl.dart:669:7)[39;49m
[38;5;244mE/flutter (21092): #31     _rootRun  (dart:async/zone.dart:1122:38)[39;49m
[38;5;244mE/flutter (21092): #32     _CustomZone.run  (dart:async/zone.dart:1023:19)[39;49m
[38;5;244mE/flutter (21092): #33     _CustomZone.runGuarded  (dart:async/zone.dart:925:7)[39;49m
[38;5;244mE/flutter (21092): #34     _CustomZone.bindCallbackGuarded.<anonymous closure>  (dart:async/zone.dart:965:23)[39;49m
[38;5;244mE/flutter (21092): #35     _rootRun  (dart:async/zone.dart:1126:13)[39;49m
[38;5;244mE/flutter (21092): #36     _CustomZone.run  (dart:async/zone.dart:1023:19)[39;49m
[38;5;244mE/flutter (21092): #37     _CustomZone.runGuarded  (dart:async/zone.dart:925:7)[39;49m
[38;5;244mE/flutter (21092): #38     _CustomZone.bindCallbackGuarded.<anonymous closure>  (dart:async/zone.dart:965:23)[39;49m
[38;5;244mE/flutter (21092): #39     _microtaskLoop  (dart:async/schedule_microtask.dart:43:21)[39;49m
[38;5;244mE/flutter (21092): #40     _startMicrotaskLoop  (dart:async/schedule_microtask.dart:52:5)[39;49m
E/flutter (21092):
D/com.polidea.flutter_ble_lib.FlutterBleLibPlugin(21092): on native side observed method: cancelTransaction
D/com.polidea.flutter_ble_lib.FlutterBleLibPlugin(21092): on native side observed method: createClient
D/com.polidea.flutter_ble_lib.FlutterBleLibPlugin(21092): on native side observed method: startDeviceScan
```

slole, 2020-02-25:
The problem was the I wasn't closing the streams. I solved the problem by calling `connectionStream?.cancel();` and `rxMonitor?.cancel();` in my disconnect function.

closed 2020-02-25 by slole