dotintent/FlutterBleLib #435: [Android][iOS] Throughput information

rmorbach, 2020-02-27:
I wonder if there is a throughput limitation for the library. While using `BLEEmulator` with a sampling rate of 100ms the application behaves normally.
However, while connecting to a real device with the same parameters, the application starts to behave oddly, lagging and impossible to use.

Have you ever experienced something like that?

mikolak, 2020-02-27:
Hi! To be honest it never happened to me. 

By sampling rate you mean advertisement interval? 
What platforms/devices are you checking against?
Could you make a video?
Could you recreate this behaviour with profiler turned on and record the session?

rmorbach, 2020-02-27:
Thanks for the quick reply.
The interval is for the device to send value of a given Characteristic.
So I added an observer to that Characteristic and receive some value each 100 ms.
I am not able to make a video, sorry.
Do you think this can be related to the interval?

mikolak, 2020-02-28:
Ah, notifications then. Are you using notifications or indications? Do you think it might be connected to #362? Does the issue occur on both platforms?

rmorbach, 2020-03-03:
@mikolak the problem was raising by an issue in the Flutter rendering process. Nothing related to this awesome library. 

Thank you for your time.

closed 2020-03-03 by rmorbach