dotintent/FlutterBleLib #443: Can you provide a sample app with only FlutterBleLib plugin?

NOA-Alain, 2020-03-06:
Hi, i am new to Flutter and Dart. I find it difficult to understand the usage of FlutterBleLib using the sample app.  
There is lots of code about  ```Bloc``` and ```rxDart``` which is difficult to read. Can you provide a sample app without design patten or unrelated plugins?

mikolak, 2020-03-06:
Hi!

Creating a new simplified sample app will take time.

For the time being, you can start with just two files:
Initialisation https://github.com/Polidea/FlutterBleLib/blob/develop/example/lib/devices_list/devices_bloc.dart#L52

Scanning for peripherals:
https://github.com/Polidea/FlutterBleLib/blob/develop/example/lib/devices_list/devices_bloc.dart#L111
(same file)

Most of operations on a connected peripheral
https://github.com/Polidea/FlutterBleLib/blob/develop/example/lib/test_scenarios/peripheral_test_operations.dart

leech001, 2020-04-21:
Still, if it is not difficult, I would like to see a simple example of using the library.

sonnny, 2020-04-22:
See link below simple example, assumed you know device name, service and char uuid, use nrf connect to find out. It just writes to the device.

[link for simple ble write example](http://sayaw.net/flutter/another_ble.html)

JasonEdinburgh, 2020-09-23:
@sonnny link http://sayaw.net/flutter/another_ble.html is broken

sonnny, 2020-09-24:
@JasonEdinburgh good link http://orasna.com/flutter/another_ble.html

This is a basic example.

JasonEdinburgh, 2020-09-27:
Thank you @sonnny 

But I need something that shows how to catch the exceptions that are thrown when calling peripheral.connect etc. 

My current code does this to initiate a connection and send/receive via characteristics (nordic uart service), but I occasionally get unhandled exceptions. I am not sure yet if they originate in other threads/isolates. 

`    try {
      _connectPressed = true;
      Peripheral peripheral = myPeripherals[widget.deviceId];
        await peripheral.connect(timeout: Duration(seconds: 2)).catchError((theError) {
          print("Caught Error in peripheral.connect ($theError)");
          return;
        });
        print("connected");
        await peripheral.discoverAllServicesAndCharacteristics(transactionId: "discovery").catchError((theError) {
          print("Caught Error in peripheral.discoverAllServicesAndCharacteristics ($theError)");
          return;
        });
        print("discoverAllServicesAndCharacteristics completed");
        await peripheral.services().catchError((theError) {
          print("Caught Error in peripheral.services ($theError)");
          return;
        });
        await peripheral.writeCharacteristic(NUS_SERVICE_UUID, WRITE_TO_DEVICE_CHARACTERISTIC_UUID, dataToSend, withResponse).catchError((theError) {
          print("Caught Error in peripheral.writeCharacteristic ($theError)");
          return;
        });

        peripheral.monitorCharacteristic(NUS_SERVICE_UUID, READ_FROM_DEVICE_CHARACTERISTIC_UUID).listen((characteristicWithValue) {
          talkToDevice(characteristicWithValue, peripheral);
        }).onError((theError) {
          print("Caught Error in peripheral.monitorCharacteristic ($theError)");
          return;
        });

 } catch (ex, st) {
      print("exception caught ${ex.toString()} at $st");
    } finally {
      print("finally setting _connectPressed = false");
      _connectPressed = false;
    }`

ride4sun, 2020-12-09:
Is there any progress on this issue?