dotintent/FlutterBleLib #447: Unable to listen characteristics for a second time.

NailAgliev, 2020-03-18:
Hello all. I connect to the device and begin to listen to the characteristics and for the first time everything works fine.
``` 
weightStream = characteristics1[0]
          .monitor(transactionId: 'weightStream')
          .listen((data) => _handleWeightStream(data, _peripheral));
```
Then device which i was connected to simply restarts for emulating going out of range. 
```
await connectionStream?.cancel();
await weightStream?.cancel();
add(StartScan());
```
After that, I connect to the device a second time, and when I start listening again, an error occurs.
```
D/com.polidea.flutter_ble_lib.FlutterBleLibPlugin(16120): on native side observed method: monitorCharacteristicForIdentifier
D/BluetoothGatt(16120): setCharacteristicNotification() - uuid: 00002a98-0000-1000-8000-00805f9b34fb enable: false
E/flutter (16120): [ERROR:flutter/lib/ui/ui_dart_state.cc(157)] Unhandled Exception: BleError (Error code: 2, ATT error code: null, iOS error code: null, Android error code: null, reason: null, internal message: null, device ID: null, service UUID: null, characteristic UUID: null, descriptor UUID: null)
E/flutter (16120): #0      CharacteristicsMixin._throwErrorIfMatchesWithTransactionId (package:flutter_ble_lib/src/bridge/characteristics_mixin.dart:287:7)
E/flutter (16120): #1      CharacteristicsMixin._createMonitoringStream.<anonymous closure> (package:flutter_ble_lib/src/bridge/characteristics_mixin.dart:247:13)
E/flutter (16120): #2      _invokeErrorHandler (dart:async/async_error.dart:19:29)
E/flutter (16120): #3      _HandleErrorStream._handleError (dart:async/stream_pipe.dart:288:9)
E/flutter (16120): #4      _ForwardingStreamSubscription._handleError (dart:async/stream_pipe.dart:170:13)
E/flutter (16120): #5      _rootRunBinary (dart:async/zone.dart:1146:38)
E/flutter (16120): #6      _CustomZone.runBinary (dart:async/zone.dart:1039:19)
E/flutter (16120): #7      _CustomZone.runBinaryGuarded (dart:async/zone.dart:941:7)
E/flutter (16120): #8      _BufferingStreamSubscription._sendError.sendError (dart:async/stream_impl.dart:357:15)
E/flutter (16120): #9      _BufferingStreamSubscription._sendError (dart:async/stream_impl.dart:375:16)
E/flutter (16120): #10     _BufferingStreamSubscription._addError (dart:async/stream_impl.dart:274:7)
E/flutter (16120): #11     _ForwardingStreamSubscription._addError (dart:async/stream_pipe.dart:139:11)
E/flutter (16120): #12     _ForwardingStream._handleError (dart:async/stream_pipe.dart:104:10)
E/flutter (16120): #13     _ForwardingStreamSubscription._handleError (dart:async/stream_pipe.dart:170:13)
E/flutter (16120): #14     _rootRunBinary (dart:async/zone.dart:1146:38)
E/flutter (16120): #15     _CustomZone.runBinary (dart:async/zone.dart:1039:19)
E/flutter (16120): #16     _CustomZone.runBinaryGuarded (dart:async/zone.dart:941:7)
E/flutter (16120): #17     _BufferingStreamSubscription._sendError.sendError (dart:async/stream_impl.dart:357:15)
E/flutter (16120): #18     _BufferingStreamSubscription._sendError (dart:async/stream_impl.dart:375:16)
E/flutter (16120): #19     _BufferingStreamSubscription._addError (dart:async/stream_impl.dart:274:7)
E/flutter (16120): #20     _ForwardingStreamSubscription._addError (dart:async/stream_pipe.dart:139:11)
E/flutter (16120): #21     _ForwardingStream._handleError (dart:async/stream_pipe.dart:104:10)
E/flutter (16120): #22     _ForwardingStreamSubscription._handleError (dart:async/stream_pipe.dart:170:13)
E/flutter (16120): #23     _rootRunBinary (dart:async/zone.dart:1146:38)
E/flutter (16120): #24     _CustomZone.runBinary (dart:async/zone.dart:1039:19)
E/flutter (16120): #25     _CustomZone.runBinaryGuarded (dart:async/zone.dart:941:7)
E/flutter (16120): #26     _BufferingStreamSubscription._sendError.sendError (dart:async/stream_impl.dart:357:15)
E/flutter (16120): #27     _BufferingStreamSubscription._sendError (dart:async/stream_impl.dart:375:16)
E/flutter (16120): #28     _DelayedError.perform (dart:async/stream_impl.dart:604:14)
E/flutter (16120): #29     _StreamImplEvents.handleNext (dart:async/stream_impl.dart:709:11)
E/flutter (16120): #30     _PendingEvents.schedule.<anonymous closure> (dart:async/stream_impl.dart:669:7)
E/flutter (16120): #31     _rootRun (dart:async/zone.dart:1122:38)
E/flutter (16120): #32     _CustomZone.run (dart:async/zone.dart:1023:19)
E/flutter (16120): #33     _CustomZone.runGuarded (dart:async/zone.dart:925:7)
E/flutter (16120): #34     _CustomZone.bindCallbackGuarded.<anonymous closure> (dart:async/zone.dart:965:23)
E/flutter (16120): #35     _rootRun (dart:async/zone.dart:1126:13)
E/flutter (16120): #36     _CustomZone.run (dart:async/zone.dart:1023:19)
E/flutter (16120): #37     _CustomZone.runGuarded (dart:async/zone.dart:925:7)
E/flutter (16120): #38     _CustomZone.bindCallbackGuarded.<anonymous closure> (dart:async/zone.dart:965:23)
E/flutter (16120): #39     _microtaskLoop (dart:async/schedule_microtask.dart:43:21)
E/flutter (16120): #40     _startMicrotaskLoop (dart:async/schedule_microtask.dart:52:5)
E/flutter (16120): 
E/flutter (16120): [ERROR:flutter/lib/ui/ui_dart_state.cc(157)] Unhandled Exception: BleError (Error code: 2, ATT error code: null, iOS error code: null, Android error code: null, reason: null, internal message: null, device ID: null, service UUID: null, characteristic UUID: null, descriptor UUID: null)
E/flutter (16120): #0      CharacteristicsMixin._throwErrorIfMatchesWithTransactionId (package:flutter_ble_lib/src/bridge/characteristics_mixin.dart:287:7)
E/flutter (16120): #1      CharacteristicsMixin._createMonitoringStream.<anonymous closure> (package:flutter_ble_lib/src/bridge/characteristics_mixin.dart:247:13)
E/flutter (16120): #2      _invokeErrorHandler (dart:async/async_error.dart:19:29)
E/flutter (16120): #3      _HandleErrorStream._handleError (dart:async/stream_pipe.dart:288:9)
E/flutter (16120): #4      _ForwardingStreamSubscription._handleError (dart:async/stream_pipe.dart:170:13)
E/flutter (16120): #5      _rootRunBinary (dart:async/zone.dart:1146:38)
E/flutter (16120): #6      _CustomZone.runBinary (dart:async/zone.dart:1039:19)
E/flutter (16120): #7      _CustomZone.runBinaryGuarded (dart:async/zone.dart:941:7)
E/flutter (16120): #8      _BufferingStreamSubscription._sendError.sendError (dart:async/stream_impl.dart:357:15)
E/flutter (16120): #9      _BufferingStreamSubscription._sendError (dart:async/stream_impl.dart:375:16)
[... 9 lines cut ...]
E/flutter (16120): #19     _BufferingStreamSubscription._addError (dart:async/stream_impl.dart:274:7)
E/flutter (16120): #20     _ForwardingStreamSubscription._addError (dart:async/stream_pipe.dart:139:11)
E/flutter (16120): #21     _ForwardingStream._handleError (dart:async/stream_pipe.dart:104:10)
E/flutter (16120): #22     _ForwardingStreamSubscription._handleError (dart:async/stream_pipe.dart:170:13)
E/flutter (16120): #23     _rootRunBinary (dart:async/zone.dart:1146:38)
E/flutter (16120): #24     _CustomZone.runBinary (dart:async/zone.dart:1039:19)
E/flutter (16120): #25     _CustomZone.runBinaryGuarded (dart:async/zone.dart:941:7)
E/flutter (16120): #26     _BufferingStreamSubscription._sendError.sendError (dart:async/stream_impl.dart:357:15)
E/flutter (16120): #27     _BufferingStreamSubscription._sendError (dart:async/stream_impl.dart:375:16)
E/flutter (16120): #28     _DelayedError.perform (dart:async/stream_impl.dart:604:14)
E/flutter (16120): #29     _StreamImplEvents.handleNext (dart:async/stream_impl.dart:709:11)
E/flutter (16120): #30     _PendingEvents.schedule.<anonymous closure> (dart:async/stream_impl.dart:669:7)
E/flutter (16120): #31     _rootRun (dart:async/zone.dart:1122:38)
E/flutter (16120): #32     _CustomZone.run (dart:async/zone.dart:1023:19)
E/flutter (16120): #33     _CustomZone.runGuarded (dart:async/zone.dart:925:7)
E/flutter (16120): #34     _CustomZone.bindCallbackGuarded.<anonymous closure> (dart:async/zone.dart:965:23)
E/flutter (16120): #35     _rootRun (dart:async/zone.dart:1126:13)
E/flutter (16120): #36     _CustomZone.run (dart:async/zone.dart:1023:19)
E/flutter (16120): #37     _CustomZone.runGuarded (dart:async/zone.dart:925:7)
E/flutter (16120): #38     _CustomZone.bindCallbackGuarded.<anonymous closure> (dart:async/zone.dart:965:23)
E/flutter (16120): #39     _microtaskLoop (dart:async/schedule_microtask.dart:43:21)
E/flutter (16120): #40     _startMicrotaskLoop (dart:async/schedule_microtask.dart:52:5)
E/flutter (16120): 
D/com.polidea.flutter_ble_lib.FlutterBleLibPlugin(16120): on native side observed method: cancelTransaction
D/com.polidea.flutter_ble_lib.FlutterBleLibPlugin(16120): on native side observed method: cancelTransaction

NailAgliev, 2020-03-18:
Not sure how but it work now.  
```
 weightStream = _peripheral.monitorCharacteristic(
        WEIGHT_SCALE_SERVICE_UUID, WEIGHT_CHARACTERISTIC_UUID)
       .listen((data) => _handleWeightStream(data.value, _peripheral));
```

closed 2020-03-18 by NailAgliev

mikolak, 2020-03-19:
Hi!

Error code 2 means `BleErrorCode.OperationCancelled`. If you don't provide your own transaction ID, the library generates one itself.

Weird thing is that unsubscribing from the monitoring stream should've called `BleManager.cancelTransaction()`. Another weird thing is that you should've gotten peripheral disconnected error on the stream which would cancel it.