dotintent/FlutterBleLib #450: Can not work on iPad and iPhone

hu70258tao, 2020-03-31:
both system is 13.4，
iPad sometime can not read and write，and tall bleerro code 205,means disconnect，but I try check bledevice,it still connected。
iPhone can not scan any bledevice，and always BluetoothState.UNKNOWN 。

mikolak, 2020-03-31:
Hi!

Have you followed the installation guide? https://github.com/Polidea/FlutterBleLib#ios
Can you run the sample the from the library and see if you encounter the same issues?

hu70258tao, 2020-04-01:
my English is very weak，I test as soon as posible

hu70258tao, 2020-04-01:
it can run on iPad,but throw erro on iPhone,
[image]

hu70258tao, 2020-04-01:
and this project is your example

hu70258tao, 2020-04-01:
[image]
Ipad can scan BleDevice,but iPhone can not

hu70258tao, 2020-04-01:
And I user a iOS App called LightBlue,it can scan many device
[image]

hu70258tao, 2020-04-01:
I test iPhone8 system is 13.4，Iphone7 system 13.31,both not work,
and I test IphoneX just now,it is work,and system is 13.3.1

hu70258tao, 2020-04-01:
[image]
and when I write anything,it return erro coed 205 like this.i finad it is mean device disconnect,but my bluedevice received message and it is still connected.

john-yick, 2020-04-03:
@hu70258tao It should work fine, I am testing on an iPhone X on ios 13.4 with running flutter stable version.

mikolak, 2020-04-03:
I have no idea about iPad and currently have no means to test it, but the example filters out peripherals without a name.

https://github.com/Polidea/FlutterBleLib/blob/79ebf4423969dedaca3cca0b9220504d121aab80/example/lib/devices_list/devices_bloc.dart#L113

Change the condition in _startScan() to:

```
void _startScan() {
    Fimber.d("Ble client created");
    _scanSubscription =
        _bleManager.startPeripheralScan().listen((ScanResult scanResult) {
      var bleDevice = BleDevice(scanResult);
      if (!bleDevices.contains(bleDevice)) {
        Fimber.d(
            'found new device ${scanResult.advertisementData.localName} ${scanResult.peripheral.identifier}');
        bleDevices.add(bleDevice);
        _visibleDevicesController.add(bleDevices.sublist(0));
      }
    });
  }
```
And it should find your unnamed peripherals.

I'll change the example to stop omitting those.

hu70258tao, 2020-04-14:
thanks very much,i will try and test

hu70258tao, 2020-04-14:
and I write ,it always err like this
[image]

mikolak, 2021-04-07:
It'd seem you're trying to write an unsupported value.
https://stackoverflow.com/a/63341036

closed 2021-04-07 by mikolak