dotintent/FlutterBleLib #456: Unhandled Exception occurs when disconnecting from a BLE device on Android OS

t-girniak, 2020-04-24:
When I disconnect from a BLE device, my main.dart class catches unhandled Exception which occurs on Native Android part of the plugin (While the same codebase on iOS doesn't). Device disconnects, but still I'm getting the following error in Logs:

```
E/flutter (19065): [ERROR:flutter/lib/ui/ui_dart_state.cc(157)] Unhandled Exception: 'package:appName/main_development.dart': Failed assertion: line 39 pos 12: 'false': Unhandled error: BleError (Error code: 201, ATT error code: 0, iOS error code: null, Android error code: null, reason: Disconnected from C2:80:ED:C1:6E:B5 with status 0 (GATT_SUCCESS), internal message: null, device ID: C2:80:ED:C1:6E:B5, service UUID: null, characteristic UUID: null, descriptor UUID: null)
E/flutter (19065): #0      _AssertionError._doThrowNew (dart:core-patch/errors_patch.dart:42:39)
E/flutter (19065): #1      _AssertionError._throwNew (dart:core-patch/errors_patch.dart:38:5)
E/flutter (19065): #2      main.<anonymous closure> (package:appName/main_development.dart:39:12)
E/flutter (19065): #3      _rootRunBinary (dart:async/zone.dart:1150:13)
E/flutter (19065): #4      _CustomZone.runBinary (dart:async/zone.dart:1039:19)
E/flutter (19065): #5      runZoned.<anonymous closure> (dart:async/zone.dart:1481:21)
E/flutter (19065): #6      _CustomZone.handleUncaughtError (dart:async/zone.dart:1005:19)
E/flutter (19065): #7      _nullErrorHandler (dart:async/stream_impl.dart:574:16)
E/flutter (19065): #8      _rootRunBinary (dart:async/zone.dart:1146:38)
E/flutter (19065): #9      _CustomZone.runBinary (dart:async/zone.dart:1039:19)
E/flutter (19065): #10     _CustomZone.runBinaryGuarded (dart:async/zone.dart:941:7)
E/flutter (19065): #11     _BufferingStreamSubscription._sendError.sendError (dart:async/stream_impl.dart:357:15)
E/flutter (19065): #12     _BufferingStreamSubscription._sendError (dart:async/stream_impl.dart:375:16)
E/flutter (19065): #13     _DelayedError.perform (dart:async/stream_impl.dart:604:14)
E/flutter (19065): #14     _StreamImplEvents.handleNext (dart:async/stream_impl.dart:709:11)
E/flutter (19065): #15     _PendingEvents.schedule.<anonymous closure> (dart:async/stream_impl.dart:669:7)
E/flutter (19065): #16     _rootRun (dart:async/zone.dart:1122:38)
E/flutter (19065): #17     _CustomZone.run (dart:async/zone.dart:1023:19)
E/flutter (19065): #18     _CustomZone.runGuarded (dart:async/zone.dart:925:7)
E/flutter (19065): #19     _CustomZone.bindCallbackGuarded.<anonymous closure> (dart:async/zone.dart:965:23)
E/flutter (19065): #20     _rootRun (dart:async/zone.dart:1126:13)
E/flutter (19065): #21     _CustomZone.run (dart:async/zone.dart:1023:19)
E/flutter (19065): #22     _CustomZone.runGuarded (dart:async/zone.dart:925:7)
E/flutter (19065): #23     _CustomZone.bindCallbackGuarded.<anonymous closure> (dart:async/zone.dart:965:23)
E/flutter (19065): #24     _microtaskLoop (dart:async/schedule_microtask.dart:43:21)
E/flutter (19065): #25     _startMicrotaskLoop (dart:async/schedule_microtask.dart:52:5)
```

The main part of it says: " Unhandled error: BleError (**Error code: 201**, ATT error code: 0, iOS error code: null, Android error code: null, reason: Disconnected from C2:80:ED:C1:6E:B5 **with status 0 (GATT_SUCCESS)**"

The main_development.dart class:
```
void main() {
  WidgetsFlutterBinding.ensureInitialized();
  final FlavorConfig config = FlavorConfig(
    flavor: Flavor.DEVELOPMENT,
    apiBaseUrl: 'https://apisBaseUrl.net',
    appName: 'appName',
  );
  Crashlytics.instance.enableInDevMode = true;

  FlutterError.onError = Crashlytics.instance.recordFlutterError;
  runZoned<Future<void>>(() async {
    registerGlobalDeps(config);
    SystemChrome.setPreferredOrientations([DeviceOrientation.portraitUp]);
    runApp(
      App(
        connectivity: getIt<Connectivity>(),
        db: getIt<Database>(),
        apiClient: getIt<ApiClient>(),
        log: getIt<Logger>(),
      ),
    );
  }, onError: (e, i) {
    assert(false, 'Unhandled error: $e');
    Crashlytics.instance.recordError(e, i);
  });
}
```

Do you know why does this error occur and how to fix it? Thanks in advance.

mikolak, 2020-04-24:
It comes up in main because you're catching the error there, but the stack trace is incorrect.

201 is peripheral disconnected, so I'm guessing you have some dangling operation, which throws up an error because you've disconnected before it finished. Can't tell more without looking at the BLE code.

t-girniak, 2020-04-27:
Thanks for a quick response, @mikolak . I wasn't unsubscribing from the monitorCharacteristic listeners before disconnecting from the device. After doing so this error had disappeared. 

But how to avoid the "**Disconnected from C2:80:ED:C1:6E:B5 with status 8 (GATT_INSUF_AUTHORIZATION)**" error after the BLE device was turned off in the process of connection?
```
Unhandled Exception: 'package:appName/main_development.dart': Failed assertion: line 39 pos 12: 'false': Unhandled error: BleError (Error code: 201, ATT error code: 8, iOS error code: null, Android error code: null, reason: Disconnected from C2:80:ED:C1:6E:B5 with status 8 (GATT_INSUF_AUTHORIZATION), internal message: null, device ID: C2:80:ED:C1:6E:B5, service UUID: null, characteristic UUID: null, descriptor UUID: null)
```
We need to unsubscribe from monitoring characteristics before we actually disconnect. But what to do if the BLE device was turned off manually? The listeners will be listening for characteristics indicate while the connection was lost and we got this exception.

mikolak, 2020-04-27:
To be honest being disconnected is not that extraordinary, so I guess you should be handling it in all operations, one way or another. User could also turn off BT, which would also result in error being returned by any ongoing operation. Here's the complete list of error codes https://github.com/Polidea/FlutterBleLib/blob/develop/lib/error/ble_error.dart

On Futures you can use `.catchError()`, and on Streams (like characteristic monitoring) you can pass `onError` to `listen()`. 
If the error cannot be handled (ie. is not any expected issue, like peripheral disconnected), just rethrow it from there.

t-girniak, 2020-04-27:
Thanks for your help! 👍 
Maybe this thread will help someone else.
We can close this issue, I think

closed 2020-04-27 by t-girniak