dotintent/FlutterBleLib #458: "writing is not permitted" error on native iOS when writing to the characteristic on iPhone X

t-girniak, 2020-05-04:
When I try to write [0x01] on **iPhone X** to the characteristic on a BLE device I got a platform error, which says: **"writing is not permitted"**. It works well on any Android phone. It also works on iPhone 6S (iOS 13.3), but **on iPhone X it doesn't**. I traced this error up to rejectForFlutterResult method:
[image]
When debugging in Android studio I catch the following error: 

> BleError (Error code: 401, ATT error code: 3, iOS error code: null, Android error code: null, reason: Writing is not permitted.,

[image]

I reproduced the same characteristic write sequence in the Nordic nRF Connect app, and everything worked ok (on the same iPhone X). Do you know why this error occurs?

P.S. Thanks for this library, it's the most stable BLE library for the Flutter out there yet :)

mikolak, 2020-05-06:
Same code works for iPhone 6s with iOS 13.3, but does not work for iPhone X, yes? What's the system version on X? Could  you check the permission flags on that characteristic (`isWritableWithResponse`, `isWritableWithoutResponse`)?

t-girniak, 2020-05-06:
Yes, the same code works on iPhone 6s (iOS 13.3) and any Android device, but doesn't on iPhone X (iOS 13.4). The characteristic is "WRITE_NO_RESPONSE". I use the following code:
```
Future<Characteristic> writeRfidControl(Peripheral peripheral, List<int> intList) async {
  return await peripheral.writeCharacteristic(
      StargateAttributes.RFID_SERVICE, StargateAttributes.CONTROL, toBytes(intList), false);
}
```
Here's nRF connect screenshot:
[image]

mikolak, 2020-05-07:
Are you using the newest version of the library?

t-girniak, 2020-05-07:
Yes, I am using the latest version (flutter_ble_lib: ^2.2.3). I'd just tested develop branch ref, but still, iPhone X can't write the characteristic :(

mikolak, 2020-05-12:
Are both iPhones running the same OS? 

The error occurs in step 4 or 5 of your stacktrace (safeWriteCharacteristicForDevice or subscribe). The one mentioned in the selected object leads to Multiplatform BLE Adapter and somewhere here: https://github.com/Polidea/MultiPlatformBleAdapter/blob/master/iOS/RxBluetoothKit/RxCBPeripheral.swift#L306

Could you try setting logs to verbose in FlutterBleLib and see if there's any more useful information there? 

I don't have access to iPhone X for the foreseeable future, so you'll have to help me if you want it fixed. ;)

t-girniak, 2020-05-12:
Of course, I will try help :)
iPhone 6s currently runs on iOS 13.4.1, while iPhone X is on iOS 13.4. I will have access to the iPhone 6S in a day or two so I can debug app personally on it and see if the issue lies in the iPhone, in iOS version or in something else

mikolak, 2020-05-13:
Start with verbose logs, if possible.

t-girniak, 2020-05-13:
That's all I managed to track in Xcode, all the rest is leading to your RxBluetooth library. Unfortunately, I can't send you more data from our app logs. Also, the same codebase suddenly stopped working on iPhone 6S. I am trying to figure out why.
[image]
[image]

austbot, 2020-05-22:
@mikolak Any Ideas on this I work with @t-girniak

austbot, 2020-05-22:
@mikolak I have iphonex and Iphone 11 both are throwing the same error @t-girniak listed,how can I help?

austbot, 2020-05-22:
BleError (Error code: 401, ATT error code: 3, iOS error code: null, Android error code: null, reason: Writing is not permitted.,

austbot, 2020-05-22:
@mikolak Some notable things I see in verbose logging are 
```
[CoreBluetooth] API MISUSE: <CBCentralManager: 0x282b54e00> has no restore identifier but the delegate implements the centralManager:willRestoreState: method. Restoring will not be supported
2020-05-22 11:40:25.051587-0500 Runner[3463:1262593] User granted permissions
[RxBLEKit|DEBG|16:40:25.058]: CentralManager(10782854656) didUpdateState(state: poweredOn)

--- SCanning 
RxBLEKit|DEBG|16:41:40.743]: CentralManager(10782854656) scanForPeripherals(
withServices: Optional("[]"),
options: Optional([:]))
[RxBLEKit|VERB|16:41:40.860]: CentralManager(10782854656) didDiscover(peripheral: Peripheral(uuid: 1BBD8B5E-6750-97E5-976D-4091C853E05B, name: Optional("Apple TV")),
rssi: -45)
[RxBLEKit|VERB|16:41:40.939]: CentralManager(10782854656) didDiscover(peripheral: Peripheral(uuid: 54478E68-FE26-468E-2FF1-63B7F0C1681A, name: Optional("iPhone")),
rssi: -36)
[RxBLEKit|VERB|16:41:40.940]: CentralManager(10782854656) didDiscover(peripheral: Peripheral(uuid: 4EE3DA7B-6B8A-2EB5-813B-D295C9F37020, name: nil),
rssi: -61)
[RxBLEKit|VERB|16:41:40.951]: CentralManager(10782854656) didDiscover(peripheral: Peripheral(uuid: 26D3D08E-F5DF-C588-B5FB-63687DCDD9A6, name: nil),
rssi: -43)
[RxBLEKit|VERB|16:41:40.993]: CentralManager(10782854656) didDiscover(peripheral: Peripheral(uuid: C6053FA9-EFB8-C450-315B-84C289024130, name: nil),
rssi: -58)
[RxBLEKit|VERB|16:41:41.055]: CentralManager(10782854656) didDiscover(peripheral: Peripheral(uuid: 22A78875-CF5A-4BEE-9EF8-AB94A1DB6C57, name: Optional("Austin’s Apple Watch")),
rssi: -37)
[RxBLEKit|VERB|16:41:41.137]: CentralManager(10782854656) didDiscover(peripheral: Peripheral(uuid: 4A31F525-52A1-7AEA-ED97-C1B5F88C6392, name: Optional("stargate-sg1-4")),
rssi: -48)
2020-05-22 11:41:41.182437-0500 Runner[3463:1262781] flutter: Error caught by Crashlytics plugin <recordError>:
2020-05-22 11:41:41.200333-0500 Runner[3463:1262781] flutter: firebase_crashlytics: Error reported to Crashlytics.
[RxBLEKit|VERB|16:41:41.290]: CentralManager(10782854656) didDiscover(peripheral: Peripheral(uuid: D47F84DD-2DDF-D70D-F963-1026979C1537, name: nil),
rssi: -86)
[RxBLEKit|VERB|16:41:41.331]: CentralManager(10782854656) didDiscover(peripheral: Peripheral(uuid: 2587ED2D-2847-B5B6-BC18-9AA95EE3F502, name: Optional("LE-reserved_c")),
rssi: -84)
[RxBLEKit|VERB|16:41:41.617]: CentralManager(10782854656) didDiscover(peripheral: Peripheral(uuid: 5F84AF58-8F75-B8A3-0475-F5F79DB8B162, name: Optional("LE-booty buds")),
rssi: -82)
[RxBLEKit|VERB|16:41:43.035]: CentralManager(10782854656) didDiscover(peripheral: Peripheral(uuid: 7F66D055-D1A9-E09F-A329-67B1C55A0901, name: Optional("LE-brains !!")),
rssi: -52)
[RxBLEKit|VERB|16:41:43.721]: CentralManager(10782854656) didDiscover(peripheral: Peripheral(uuid: D47F84DD-2DDF-D70D-F963-1026979C1537, name: nil),
rssi: -90)
[RxBLEKit|VERB|16:41:54.102]: CentralManager(10782854656) didDiscover(peripheral: Peripheral(uuid: D47F84DD-2DDF-D70D-F963-1026979C1537, name: nil),
rssi: -90)

-----------------------------
[RxBLEKit|DEBG|16:49:06.178]: Peripheral(uuid: 4A31F525-52A1-7AEA-ED97-C1B5F88C6392, name: Optional("stargate-sg1-4")) writeValue(
data: 935e32cef5b745f2b9b9d0d00d0bee7b,
for: Characteristic(uuid: 3FD70402-709E-4D96-9F81-3D37FC17E86D, id: 10802232704),
type: withoutResponse)
**2020-05-22 11:49:06.178609-0500 Runner[3484:1266477] [CoreBluetooth] WARNING: Characteristic <CBCharacteristic: 0x283dcfd80, UUID = 3FD70402-709E-4D96-9F81-3D37FC17E86D, properties = 0x8A, value = (null), notifying = NO> does not specify the "Write Without Response" property - ignoring response-less write**
[RxBLEKit|DEBG|16:49:08.832]: Peripheral(uuid: 4A31F525-52A1-7AEA-ED97-C1B5F88C6392, name: Optional("stargate-sg1-4")) didUpdateValueFor(for:Characteristic(uuid: 3FD70202-709E-4D96-9F81-3D37FC17E86D, id: 10801973216),
value: Optional("1c"),
error: nil)
[RxBLEKit|DEBG|16:49:10.076]: Peripheral(uuid: 4A31F525-52A1-7AEA-ED97-C1B5F88C6392, name: Optional("stargate-sg1-4")) didUpdateValueFor(for:Characteristic(uuid: Battery Level, id: 10802030144),
value: Optional("2c"),
error: nil)
[RxBLEKit|DEBG|16:49:15.835]: Peripheral(uuid: 4A31F525-52A1-7AEA-ED97-C1B5F88C6392, name: Optional("stargate-sg1-4")) didUpdateValueFor(for:Characteristic(uuid: 3FD70202-709E-4D96-9F81-3D37FC17E86D, id: 10801973216),
value: Optional("1c"),
error: nil)
[RxBLEKit|DEBG|16:49:20.104]: Peripheral(uuid: 4A31F525-52A1-7AEA-ED97-C1B5F88C6392, name: Optional("stargate-sg1-4")) didUpdateValueFor(for:Characteristic(uuid: Battery Level, id: 10802030144),
value: Optional("2c"),
error: nil)
[RxBLEKit|DEBG|16:49:22.834]: Peripheral(uuid: 4A31F525-52A1-7AEA-ED97-C1B5F88C6392, name: Optional("stargate-sg1-4")) didUpdateValueFor(for:Characteristic(uuid: 3FD70202-709E-4D96-9F81-3D37FC17E86D, id: 10801973216),
value: Optional("1c"),
error: nil)
```

austbot, 2020-05-22:
@t-girniak @mikolak I changed something in the device firmware to allow the WRITE no response cmd flag and it seems to allow those characteristics to be written to again.

mikolak, 2020-05-22:
Hi, @austbot! Thanks for following up! I think system might cache attributes of a peripheral, so perhaps something changed in the firmware prior to this?

There was an issue fixed in 2.2.3 where `withResponse` flag was ignored on iOS (always had the same value, true I think), but it was fixed in 2.2.4.

Let me know if the issue occurs again, closing this for now.

closed 2020-05-22 by mikolak

jhionan, 2020-06-25:
I'm having this exactly issue on version 2.2.4