dotintent/FlutterBleLib #462: await for enableRadio future stuck e

xandros6, 2020-05-10:


mikolak, 2020-05-11:
I'll see about fixing it, but you can call `BleManager.bluetoothState()` to check if it's enabled or `BleManager.observeBluetoothState()` to monitor the state.

mikolak, 2020-05-11:
From what I see the adapter is returning an error if you're trying to enable an enabled radio.
https://github.com/Polidea/MultiPlatformBleAdapter/blob/master/android/library/src/main/java/com/polidea/multiplatformbleadapter/BleModule.java#L1188

The returned error has error code: BleErrorCode.BluetoothStateChangeFailed (105). Can you surround your call to enable with try/catch and confirm that it really is stuck and not just terminating the rest of the flow because of an uncaught error?

xandros6, 2020-05-20:
BleErrorCode.BluetoothStateChangeFailed is not emitted, so I use BleManager.bluetoothState() as you suggest.

Thanks!

mikolak, 2020-05-20:
Oh! It should be emitted! I'll use this issue to track it, if you don't mind? I'll have to fix it sometime.