dotintent/FlutterBleLib #464: About MTU

xandros6, 2020-05-18:
Hi, 
can you give more information about auto sending multiple chunk of data with MTU=20byte?

Thanks.

mikolak, 2020-05-18:
Hi!

What do you mean autosending? iOS by default tries to negotiate MTU to ~186 bytes, on Android you have to specify it manually, either by passing optional argument to `peripheral.connect()`, or by calling `peripheral.requestMtu()` after connecting.

xandros6, 2020-05-18:
Yes ok, but if I need to send more than MTU bytes, I need to manage it manually or there is some feature of the lib that automates the sending of the chunk of bytes.

In native Android implementations there are 3 options how you can achieve it:
-  You may request higher MTU from the phone using this method.
-  If you are using Write With Response (write type WRITE_TYPE_DEFAULT in Android), and your device supports Long Write (that means it reserves more bytes in a buffer for each writes), you may just write longer packet despite having shorter MTU. Android will automatically chunk data into at-most-MTU-size packets and will send it to the peripheral. For you, it will be transparent. This, however, does not work with Write Without Response (WRITE_TYPE_NO_RESPONSE).
- If your device does not support higher MTUs, or you are using Write Without Response, or Long Write is not possible, you have to split your data into at most 20-bytes chunks.

mikolak, 2020-05-20:
We're not handling it automatically and do not support long write.

`peripheral.requestMtu()` returns the negotiated mtu and you can use that info for splitting chunks of data.

xandros6, 2020-05-20:
Ok, thanks!

closed 2020-05-20 by xandros6