dotintent/FlutterBleLib #466: How to define a discoverable service?

SharonU, 2020-05-25:
It is not clear how do define a service, which would be discoverable on the remote end's side upon connection.

I am developing an app that connects to another BLE device which implements the "Wireless UART" service. I am able to connect and transmit but not receive messages from it. I assume it is because my app is not recognisable to the remote app as one which implements the "Wireless UART" service.
I know as a fact that the remote device expects which ever it connects to to "have" this service; It even initiates a disconnect if the service is missing.

mikolak, 2020-05-26:
Hi! 

I didn't get a chance to look into your issue before you closed it. From what you're writing I assume you need to be in both central and peripheral mode, and the latter is as of now not supported in the library.

If you somehow resolved your issue (possibly using a different library), could you post the solution for future reference?

SharonU, 2020-05-27:
Hi, 

Thanks for your reply, I wasn't sure about this actually being an issue not was I sure wether this was the right place to post it. 
I wasn't able to resolve this yet; Using a reference project (embedded) proves that being both in central and peripheral mode is not needed.
When the remote end initiates the discovery process it receives none - so what is required is make the mobile app respond with the expected service. The question is how...?

mikolak, 2020-05-27:
AFAIK for your mobile app to have any services at all, it must be in the peripheral mode. I'd think your BLE device is transmitting its data as notification, to which you must listen? 

@dariuszseweryn do you have any other ideas?

SharonU, 2020-05-28:
I managed to monitor the the battery service and get actual events, but I still can't monitor the "Wireless UART" service. I guess my solution is to find the library's implementation of the battery service and implement my own "Wireless UART" service. And I also need to see how the battery service is hooked in to the manager.

dariuszseweryn, 2020-06-01:
Hello @SharonU 
I am afraid that with the provided information we cannot help you. We do not know what a "Wireless UART" service is in your case. Generally speaking to connect two devices via BLE one should be a server and another one a client. Currently this library supports only client role.
Usually if the server is to inform the client about something it sends notifications which can be monitored by the client. If the client wants to send something to the server it uses characteristic write.

SharonU, 2020-06-03:
Hi,

Please, never mind the nonsense I wrote about finding the library's implementation of the battery service.

Now, I understand that the library only supports client mode - and that's fine.

The question remains, does the library support being service-discovered by the server? Becase this is exactly my scenario.

Thanks

SharonU, 2020-06-04:
I'm looking for the equivalent of this in FlutterBleLib:

https://doc.qt.io/qt-5/qlowenergycontroller.html#addService

mikolak, 2020-06-08:
I'm afraid FlutterBleLib does not support Peripheral/server role for now. The docs you linked mention it:
> The controller must be in the PeripheralRole

You might want to check out this: https://pub.dev/packages/flutter_ble_peripheral
or look for something similar
Don't know if it's good, as I haven't used this.

closed 2020-06-16 by mikolak