dotintent/FlutterBleLib #467: WebBluetooth API

martinportmann, 2020-05-26:
The package is marked as compatible with flutter web at
https://pub.dev/packages/flutter_ble_lib

I do not see any code in the repository to support that.
This would be a very fine addition to the library if it would also work on the upcoming flutter web.

mikolak, 2020-05-26:
Hi!

It's marked as compatible because we haven't switched to new pubspec.yaml format, I'll try to fix it next week. 

I haven't researched WebBluetooth API, but my colleague mentioned once it might be too different from the mobile one. I'll leave the ticket open to check for compatibility some time, but there are no plans to support it as of now.

martinportmann, 2020-05-26:
I tested the WebBluetooth API locally with some samples and the workflow is a little different because before connecting a dialog pops up to authorize a specific device. Probably a good thing for privacy ...

The API seems on paper easy enough to be integrated.
We would help to test and perhaps will even be willing to implement as soon as we enter into production with a specific mobile app.

dariuszseweryn, 2020-05-26:
If we are about to have the same API for Android, iOS and WebBluetooth then the lowest common demnominator would be WebBluetooth API — it requires a specific type of handling. As far as I know one cannot create a generic sniffer app using WebBluetooth API because to be able to interact with some services these need to be defined _before_ establishing a connection. That's for starters.

Anyway we have that on our stack to research and think of how to address API differences.