dotintent/FlutterBleLib #470: manufacturerData is empty when BLE packet is a Beacon

isaiahtaylor, 2020-05-30:
Hello,

When scanning a BLE device that has the Beacon metadata (adding an Apple-specific ID to the manufacturer data followed by a proximity UUID), I get a scan result, but everything is nulled out. There is no manufacturerData.

mikolak, 2020-06-01:
iOS treats iBeacon differently AFAIK (CoreLocation instead of CoreBluetooth). 

There are dedicated libraries for working with beacons:
https://pub.dev/packages/flutter_beacon
https://pub.dev/packages/beacons

I don't know them and haven't used them, so I can't really recommend anything to you. Might be be worth giving it a shot though.

Does the issue occur on Android as well?

isaiahtaylor, 2020-06-01:
@mikolak you are correct, on more research it appears that iOS sanitizes packets that have iBeacon bits in them.

closed 2020-06-01 by isaiahtaylor