dotintent/FlutterBleLib #475: writeCharacteristic for peripheral not working as expected

sandeepcmsm, 2020-06-10:
Hi,
I'm using 2.2.4, when I tried to write using writeCharacteristic method its giving error I/flutter (14171): 2020-06-10T11:55:15.431690	E	new DeviceDetailsBloc.<ac>:	 BleError caught: 4 GATT exception from MAC address 0C:8C:DC:2E:8B:EA, with type BleGattOperation{description='CHARACTERISTIC_WRITE'}.  

You can see even though Characteristic is writable its saying its read only. 
and multiplatformbleadapter android lib exact cause error is com.polidea.rxandroidble.exceptions.BleGattCannotStartException: GATT exception from MAC address 0C:8C:DC:2E:8B:EA, with type BleGattOperation{description='CHARACTERISTIC_WRITE'}

Here are logs:

D/RxBle#ConnectionOperationQueue(14171): QUEUED   ServiceDiscoveryOperation(40802563)
D/RxBle#ConnectionOperationQueue(14171): STARTED  ServiceDiscoveryOperation(40802563)
D/BluetoothGatt(14171): discoverServices() - device: 0C:8C:DC:2E:8B:EA
D/BluetoothGatt(14171): onSearchComplete() = Device=0C:8C:DC:2E:8B:EA Status=0
D/RxBle#BluetoothGatt$1(14171): onServicesDiscovered status=0
V/RxBle#ServiceDiscoveryOperation(14171): Preparing services description
V/RxBle#ServiceDiscoveryOperation(14171): --------------- ====== Printing peripheral content ====== ---------------
V/RxBle#ServiceDiscoveryOperation(14171): PERIPHERAL ADDRESS: 0C:8C:DC:2E:8B:EA
V/RxBle#ServiceDiscoveryOperation(14171): PERIPHERAL NAME: Movesense 192830000572
V/RxBle#ServiceDiscoveryOperation(14171): -------------------------------------------------------------------------
V/RxBle#ServiceDiscoveryOperation(14171): 
V/RxBle#ServiceDiscoveryOperation(14171): Primary Service - Generic Access (00001800-0000-1000-8000-00805f9b34fb)
V/RxBle#ServiceDiscoveryOperation(14171): Instance ID: 1
V/RxBle#ServiceDiscoveryOperation(14171): -> Characteristics:
V/RxBle#ServiceDiscoveryOperation(14171): 	* Device Name (00002a00-0000-1000-8000-00805f9b34fb)
V/RxBle#ServiceDiscoveryOperation(14171): 	  Properties: [ READ WRITE ]
V/RxBle#ServiceDiscoveryOperation(14171): 	* Appearance (00002a01-0000-1000-8000-00805f9b34fb)
V/RxBle#ServiceDiscoveryOperation(14171): 	  Properties: [ READ ]
V/RxBle#ServiceDiscoveryOperation(14171): 	* Peripheral Preferred Connection Parameters (00002a04-0000-1000-8000-00805f9b34fb)
V/RxBle#ServiceDiscoveryOperation(14171): 	  Properties: [ READ ]
V/RxBle#ServiceDiscoveryOperation(14171): 	* Central Address Resolution (00002aa6-0000-1000-8000-00805f9b34fb)
V/RxBle#ServiceDiscoveryOperation(14171): 	  Properties: [ READ ]
V/RxBle#ServiceDiscoveryOperation(14171): 
V/RxBle#ServiceDiscoveryOperation(14171): Primary Service - Generic Attribute (00001801-0000-1000-8000-00805f9b34fb)
V/RxBle#ServiceDiscoveryOperation(14171): Instance ID: 10
V/RxBle#ServiceDiscoveryOperation(14171): -> Characteristics:
V/RxBle#ServiceDiscoveryOperation(14171): 	* Service Changed (00002a05-0000-1000-8000-00805f9b34fb)
V/RxBle#ServiceDiscoveryOperation(14171): 	  Properties: [ INDICATE ]
V/RxBle#ServiceDiscoveryOperation(14171): 	  -> Descriptors: 
V/RxBle#ServiceDiscoveryOperation(14171): 		* Client Characteristic Configuration (00002902-0000-1000-8000-00805f9b34fb)
V/RxBle#ServiceDiscoveryOperation(14171): 
V/RxBle#ServiceDiscoveryOperation(14171): Primary Service - Device Information (0000180a-0000-1000-8000-00805f9b34fb)
V/RxBle#ServiceDiscoveryOperation(14171): Instance ID: 14
V/RxBle#ServiceDiscoveryOperation(14171): -> Characteristics:
V/RxBle#ServiceDiscoveryOperation(14171): 	* Manufacturer Name String (00002a29-0000-1000-8000-00805f9b34fb)
V/RxBle#ServiceDiscoveryOperation(14171): 	  Properties: [ READ ]
V/RxBle#ServiceDiscoveryOperation(14171): 	* Serial Number String (00002a25-0000-1000-8000-00805f9b34fb)
V/RxBle#ServiceDiscoveryOperation(14171): 	  Properties: [ READ ]
V/RxBle#ServiceDiscoveryOperation(14171): 
V/RxBle#ServiceDiscoveryOperation(14171): Primary Service - Battery Service (0000180f-0000-1000-8000-00805f9b34fb)
V/RxBle#ServiceDiscoveryOperation(14171): Instance ID: 19
V/RxBle#ServiceDiscoveryOperation(14171): -> Characteristics:
V/RxBle#ServiceDiscoveryOperation(14171): 	* Battery Level (00002a19-0000-1000-8000-00805f9b34fb)
V/RxBle#ServiceDiscoveryOperation(14171): 	  Properties: [ READ NOTIFY ]
V/RxBle#ServiceDiscoveryOperation(14171): 	  -> Descriptors: 
V/RxBle#ServiceDiscoveryOperation(14171): 		* Client Characteristic Configuration (00002902-0000-1000-8000-00805f9b34fb)
V/RxBle#ServiceDiscoveryOperation(14171): 
V/RxBle#ServiceDiscoveryOperation(14171): Primary Service - Unknown service (6e400001-b5a3-f393-e0a9-e50e24dcca9e)
V/RxBle#ServiceDiscoveryOperation(14171): Instance ID: 23
V/RxBle#ServiceDiscoveryOperation(14171): -> Characteristics:
V/RxBle#ServiceDiscoveryOperation(14171): 	* Unknown characteristic (6e400002-b5a3-f393-e0a9-e50e24dcca9e)
V/RxBle#ServiceDiscoveryOperation(14171): 	  Properties: [ WRITE_NO_RESPONSE WRITE ]
V/RxBle#ServiceDiscoveryOperation(14171): 	* Unknown characteristic (6e400003-b5a3-f393-e0a9-e50e24dcca9e)
V/RxBle#ServiceDiscoveryOperation(14171): 	  Properties: [ NOTIFY ]
V/RxBle#ServiceDiscoveryOperation(14171): 	  -> Descriptors: 
V/RxBle#ServiceDiscoveryOperation(14171): 		* Client Characteristic Configuration (00002902-0000-1000-8000-00805f9b34fb)
V/RxBle#ServiceDiscoveryOperation(14171): 
V/RxBle#ServiceDiscoveryOperation(14171): Primary Service - Unknown service (61353090-8231-49cc-b57a-886370740041)
V/RxBle#ServiceDiscoveryOperation(14171): Instance ID: 29
V/RxBle#ServiceDiscoveryOperation(14171): -> Characteristics:
V/RxBle#ServiceDiscoveryOperation(14171): 	* Unknown characteristic (34802252-7185-4d5d-b431-630e7050e8f0)
V/RxBle#ServiceDiscoveryOperation(14171): 	  Properties: [ NOTIFY ]
V/RxBle#ServiceDiscoveryOperation(14171): 	  -> Descriptors: 
V/RxBle#ServiceDiscoveryOperation(14171): 		* Client Characteristic Configuration (00002902-0000-1000-8000-00805f9b34fb)
V/RxBle#ServiceDiscoveryOperation(14171): 	* Unknown characteristic (17816557-5652-417f-909f-3aee61e5fa85)
V/RxBle#ServiceDiscoveryOperation(14171): 	  Properties: [ WRITE_NO_RESPONSE WRITE ]
V/RxBle#ServiceDiscoveryOperation(14171): 	  -> Descriptors: 
V/RxBle#ServiceDiscoveryOperation(14171): 		* Characteristic Extended Properties (00002900-0000-1000-8000-00805f9b34fb)
V/RxBle#ServiceDiscoveryOperation(14171): 
V/RxBle#ServiceDiscoveryOperation(14171): Primary Service - User Data (0000181c-0000-1000-8000-00805f9b34fb)
V/RxBle#ServiceDiscoveryOperation(14171): Instance ID: 36
V/RxBle#ServiceDiscoveryOperation(14171): -> Characteristics:
V/RxBle#ServiceDiscoveryOperation(14171): 	* Serial Number String (00002a25-0000-1000-8000-00805f9b34fb)
V/RxBle#ServiceDiscoveryOperation(14171): 	  Properties: [ WRITE ]
V/RxBle#ServiceDiscoveryOperation(14171): 	  -> Descriptors: 
V/RxBle#ServiceDiscoveryOperation(14171): 		* Characteristic Extended Properties (00002900-0000-1000-8000-00805f9b34fb)
V/RxBle#ServiceDiscoveryOperation(14171): --------------- ====== Finished peripheral content ====== ---------------
D/RxBle#ConnectionOperationQueue(14171): FINISHED ServiceDiscoveryOperation(40802563) in 51 ms

D/com.polidea.flutter_ble_lib.FlutterBleLibPlugin(14171): on native side observed method: monitorCharacteristicForDevice
D/BluetoothGatt(14171): setCharacteristicNotification() - uuid: 6e400003-b5a3-f393-e0a9-e50e24dcca9e enable: true

D/RxBle#ConnectionOperationQueue(14171): QUEUED   DescriptorWriteOperation(92667227)
D/RxBle#ConnectionOperationQueue(14171): STARTED  DescriptorWriteOperation(92667227)
D/RxBle#BluetoothGatt$1(14171): onDescriptorWrite descriptor=00002902-0000-1000-8000-00805f9b34fb status=0

D/com.polidea.flutter_ble_lib.FlutterBleLibPlugin(14171): on native side observed method: writeCharacteristicForDevice
W/RxBle#IllegalOperationChecker(14171): Characteristic 00002a25-0000-1000-8000-00805f9b34fb supports properties: [ READ ] (2) does not have any property matching [ WRITE_NO_RESPONSE WRITE SIGNED_WRITE ] (76)
D/RxBle#ConnectionOperationQueue(14171): QUEUED   CharacteristicWriteOperation(112848188)
D/RxBle#ConnectionOperationQueue(14171): STARTED  CharacteristicWriteOperation(112848188)
D/RxBle#ConnectionOperationQueue(14171): FINISHED CharacteristicWriteOperation(112848188) in 24 ms
I/flutter (14171): 2020-06-10T11:55:15.431690	E	new DeviceDetailsBloc.<ac>:	 BleError caught: 4 GATT exception from MAC address 0C:8C:DC:2E:8B:EA, with type BleGattOperation{description='CHARACTERISTIC_WRITE'}

mikolak, 2020-06-10:
Hi!

There are two characteristics with the same uuid.
```
V/RxBle#ServiceDiscoveryOperation(14171): Primary Service - Device Information (0000180a-0000-1000-8000-00805f9b34fb)
V/RxBle#ServiceDiscoveryOperation(14171): Instance ID: 14
V/RxBle#ServiceDiscoveryOperation(14171): -> Characteristics:
V/RxBle#ServiceDiscoveryOperation(14171): 	* Manufacturer Name String (00002a29-0000-1000-8000-00805f9b34fb)
V/RxBle#ServiceDiscoveryOperation(14171): 	  Properties: [ READ ]
V/RxBle#ServiceDiscoveryOperation(14171): 	* Serial Number String (00002a25-0000-1000-8000-00805f9b34fb)
V/RxBle#ServiceDiscoveryOperation(14171): 	  Properties: [ READ ]
...
V/RxBle#ServiceDiscoveryOperation(14171): Primary Service - User Data (0000181c-0000-1000-8000-00805f9b34fb)
V/RxBle#ServiceDiscoveryOperation(14171): Instance ID: 36
V/RxBle#ServiceDiscoveryOperation(14171): -> Characteristics:
V/RxBle#ServiceDiscoveryOperation(14171): 	* Serial Number String (00002a25-0000-1000-8000-00805f9b34fb)
V/RxBle#ServiceDiscoveryOperation(14171): 	  Properties: [ WRITE ]
V/RxBle#ServiceDiscoveryOperation(14171): 	  -> Descriptors: 
V/RxBle#ServiceDiscoveryOperation(14171): 		* Characteristic Extended Properties (00002900-0000-1000-8000-00805f9b34fb)
```

How does your writing look like?

sandeepcmsm, 2020-06-10:
Thanks  mikolak,
Just for clarification, does characteristic UUID have to be different even they have different service UUID? 
In the API, the write methods at peripheral, service, characteristic at the end passing all the UUIDs related to peripheral, service, characteristic.

mikolak, 2020-06-10:
We had an issue with matching characteristics by UUID instead of gattCharacteristic objects, which is fixed in the underlying library, but the version hasn't been bumped here yet.
I think that's the cause of your issue. I'll try to release it today or next week.

To be honest I don't know if characteristic UUIDs have to be unique in the peripheral-scope, though name (universally unique identifier) would imply so. It's not enforced in any way though.

sandeepcmsm, 2020-06-10:
Thanks for the clarification. Il verify and close it on next release.

sandeepcmsm, 2020-06-11:
@mikolak its working fine now.

closed 2020-06-11 by sandeepcmsm