dotintent/FlutterBleLib #477: the manufacturerData of AdvertisementData  is not correct

philos3, 2020-06-18:
{localName: , txPowerLevel: null, connectable: false, manufacturerData: {6: [1, 9, 32, 2, 192, 182, 201, 73, 53, 201, 247, 233, 190, 35, 229, 229, 221, 22, 151, 237, 98, 130, 137, 138, 200, 224, 77]}, serviceData: {}, serviceUuids: []}
==========================================

AdvertisementData as above,the manufacturerData is {6: [1, 9, 32, 2, 192, 182, 201, 73, 53, 201, 247, 233, 190, 35, 229, 229, 221, 22, 151, 237, 98, 130, 137, 138, 200, 224, 77]}
but scanResult.advertisementData.manufacturerData.toString() is a list :  [1, 9, 32, 2, 192, 182, 201, 73, 53, 201, 247, 233, 190, 35, 229, 229, 221, 22, 151, 237, 98, 130, 137, 138, 200, 224, 77]

mikolak, 2020-06-18:
That `{6:` seems to be an anomaly. What platform is this?
`manufacturerData` on Android is just a byte array and it's the same in Dart.
https://github.com/Polidea/MultiPlatformBleAdapter/blob/master/android/library/src/main/java/com/polidea/multiplatformbleadapter/AdvertisementData.java#L14
https://github.com/Polidea/FlutterBleLib/blob/develop/lib/scan_result.dart#L45

philos3, 2020-07-28:
> That `{6:` seems to be an anomaly. What platform is this?
> `manufacturerData` on Android is just a byte array and it's the same in Dart. [...]

IOS