dotintent/FlutterBleLib #478: [iOS] Connect to device doesn't seem to respect timeout

cbreezier, 2020-06-19:
```
      try {
        await device.connect(timeout: Duration(seconds: 3));
      } catch (e) {
        debugPrint('Failed to connect to device: $e');
        Util.showError(context, 'Failed to connect to device');

        return await onError.call(device, e);
      }
```

`device` is a `Peripheral` object. This dart code works fine on Android, where a connection attempt is cancelled after 3 seconds and an exception is thrown (and caught).

However, the same code doesn't seem to do anything on iOS. The connection attempt seems to last forever.

Is the `timeout` parameter supported on iOS?

mikolak, 2020-06-19:
It should work the same way.
https://github.com/Polidea/MultiPlatformBleAdapter/blob/master/iOS/classes/BleModule.swift#L400

dokinkon, 2020-07-17:
Same issue with version 2.2.6

jaromeyer, 2020-07-29:
i have the same issue

jaromeyer, 2020-08-05:
As a workaround you can use flutters timeout method:
`await device.connect().timeout(Duration(seconds: 10));`

If you have some issues with scanning or connecting to the device once it's available/in range again, also add this hack:
```
try {
   await device.disconnectOrCancelConnection();
} catch (e) {}
```

HerrNiklasRaab, 2020-11-03:
@mikolak Any upates?

mikolak, 2020-11-09:
No news, sorry. I'll try to fix it this month.

mikolak, 2020-11-16:
Almost a half year later, but the issue will be fixed in the nearest release, probably by the end of the week.

PR #546 (2020-11-16): Fix connection timeout on iOS
Fixes #478

iOS wasn't remapping `timeoutMillis` to `timeout` (left for legacy reasons due to react-native-ble-plx compatibility). Since I'm no good with obj-c it was easier to just rename the parameter.
commit dc3f8f50 (2020-11-16): Fix connection timeout on iOS (#546)

Fixes #478
commit cab2dcc5 (2020-11-19): Release v2.3.1 (#548)

* Minor doc fixes (#544)

* Regenerate podfile for example (#545)

* Fix connection timeout on iOS (#546)

Fixes #478

* Fix emitting current connection state on iOS (#547)

Fixes #540

* Release 2.3.1
closed 2020-11-16 by mikolak