dotintent/FlutterBleLib #479: Official Example crash when enter detail router the second time

debugerer, 2020-06-20:
I downloaded the latest zip file, open the example project in example folder it worked fine on my iphone7+ with iOS 13.4.1.
And I can click a item in device list. it will enter the detail page, auto test and manually test is both OK.
Then i return the device list page.
When i click a item again, it crash with following logs:

2020-06-20 10:16:06.550399+0800 Runner[10894:1933409] flutter: 2020-06-20T10:16:06.550312	D	DevicesList._createTapListener.<ac>:	 clicked device :MacBook Pro
2020-06-20 10:16:06.550741+0800 Runner[10894:1933409] flutter: Another exception was thrown: Bad state: Cannot add event after closing.

I trace the code is here:
  static DeviceTapListener _createTapListener(
      DevicesBloc devicesBloc, BleDevice bleDevice) {
    return () {
      Fimber.d("clicked device: ${bleDevice.name}");
      **_devicesBloc.devicePicker.add(bleDevice);_**  //crash here
    };
  }

This is a reproducible problem,
I researched for a long time and didn’t know how to fix it. For it is a official Example, Is there any solution? Thanks very much.

mikolak, 2020-06-20:
Hi!

This is caused by closing a stream and adding an event after it is closed. The example should be simplified. I'll try to take a look at it next week.

debugerer, 2020-06-20:
@mikolak, OK, Thank you very much!

debugerer, 2020-06-20:
> Hi!
> 
> This is caused by closing a stream and adding an event after it is closed. The example should be simplified. I'll try to take a look at it next week.

@mikolak, OK, Thank you very much!
I also think the current example which use the bloc framework is too complicated for novices.
BTW, I just want to implement a simple scan result list, click on each item to enter a new router show the simple info of the device.
I have been researching this library for three days, the current example is good, but I can't to fix this crash . Is there a simple example for reference now? Or any the code flow for above function is all appreciated, Thank you.

mikolak, 2020-06-20:
Can you take a look at [BLEmulator's example](https://github.com/Polidea/blemulator_flutter)?
Before you run it flip this flag to true: https://github.com/Polidea/blemulator_flutter/blob/master/example/lib/main.dart#L23

mikolak, 2020-06-20:
`manager.startScan()` returns a stream of received scan results, so you'll need to accumulate them somewhere (and keep only unique devices). Once you have that accumulated list, you can do whatever you want with it.

`BleManager` is a singleton right now, so you don't have to pass it around and can just call `BleManager()` everywhere, though to only show info from the scan result, you don't even need that.

debugerer, 2020-06-20:
@mikolak, Thank you very much for your quick response!  I will reference it and try.

PR #484 (2020-06-23): Fix example
Fixes #479
closed 2020-06-23 by mikolak