dotintent/FlutterBleLib #480: Sometimes startPeripheralScan doesn't return any result

fpabl0, 2020-06-20:
Hi!
I am using your library for a personal project. The problem I have is that when I try to scan many times, the function startPeripheralScan doesn't return anything, or some devices aren't shown. Here is my code:

`   
  void startScan() {
      _devicesList.clear();
      _devicesSink.add(null);
      _scanSubscription = _bleManager
         .startPeripheralScan(scanMode: ScanMode.balanced, allowDuplicates: false)
         .listen((device) {
             // exclude duplicated devices
             for (var d in _devicesList) {
                 if (device.peripheral.identifier == d.peripheral.identifier) return;
             }
             _devicesList.add(device);
            _devicesSink.add(_devicesList.sublist(0));
      });
      _timeoutTimer = new Timer(const Duration(seconds: 5), () => stopScan());
   }

   // Stop BLE devices scan
  Future<void> stopScan() async {
     _timeoutTimer?.cancel();
     if (_devicesList.length == 0) {
       _devicesSink.add([]);
     }
     await _scanSubscription?.cancel();
     await _bleManager?.stopPeripheralScan();
  }
`

mikolak, 2020-06-21:
Hi!

Could you set logs to verbose, run the application from native IDE (XCode or Android Studio, but Android project, not Flutter) and check the logs? Could you paste them if that didn't give you an idea as to what happened?

fpabl0, 2020-06-21:
Yes, ok this is log when it discovers all the devices:

```
2020-06-21 10:04:01.311 10474-10474/com.example.otter_feeder D/com.polidea.flutter_ble_lib.FlutterBleLibPlugin: on native side observed method: startDeviceScan
2020-06-21 10:04:01.337 10474-10789/com.example.otter_feeder D/BluetoothAdapter: isLeEnabled(): ON
2020-06-21 10:04:01.337 10474-10789/com.example.otter_feeder E/CheckPermission: _bluetooth code = 10 
2020-06-21 10:04:01.344 2600-3048/? I/bt_stack: [INFO:gatt_api.cc(1006)] GATT_Register
2020-06-21 10:04:01.344 2600-3048/? I/bt_stack: [INFO:gatt_api.cc(1029)] allocated gatt_if=5
2020-06-21 10:04:01.348 10474-10496/com.example.otter_feeder D/BluetoothLeScanner: onScannerRegistered() - status=0 scannerId=5 mScannerId=0
2020-06-21 10:04:01.354 2600-3016/? D/BtGatt.CustomScanStrategy: checkAndNofifyScanManager bConcession =false
2020-06-21 10:04:01.354 2600-3016/? D/BtGatt.CustomScanStrategy: bOshareTransfering = false bGameSpace = false bWIFIHeavyTraffic = false
2020-06-21 10:04:01.364 639-3045/? D/vendor.qti.bluetooth@1.0-bluetooth_hci: BluetoothHci::event call back-->
2020-06-21 10:04:01.364 2600-3024/? D/bt_hci: hciEventReceived-->
2020-06-21 10:04:01.364 2600-3024/? D/bt_hci: hciEventReceived<--
2020-06-21 10:04:01.365 639-3045/? D/vendor.qti.bluetooth@1.0-bluetooth_hci: BluetoothHci::event call back<--
2020-06-21 10:04:01.366 639-3045/? D/vendor.qti.bluetooth@1.0-bluetooth_hci: BluetoothHci::event call back-->
2020-06-21 10:04:01.367 2600-3024/? D/bt_hci: hciEventReceived-->
2020-06-21 10:04:01.367 2600-3024/? D/bt_hci: hciEventReceived<--
2020-06-21 10:04:01.367 2600-3016/? D/BtGatt.CustomScanStrategy: CustomScanStrategy startWatchWIFITraffic need reduce for wifi heavy traffic
2020-06-21 10:04:01.367 2600-3016/? D/BtGatt.CustomScanStrategy: MonitorThread wlan0
2020-06-21 10:04:01.368 639-3045/? D/vendor.qti.bluetooth@1.0-bluetooth_hci: BluetoothHci::event call back<--
2020-06-21 10:04:01.368 2600-3016/? D/BtGatt.ScanManager: startWatchWIFITraffic
2020-06-21 10:04:01.368 2600-3016/? D/BtGatt.CustomScanStrategy: getCurrAction: 0
2020-06-21 10:04:01.369 639-3045/? D/vendor.qti.bluetooth@1.0-bluetooth_hci: BluetoothHci::event call back-->
2020-06-21 10:04:01.369 2600-3024/? D/bt_hci: hciEventReceived-->
2020-06-21 10:04:01.370 2600-3024/? D/bt_hci: hciEventReceived<--
2020-06-21 10:04:01.370 639-3045/? D/vendor.qti.bluetooth@1.0-bluetooth_hci: BluetoothHci::event call back<--
2020-06-21 10:04:01.372 639-3045/? D/vendor.qti.bluetooth@1.0-bluetooth_hci: BluetoothHci::event call back-->
2020-06-21 10:04:01.373 2600-3024/? D/bt_hci: hciEventReceived-->
2020-06-21 10:04:01.373 2600-3024/? D/bt_hci: hciEventReceived<--
2020-06-21 10:04:01.373 639-3045/? D/vendor.qti.bluetooth@1.0-bluetooth_hci: BluetoothHci::event call back<--
2020-06-21 10:04:01.375 639-3045/? D/vendor.qti.bluetooth@1.0-bluetooth_hci: BluetoothHci::event call back-->
2020-06-21 10:04:01.376 2600-3024/? D/bt_hci: hciEventReceived-->
2020-06-21 10:04:01.376 2600-3024/? D/bt_hci: hciEventReceived<--
2020-06-21 10:04:01.376 639-3045/? D/vendor.qti.bluetooth@1.0-bluetooth_hci: BluetoothHci::event call back<--
2020-06-21 10:04:01.377 639-3045/? D/vendor.qti.bluetooth@1.0-bluetooth_hci: BluetoothHci::event call back-->
2020-06-21 10:04:01.377 2600-3024/? D/bt_hci: hciEventReceived-->
2020-06-21 10:04:01.378 2600-3024/? D/bt_hci: hciEventReceived<--
2020-06-21 10:04:01.378 639-3045/? D/vendor.qti.bluetooth@1.0-bluetooth_hci: BluetoothHci::event call back<--
2020-06-21 10:04:01.379 639-3045/? D/vendor.qti.bluetooth@1.0-bluetooth_hci: BluetoothHci::event call back-->
2020-06-21 10:04:01.379 2600-3024/? D/bt_hci: hciEventReceived-->
2020-06-21 10:04:01.380 2600-3024/? D/bt_hci: hciEventReceived<--
2020-06-21 10:04:01.380 639-3045/? D/vendor.qti.bluetooth@1.0-bluetooth_hci: BluetoothHci::event call back<--
2020-06-21 10:04:01.384 639-3045/? D/vendor.qti.bluetooth@1.0-bluetooth_hci: BluetoothHci::event call back-->
2020-06-21 10:04:01.384 2600-3024/? D/bt_hci: hciEventReceived-->
2020-06-21 10:04:01.384 2600-3024/? D/bt_hci: hciEventReceived<--
2020-06-21 10:04:01.384 639-3045/? D/vendor.qti.bluetooth@1.0-bluetooth_hci: BluetoothHci::event call back<--
2020-06-21 10:04:01.399 639-3045/? D/vendor.qti.bluetooth@1.0-bluetooth_hci: BluetoothHci::event call back-->
2020-06-21 10:04:01.399 2600-3024/? D/bt_hci: hciEventReceived-->
2020-06-21 10:04:01.399 2600-3024/? D/bt_hci: hciEventReceived<--
2020-06-21 10:04:01.399 639-3045/? D/vendor.qti.bluetooth@1.0-bluetooth_hci: BluetoothHci::event call back<--
2020-06-21 10:04:01.399 639-3045/? D/vendor.qti.bluetooth@1.0-bluetooth_hci: BluetoothHci::event call back-->
2020-06-21 10:04:01.400 2600-3024/? D/bt_hci: hciEventReceived-->
2020-06-21 10:04:01.400 2600-3024/? D/bt_hci: hciEventReceived<--
2020-06-21 10:04:01.400 639-3045/? D/vendor.qti.bluetooth@1.0-bluetooth_hci: BluetoothHci::event call back<--
2020-06-21 10:04:01.415 2600-2616/? D/A2dpService: getA2DPService(): returning com.android.bluetooth.a2dp.A2dpService@a9680c6
2020-06-21 10:04:01.418 2600-2616/? D/A2dpService: getA2DPService(): returning com.android.bluetooth.a2dp.A2dpService@a9680c6
2020-06-21 10:04:01.580 639-3045/? D/vendor.qti.bluetooth@1.0-bluetooth_hci: BluetoothHci::event call back-->
2020-06-21 10:04:01.581 2600-3024/? D/bt_hci: hciEventReceived-->
2020-06-21 10:04:01.581 2600-3024/? D/bt_hci: hciEventReceived<--
[... cut ...]
2020-06-21 10:11:31.264 2638-3311/? D/TrafficStatsHelper: transmitted_validWifiData.get(0) = wlan0
2020-06-21 10:11:32.148 10474-10474/com.example.otter_feeder D/com.polidea.flutter_ble_lib.FlutterBleLibPlugin: on native side observed method: stopDeviceScan
2020-06-21 10:11:35.264 2638-3311/? D/TrafficStatsHelper: transmitted_validWifiData.get(0) = wlan0
2020-06-21 10:11:37.200 2600-2600/? D/HeadsetPhoneState: Enter onSignalStrengthsChanged
2020-06-21 10:11:37.200 2600-2600/? D/HeadsetPhoneState: Exit onSignalStrengthsChanged
```

I'm using your last library version and Flutter 1.17.4.

mikolak, 2020-06-21:
Did you do `BleManager().setLogLevel(LogLevel.verbose)`? (something like that, don't remember the API right now)

fpabl0, 2020-06-21:
Yes, here:

```
  final _bleManager = new BleManager();
  ...
  // Init bluetooth
  Future<void> _init() async {
      _bleManager.setLogLevel(LogLevel.verbose);
      await _bleManager.createClient();
      await _checkPermissions();
      await _waitForBluetoothPoweredOn();
  }
```

mikolak, 2020-06-21:
It has to be called after `.createClient()`. Can you change the order of calls and take a look at the logs again?

fpabl0, 2020-06-21:
Yes, I can. I got this log when the scan returns three devices instead of four expected:

```
2020-06-21 15:19:33.954 2638-3311/? D/TrafficStatsHelper: transmitted_validWifiData.get(0) = wlan0
2020-06-21 15:19:35.200 23449-23449/com.example.otter_feeder D/com.polidea.flutter_ble_lib.FlutterBleLibPlugin: on native side observed method: startDeviceScan
2020-06-21 15:19:35.208 23449-23449/com.example.otter_feeder D/RxBle#ClientOperationQueue: QUEUED   ScanOperationApi21(116670761)
2020-06-21 15:19:35.211 23449-23603/com.example.otter_feeder D/RxBle#ClientOperationQueue: STARTED  ScanOperationApi21(116670761)
2020-06-21 15:19:35.213 23449-23615/com.example.otter_feeder I/RxBle#QueueOperation: Scan operation is requested to start.
2020-06-21 15:19:35.215 23449-23615/com.example.otter_feeder D/BluetoothAdapter: isLeEnabled(): ON
2020-06-21 15:19:35.215 23449-23615/com.example.otter_feeder E/CheckPermission: _bluetooth code = 10 
2020-06-21 15:19:35.219 2600-3048/? I/bt_stack: [INFO:gatt_api.cc(1006)] GATT_Register
2020-06-21 15:19:35.220 2600-3048/? I/bt_stack: [INFO:gatt_api.cc(1029)] allocated gatt_if=5
2020-06-21 15:19:35.220 23449-23626/com.example.otter_feeder D/BluetoothLeScanner: onScannerRegistered() - status=0 scannerId=5 mScannerId=0
2020-06-21 15:19:35.228 2600-3016/? D/BtGatt.CustomScanStrategy: checkAndNofifyScanManager bConcession =false
2020-06-21 15:19:35.228 2600-3016/? D/BtGatt.CustomScanStrategy: bOshareTransfering = false bGameSpace = false bWIFIHeavyTraffic = false
2020-06-21 15:19:35.228 23449-23603/com.example.otter_feeder D/RxBle#ClientOperationQueue: FINISHED ScanOperationApi21(116670761) in 18 ms
2020-06-21 15:19:35.235 639-3045/? D/vendor.qti.bluetooth@1.0-bluetooth_hci: BluetoothHci::event call back-->
2020-06-21 15:19:35.235 2600-3024/? D/bt_hci: hciEventReceived-->
2020-06-21 15:19:35.235 2600-3024/? D/bt_hci: hciEventReceived<--
2020-06-21 15:19:35.236 639-3045/? D/vendor.qti.bluetooth@1.0-bluetooth_hci: BluetoothHci::event call back<--
2020-06-21 15:19:35.240 639-3045/? D/vendor.qti.bluetooth@1.0-bluetooth_hci: BluetoothHci::event call back-->
2020-06-21 15:19:35.240 2600-3024/? D/bt_hci: hciEventReceived-->
2020-06-21 15:19:35.240 2600-3024/? D/bt_hci: hciEventReceived<--
2020-06-21 15:19:35.240 2600-3016/? D/BtGatt.CustomScanStrategy: CustomScanStrategy startWatchWIFITraffic need reduce for wifi heavy traffic
2020-06-21 15:19:35.241 2600-3016/? D/BtGatt.CustomScanStrategy: MonitorThread wlan0
2020-06-21 15:19:35.241 639-3045/? D/vendor.qti.bluetooth@1.0-bluetooth_hci: BluetoothHci::event call back<--
2020-06-21 15:19:35.241 2600-3016/? D/BtGatt.ScanManager: startWatchWIFITraffic
2020-06-21 15:19:35.241 2600-3016/? D/BtGatt.CustomScanStrategy: getCurrAction: 0
2020-06-21 15:19:35.249 639-3045/? D/vendor.qti.bluetooth@1.0-bluetooth_hci: BluetoothHci::event call back-->
2020-06-21 15:19:35.249 2600-3024/? D/bt_hci: hciEventReceived-->
2020-06-21 15:19:35.249 2600-3024/? D/bt_hci: hciEventReceived<--
2020-06-21 15:19:35.250 639-3045/? D/vendor.qti.bluetooth@1.0-bluetooth_hci: BluetoothHci::event call back<--
2020-06-21 15:19:35.252 639-3045/? D/vendor.qti.bluetooth@1.0-bluetooth_hci: BluetoothHci::event call back-->
2020-06-21 15:19:35.252 2600-3024/? D/bt_hci: hciEventReceived-->
2020-06-21 15:19:35.252 2600-3024/? D/bt_hci: hciEventReceived<--
2020-06-21 15:19:35.252 639-3045/? D/vendor.qti.bluetooth@1.0-bluetooth_hci: BluetoothHci::event call back<--
2020-06-21 15:19:35.261 639-3045/? D/vendor.qti.bluetooth@1.0-bluetooth_hci: BluetoothHci::event call back-->
2020-06-21 15:19:35.261 2600-3024/? D/bt_hci: hciEventReceived-->
2020-06-21 15:19:35.261 2600-3024/? D/bt_hci: hciEventReceived<--
2020-06-21 15:19:35.261 639-3045/? D/vendor.qti.bluetooth@1.0-bluetooth_hci: BluetoothHci::event call back<--
2020-06-21 15:19:35.262 639-3045/? D/vendor.qti.bluetooth@1.0-bluetooth_hci: BluetoothHci::event call back-->
2020-06-21 15:19:35.263 2600-3024/? D/bt_hci: hciEventReceived-->
2020-06-21 15:19:35.263 2600-3024/? D/bt_hci: hciEventReceived<--
2020-06-21 15:19:35.263 639-3045/? D/vendor.qti.bluetooth@1.0-bluetooth_hci: BluetoothHci::event call back<--
2020-06-21 15:19:35.263 639-3045/? D/vendor.qti.bluetooth@1.0-bluetooth_hci: BluetoothHci::event call back-->
2020-06-21 15:19:35.263 2600-3024/? D/bt_hci: hciEventReceived-->
2020-06-21 15:19:35.264 2600-3024/? D/bt_hci: hciEventReceived<--
2020-06-21 15:19:35.264 639-3045/? D/vendor.qti.bluetooth@1.0-bluetooth_hci: BluetoothHci::event call back<--
2020-06-21 15:19:35.264 639-3045/? D/vendor.qti.bluetooth@1.0-bluetooth_hci: BluetoothHci::event call back-->
2020-06-21 15:19:35.265 2600-3024/? D/bt_hci: hciEventReceived-->
2020-06-21 15:19:35.265 2600-3024/? D/bt_hci: hciEventReceived<--
2020-06-21 15:19:35.265 639-3045/? D/vendor.qti.bluetooth@1.0-bluetooth_hci: BluetoothHci::event call back<--
2020-06-21 15:19:35.970 639-3045/? D/vendor.qti.bluetooth@1.0-bluetooth_hci: BluetoothHci::event call back-->
2020-06-21 15:19:35.971 2600-3024/? D/bt_hci: hciEventReceived-->
2020-06-21 15:19:35.971 2600-3024/? D/bt_hci: hciEventReceived<--
2020-06-21 15:19:35.971 639-3045/? D/vendor.qti.bluetooth@1.0-bluetooth_hci: BluetoothHci::event call back<--
2020-06-21 15:19:35.971 639-3045/? D/vendor.qti.bluetooth@1.0-bluetooth_hci: BluetoothHci::event call back-->
2020-06-21 15:19:35.972 2600-3024/? D/bt_hci: hciEventReceived-->
2020-06-21 15:19:35.972 2600-3024/? D/bt_hci: hciEventReceived<--
[... cut ...]
2020-06-21 15:25:59.813 2600-2600/? D/HeadsetPhoneState: Exit onSignalStrengthsChanged
2020-06-21 15:26:02.012 2638-3311/? D/TrafficStatsHelper: transmitted_validWifiData.get(0) = wlan0
2020-06-21 15:26:03.174 1919-2447/? D/NetworkPolicy: uid 10138 is allowed on unmetered network
2020-06-21 15:26:03.176 1919-2447/? I/chatty: uid=1000(system) ConnectivityThr identical 12 lines
2020-06-21 15:26:03.176 1919-2447/? D/NetworkPolicy: uid 10138 is allowed on unmetered network
2020-06-21 15:26:03.176 1919-2447/? D/NetworkPolicy: uid 10140 is allowed on unmetered network
```

mikolak, 2020-06-21:
It seems as if native client was not there. Are you using `destroyClient()` between the calls?

fpabl0, 2020-06-21:
No, I am using the Bloc pattern and the only place I use it is in a dispose method:

```
  void dispose() async {
    await stopScan();
    await _bleManager?.destroyClient();
    await _scanSubscription?.cancel();
    await _connStateSubscription?.cancel();
  }
```
Then this method is called here: (I am using Provider lib)

```
@override
  Widget build(BuildContext context) {
    return MultiProvider(
      providers: [
        Provider<FeederBloc>(
          create: (_) => new FeederBloc(),
          dispose: (_, bloc) => bloc.dispose(),
        ),
       ....
```

mikolak, 2020-06-22:
Try setting a breakpoint on `destroyClient()` or somewhere in FlutterBleLibPlugin class to check if the adapter is set. Since no operation is queued in your other log, you might either be calling when there's no adapter or the adapter's too busy, ie. you're changing radio state too fast.

fpabl0, 2020-06-22:
Hmm ok thanks, changing the radio state too fast, maybe explain why sometimes I don't get any results. However what about the other problem? Sometimes the function does not show all the available devices.

mikolak, 2020-06-22:
Perhaps the advertisement simply didn't come? 5 seconds isn't a lot. Try adding a two second delay between stopping and restarting the scan and lengthen the scanning period to 15 seconds.

Can you add a log to your dispose method to see if it's not called when it shouldn't be?

fpabl0, 2020-06-22:
Thanks for the suggestion, I will take into account to add a delay between stopping and restarting the scan. On the other hand, I don't think 5 seconds is a lot. This application I am doing is a refactor of a previous application I did with Ionic framework, and in that case 5 seconds were enough to identify all the near devices.

I have debugged with a breakpoint on my dispose method, and it is never called, as it just have to be called at the end of the app.

dariuszseweryn, 2020-06-22:
5 seconds may or may not be a lot. If you cannot scan all devices you expect you can change scan mode to more aggressive.
As for the second case where you get no results – are all scans made on the same instance of the manager? Or the managers are never disposed but new ones are being created with each app reload?

fpabl0, 2020-06-22:
Hi @dariuszseweryn, yes that's why I changed the default LowPower mode to Balanced mode, I just wonder why I can get all the devices in LowPower mode with the other app.
Scans are made on the same instance of the manager, the instance is created when the bloc is created, and it is created just once when the app starts, and then it is disposed when the app finishes.

dariuszseweryn, 2020-06-22:
So I assume you do not try to start more than 5 scans in any 30 second window. I also assume you do not try to run the scan in the background. Am I correct?
As for the performance — I cannot really tell why you experience different results with different apps. I cannot find mode changes in the [readme](https://github.com/don/cordova-plugin-ble-central#parameters-2) of the plugin that is apparently used by [Ionic](https://ionicframework.com/docs/native/ble).

fpabl0, 2020-06-22:
Do you mean simultaneously? If so, then I am just one scan at time. If you do not mean simultaneously, then yes, it is possible to start more than five scans in a 30 second window, because I let the user to start or stop the scans but with a default timeout of 5 seconds (if they don't press the stop button).
In the case of Ionic, they don't have scan mode changes, because they are using deprecated android api, but according to the documentation ScanLowPowerMode is the the default one, so I assume that they used it, but honestly I am not sure.
Finally, don't get me wrong I'm not saying that Ionic plugin is better than this one, actually it isn't, that's why I migrated to flutter, your library is by far more performant than Ionic one (and even better than others in flutter), I just wanted to understand why these differences at the scanning time.

dariuszseweryn, 2020-06-22:
> In the case of Ionic, they don't have scan mode changes, because they are using deprecated android api, but according to the documentation ScanLowPowerMode is the the default one, so I assume that they used it, but honestly I am not sure.

Legacy API used aggressive scan mode.

> Do you mean simultaneously? If so, then I am just one scan at time. If you do not mean simultaneously, then yes, it is possible to start more than five scans in a 30 second window, because I let the [...]

Not necessarily simultaneously. Serial calls may trigger throttling as well. Thing is that the library should cover that situation if the manager is not being recreated in the mean time.

Do you have any specific recreation scenario under which the results are not returned? The more repeatable the better.

fpabl0, 2020-06-23:
> Legacy API used aggressive scan mode.

Oh sorry, I didn't know it.

> Do you have any specific recreation scenario under which the results are not returned? The more repeatable the better.

I have created a basic project based on the one I am developing in order to give you a way to test:
https://github.com/fpabl0/flutter_ble_test_tmp
You only have to press Scan button in an interval of 2 or 5 seconds (previously stopped) and probably you will get "Could not find near devices".

dariuszseweryn, 2020-06-24:
Do you have any error handling there? I can't see it

fpabl0, 2020-06-24:
I'm just handling the error in _init() function, then I send it to state stream:

```
  // Init bluetooth
  Future<void> _init() async {
    try {
      _stateSink.add(DeviceBlocState.INITIALIZING);
      await _bleManager.createClient();
      await _bleManager.setLogLevel(LogLevel.verbose);
      await _checkPermissions();
      await _waitForBluetoothPoweredOn();
      _stateSink.add(DeviceBlocState.IDLE);
    } catch (e) {
      _stateSink.addError("Error initializing bluetooth: ${e.toString()}");
    }
  }
```
I'm not handling errors in startScan and stopScan function, because startScan returns a stream, and an error in stopScan, at least for me, is not very important to the final user.

dariuszseweryn, 2020-06-24:
From what I know the `Stream` may error out. I am not familiar with how it should be handled. I just see that it does not seem to be handled in the code you provided.

fpabl0, 2020-06-24:
You are right, it's my mistake. I have updated my repo handling stream errors.
In these two places.
1.
```
  // Wait Bluetooth on
  Future<void> _waitForBluetoothPoweredOn() async {
    Completer completer = Completer();
    StreamSubscription<BluetoothState> subscription;
    subscription = _bleManager.observeBluetoothState(emitCurrentValue: true).listen(
      (bluetoothState) async {
        if (bluetoothState == BluetoothState.POWERED_ON && !completer.isCompleted) {
          await subscription.cancel();
          completer.complete();
        }
      },
      // handling stream error
      onError: (error) {
        completer.completeError(error);
      },
      // automatically cancel the subscription on first error
      cancelOnError: true,
    );
    return completer.future;
  }
```
2.
```
  // Start BLE devices scan
  void startScan() {
    if (lastState != DeviceBlocState.IDLE) return;
    _stateSink.add(DeviceBlocState.SCANNING);
    _devicesList.clear();
    _devicesSink.add(null);

    _scanSubscription = _bleManager
        .startPeripheralScan(scanMode: ScanMode.balanced, allowDuplicates: false)
        .listen(
      (device) {
        for (var d in _devicesList) {
          if (device.peripheral.identifier == d.peripheral.identifier) return;
        }
        _devicesList.add(device);
        _devicesSink.add(_devicesList.sublist(0));
      },
      // handling stream error
      onError: (error) async {
        _stateSink.addError(error);
        await stopScan(); // stop scan here?
      },
      // automatically cancel the subscription on first error
      cancelOnError: true,
    );

    _timeoutTimer = new Timer(_TIMEOUT_DURATION, () => stopScan());
  }
```
However, none of them are trigger so the error is not there, and the problem that sometimes does not return any result still appears.

fati-chan, 2020-06-30:
Hi !
I don't know if this might help you but you can check it anyway ...
In my case, everything was working just fine but suddenly startPeripheralScan didn't return any results anymore and no errors were shown either ... after many tries I finally figured out that the location services on my Android device got disabled 😑 
so turning on the location services on my phone solved the problem.

dariuszseweryn, 2020-07-03:
Confirmed. The problem is that the native side is reporting an issue with scanning but the flutter side is not yet listening (race condition). Working on a fix.

PR #490 (2020-07-03): Fixed race condition in scan peripherals
Under certain circumstances a problem can be reported by the native side very quickly. Since monitoring for callbacks started after the call those responses could be delivered before anyone would start to listen.

Fixes #480
closed 2020-07-03 by mikolak