dotintent/FlutterBleLib #481: There is any way to check the scan state ？

philos3, 2020-06-22:
There is any way to check the scan state like :

//    if (_scanStateSub == null) {
//      _scanStateSub = FlutterBlue.instance.isScanning.listen((result) {
//        print("_scanStateSub isScanning：====================================$result");
//        if(mounted){
//          setState(() {
//            isScanning = result;
//          });
//        }
//        if(result){
//          Future.delayed(Duration.zero, () {
//            showLoadingDialog(myContext, "正在连接设备",true);
//          });
//        }
//      });
//    }

mikolak, 2020-06-22:
No. You can wrap subscribing to `startScan()` in a function that would switch your internal flag.

philos3, 2020-06-22:
> No. You can wrap subscribing to `startScan()` in a function that would switch your internal flag.

thanks for help

closed 2020-06-22 by philos3