dotintent/FlutterBleLib #482: characteristic.monitor().listen((readValue)  only  recieve one time ,how can i listen to the notify value of my device

philos3, 2020-06-22:
when the first time i connect  to my device ,it notifies me 
"Reading Characteristic value:  [18, 0, 8, 0, 185, 0, 0, 0, 0, 0, 0, 0]=-======================" 
it is ok ,but when i write  [7, 0, 0, 0] to my device ,it can't recieve any notification data,here is my code,please figured out the problem 

########################################
```
Service service = await bleDevice.peripheral.services().then((services) =>
          services.firstWhere((service) =>(service.uuid == "7e400001-b5a3-f393-e0a9-e50e24dcca9e")));
      List<Characteristic> characteristics = await service.characteristics();
      Characteristic characteristic = characteristics.firstWhere(
              (characteristic) =>
          characteristic.uuid =="7e400003-b5a3-f393-e0a9-e50e24dcca9e",orElse: () => null);
      if (characteristic == null) {
        print("Characteristic not found");
      } else {
        print(
          "Characteristic:\n"
              "uuid: ${characteristic.uuid}\n"
              "isReadable: ${characteristic.isReadable}\n"
              "isWritableWithoutResponse: ${characteristic.isWritableWithoutResponse}\n"
              "isWritableWithResponse: ${characteristic.isWritableWithResponse}\n"
              "isIndicatable: ${characteristic.isIndicatable}\n"
              "isNotifiable: ${characteristic.isNotifiable}\n",
        );
        characteristic.monitor().listen((readValue) {
          log("Reading Characteristic value:  ${readValue.toString()}=-======================");
});

bleDevice.peripheral.writeCharacteristic(
          "7e400001xxxxxxxxxxxx",
          "7e400003xxxxxxxxxxxx",
          Uint8List.fromList(value),// [7, 0, 0, 0]
          false,
//          transactionId:transactionId
      )
```

here is my part of logs:
#########################################################
D/com.polidea.flutter_ble_lib.FlutterBleLibPlugin( 9776): on native side observed method: writeCharacteristicForDevice
D/com.polidea.flutter_ble_lib.FlutterBleLibPlugin( 9776): on native side observed method: characteristicsForService
I/flutter ( 9776): Characteristic:
I/flutter ( 9776): uuid: 7e400003-b5a3-f393-e0a9-e50e24dcca9e
I/flutter ( 9776): isReadable: false
I/flutter ( 9776): isWritableWithoutResponse: false
I/flutter ( 9776): isWritableWithResponse: true
I/flutter ( 9776): isIndicatable: false
I/flutter ( 9776): isNotifiable: true
D/com.polidea.flutter_ble_lib.FlutterBleLibPlugin( 9776): on native side observed method: monitorCharacteristicForIdentifier
W/i.flutterbeaut( 9776): Accessing hidden method Lsun/misc/Unsafe;->putOrderedLong(Ljava/lang/Object;JJ)V (greylist, linking, allowed)
W/i.flutterbeaut( 9776): Accessing hidden method Lsun/misc/Unsafe;->putOrderedObject(Ljava/lang/Object;JLjava/lang/Object;)V (greylist, linking, allowed)
W/i.flutterbeaut( 9776): Accessing hidden method Lsun/misc/Unsafe;->putOrderedLong(Ljava/lang/Object;JJ)V (greylist, linking, allowed)
W/i.flutterbeaut( 9776): Accessing hidden method Lsun/misc/Unsafe;->putOrderedLong(Ljava/lang/Object;JJ)V (greylist, linking, allowed)
I/chatty  ( 9776): uid=10285(com.radi.flutterbeauty) identical 1 line

W/i.flutterbeaut( 9776): Accessing hidden method Lsun/misc/Unsafe;->putOrderedObject(Ljava/lang/Object;JLjava/lang/Object;)V (greylist, linking, allowed)
W/i.flutterbeaut( 9776): Accessing hidden method Lsun/misc/Unsafe;->putOrderedLong(Ljava/lang/Object;JJ)V (greylist, linking, allowed)
W/i.flutterbeaut( 9776): Accessing hidden method Lsun/misc/Unsafe;->putOrderedLong(Ljava/lang/Object;JJ)V (greylist, linking, allowed)
W/i.flutterbeaut( 9776): Accessing hidden method Lsun/misc/Unsafe;->putOrderedLong(Ljava/lang/Object;JJ)V (greylist, linking, allowed)
W/i.flutterbeaut( 9776): Accessing hidden method Lsun/misc/Unsafe;->putOrderedLong(Ljava/lang/Object;JJ)V (greylist, linking, allowed)
W/i.flutterbeaut( 9776): Accessing hidden method Lsun/misc/Unsafe;->putOrderedObject(Ljava/lang/Object;JLjava/lang/Object;)V (greylist, linking, allowed)
W/i.flutterbeaut( 9776): Accessing hidden method Lsun/misc/Unsafe;->putOrderedLong(Ljava/lang/Object;JJ)V (greylist, linking, allowed)
I/flutter ( 9776): Reading Characteristic value:  [18, 0, 8, 0, 185, 0, 0, 0, 0, 0, 0, 0]=-======================

mikolak, 2020-06-22:
I'm assuming those Xs:
```
          "7e400001xxxxxxxxxxxx",
          "7e400003xxxxxxxxxxxx",
```
aren't actually in your code?

Your characteristic is writable with a response and not writable without a response, so you need to pass true in `writeCharacteristic()`.

The third thing is that it's up to peripheral to decide when to send notification, so I have actually no idea whether it should send a notification every time it is written.

philos3, 2020-06-23:
>our characteristic is writable with a response and not writable without a response, so you need to pass true in writeCharacteristic().
it is the same when i pass true in writeCharacteristic

>since the stream has to be created before the actual call to the peripheral
i  created monitor before writeCharacteristic,please figure out the problem of my code

> those Xs: aren't actually in your code?
 those Xs are the clipped part of my code ,what i expect is to receive data from my device everytime,  but i can't receive any data from monitor().listen except the firstime.

 i want to recieve   the auto-response data  from my device anytime .is there any example code to achieve what i expect to do .

it is normal when i use flutter_blue,   here is part of my code 
###########################################################
           await controldCharacteristic.setNotifyValue(true);
            _notifySubscription =controldCharacteristic.value.listen((value)  {
              // do something with new value
            }

thaks for your help~

mikolak, 2020-06-23:
Try for now:
```
characteristic.monitor(...); //same as your code
**await Future.delayed(Duration(milliseconds: 500));**
await characteristic.write(...); //same as your code
```

I'll figure out an API to be notified when the notifications are actually turned on.

philos3, 2020-06-23:
> Try for now:
> 
> ```
> characteristic.monitor(...); //same as your code
> **await Future.delayed(Duration(milliseconds: 500));**
> await characteristic.write(...); //same as your code
> ```
> 
> I'll figure out an API to be notified when the notifications are actually turned on.

it works well for me ,thanks for your help~

closed 2020-06-23 by philos3